robotology/yarp #727: Alternative for modifying saving xml applications

tanismar, 2016-04-13:
Hi all, 

right now, when an application is modified using yarpbuilder/yarpmanager++, i is by default modified on the install path, meaning that if you want to "transport" that application to other setup, it is needed to copy it by hand to your local_code/app/scripts, before committing it to github. I understand this is because applications are setup specific, and hence the general "template" should be generic, and the particular nodes used and so on modified per setup. 

However, when building a new application this turns out to be really cumbersome, and very little elegant. I was wondering if there could be a way of modifying the templates on the local code, without assigning particular node names, just as we do with normal code which gets modified locally and then installed. Other option could be to do without templates altogether, and use config files which determine the nodes (and non-default parameters) with which an application would run on each setup.

lornat75, 2016-04-14:
A possible solution could be to to modify yarpmanager++ so that:

- By default it proposes to save files in user local directories
- When loading applications it separates those xml files coming from install path/build directory, marking them as "read only"; the yarp-config utility already does this, so the yarpmanger++ could do it as well

apaikan, 2016-04-18:
@lornat75  this means we have two copies of an application loaded by manager?  Will not be a bit confusing for the users? Anyway I think it is a good idea to categorize the applications based on their installed directories! (+1)

- What if we have a feature in the manager to update a template from its application?  e.g., Updated / Create Template.

tanismar, 2016-04-20:
That sounds like a good idea, but it would have to access the local path of the application to actually prevent the issue of having to copy it by hand. One option would be that in this case, it works as some sort of "save as.. " so user can copy it on its local folder, but where the variables concerning the nodes are reset to generic values localhost/console.
In general, I was wondering if there is no way to make apps a little bit more like the rest of the code, i.e., modified locally and then installed.