robotology/yarp #728: [Portmonitor] Modified data is passed to the following connections

drdanz, 2016-04-14:
When one connection uses a portmonitor plugin that modifies the data on the sender side, the following connections may receive the data modified by that plugin, depending on which `send` is executed first.

As discussed with @apaikan, fixing this requires a copy of the data sent.

In order to reduce the number of copies required, the suggested algorithm is this:

1. reorder the vector containing the `PortCoreOutputUnits`  so that the connections using non-modifying carriers are always at the begin and the modifying ones at the end.
2. send the original data to all the connections using non-modifying carriers.
3. send a copy of the data to each connection using a modifying carrier except for the last one.
4. send the original data to the last connection using a modifying carrier.

In this way, the number of copies is always `number of modifiying carrier - 1`.

A workaround is to make a copy of the data inside the `update` method of the `MonitorObject`-derived class.

apaikan, 2016-04-15:
Let me points some issues regarding the implementation. 
- **copying data:**

  The data passed to the portmonitor is a reference of `yarp::os::PortWriter`: The most primitive data type in YARP. Hence creating a copy of that object requires casting to the actual inherited type which is unknown in PortCore. 

   A workaround would be to write the data to a dummy connection (e.g., `yarp::os::DummyConnector`) and read it back. **This also requires knowing the actual inherited type of the object!!!**  See the implementation of `yarp::os::Things::cast_as<T>()`

- **Delegating data copy to the portmonitor:**
  Same problem here! Portmonitor does not know about the data type! 

- **Delegating data copy to the `yarp::os::Thing`:**
  The `Things` class knows about the data type and theoretically can create the correct copy of the object inside ` Things::cast_as<T>()` method.  The only issue arises here is that the method should know when a copy of data is needed;  Probably `ModifyingCarrier::update()` method can have a parameter indicating when the data copy is required! 

- **Delegating data copy to the user**
  Simple but error prone!

drdanz, 2016-05-03:
@apaikan What about making the `Thing` argument in `accept`, `update` and `updateReply` `const`? In this way, the user is forced to make a copy if he wants to modify the input, and you don't need to make the copy explicitly.