robotology/yarp #729: [Portmonitor] Envelope is lost on receiver

drdanz, 2016-04-14:
When connecting a portmonitor on the receiver side, the envelope is lost.

apaikan, 2016-04-14:
@drdanz  would you show me your example? 
I have tried the following example and the envelope works fine: 

- **Console 1**
```
$ yarpdev --device test_grabber --width 1 --height 1 --framerate 1 --stamp
```
- **Console 2**
```
$ yarp read /r envelope
yarp: Port /r active at tcp://10.255.72.102:10005
yarp: Receiving input from /grabber to /r using tcp+recv.portmonitor+context.monitors+file.bot_modifier
||| did not find bot_modifier
in create!
2 1460637.03642 [mat] [rgb] (3 8 8 1 1) {0 0 0 1 4 6 0 6} "modified from Lua :)"
3 1460637.03742 [mat] [rgb] (3 8 8 1 1) {0 0 0 1 4 6 0 6} "modified from Lua :)"
4 1460637.03842 [mat] [rgb] (3 8 8 1 1) {0 0 0 1 4 6 0 6} "modified from Lua :)"
...
```

- **Console 3**
```
yarp connect /grabber /r tcp+recv.portmonitor+context.monitors+file.bot_modifier
```

drdanz, 2016-04-14:
Sorry, my fault. The issue was actually that for some linking reason the carrier was loaded only on the sender side (even if both processes were running on the same machine using the same lib) and the manager shows that the ports are connected even if the connect failed.

closed 2016-04-14 by drdanz

drdanz, 2016-09-22:
Duplicate of #909

commit f7290e5c (2016-09-26): PortMonitor,Bayer: Set parent connection reader in modifyIncomingData

Fixes #909
Fixes #782
Fixes #729