robotology/yarp #730: icub-cluster in yarpmanager++

drdanz, 2016-04-14:
It would be really nice to have all the features provided by icub-cluster in yarpmanager++.

lornat75, 2016-04-14:
@apaikan: can we rely on the SSH deployer in the yarpmanager?

drdanz, 2016-04-14:
Also it would be useful to be able to see all the resources available on the yarp network instead of just the ones configured in the xml file

apaikan, 2016-04-15:
@lornat75  Yes! we just need to implement the SSH deployer ;) It is not indeed a complicated job but would be an issue on windows!

traversaro, 2017-07-11:
@Nicogene I will offer you a beer when you solve this. :D

pattacini, 2017-07-11:
Microsoft should be supposed to make PowerShell ssh-friendly.
Don't know the current status; in the meanwhile, we could rely on [Posh-SSH](http://www.thomasmaurer.ch/2016/04/using-ssh-with-powershell).

apaikan, 2017-07-11:
one more beer from me ;)
by the way, where is @nicogene ? :D

pattacini, 2017-07-11:
You must be drinking already some 🍺 dude 😉

Nicogene, 2017-07-12:
Here I am :grinning: It is quite a messy period, but as soon as i will more free I will work on the `yarpmanager` wishlist, that's is quite long :sweat_smile:

Nicogene, 2017-08-02:
Added by #1306, for now create a folder "yarpmanager" in `.local/share/yarp/contexts` and put inside of it your cluster-config.xml. We will soon update `robots-configuration` for adding the `yarpmanager` context.

traversaro, 2017-08-02:
It should work if the `cluster-config.xml` is found in any directory present in the YARP ResourceFinder search path, right?

traversaro, 2017-08-02:
I have some other comments related to the implementation: 
* `cluster-config.xml` is a rather generic name, I would suggest to at least support another less generic default name, such as `yarpmanager-cluster-config.xml`
* I checked in `robots-configuration`, and there a few use of the `iCubCluster` with names different from `cluster-config.xml`, for example https://github.com/robotology/robots-configuration/blob/593d1d170896f14baea38a1795c7ff90af0bad0b/iCubLisboa02/scripts/icubInterface.xml , https://github.com/robotology/robots-configuration/blob/593d1d170896f14baea38a1795c7ff90af0bad0b/iCubLisboa02/scripts/no-cluster.xml and https://github.com/robotology/robots-configuration/blob/593d1d170896f14baea38a1795c7ff90af0bad0b/iCubLisboa01/vislab-cluster.xml . I would suggest to make the cluster configuration file an option in the manager : https://github.com/robotology/yarp/blob/master/src/yarpmanager/src-manager/main.cpp#L27 .

Nicogene, 2017-08-02:
Unfortunately no, it is not found if not in `yarpmanager` context, honestly I never understood how `ResourceFinder` works exactly :sweat_smile: .
The default context is set to `yarpmanager` in `main.cpp` and this is how I try to load the file:
```
#ifdef WIN32
    ui->tabWidgetLeft->tabBar()->hide();
#else
    yarp::os::ResourceFinder& rf = yarp::os::ResourceFinder::getResourceFinderSingleton();

    std::string confFile = rf.findFileByName("cluster-config.xml");
    if (!confFile.empty())
    {
        ui->clusterWidget->setConfigFile(confFile);
        ui->clusterWidget->init();
        connect(ui->clusterWidget, SIGNAL(logError(QString)), this, SLOT(onLogError(QString)));
    }
    else
    {
        ui->tabWidgetLeft->tabBar()->hide();
    }
#endif
```

traversaro, 2017-08-02:
> Unfortunately no, it is not found if not in yarpmanager context, honestly I never understood how ResourceFinder works exactly 😅 .

I am quite sure that `findFileByName` should find a file if it is present in a directory in the YARP ResourceFinder search path, for example if it is present in the directory in which you launched the yarpmanager or if it is present in `~/.local/share/yarp/robots/${YARP_ROBOT_NAME}/`.

Nicogene, 2019-04-24:
>Microsoft should be supposed to make PowerShell ssh-friendly.

The time has come, OpenSSH client was introduced in the [Windows 10 Fall Creators Update](https://support.microsoft.com/en-us/help/4028685/windows-10-get-the-fall-creators-update), then we can remove the #ifdef WIN32 inside the clusterWidget.

Nicogene, 2019-12-12:
Created a dedicated issue for this task #2152 .

This can be considered closed

PR #1306 (2017-08-02): Add iCubCluster functionality to yarpmanager
This PR import iCubCluster inside yarpmanager that looks like: 
[image]

It has been successfully tested on `iCubGenova01` setup and it has (for now) the same functionalities of `icub-cluster.py`.

on `WIN32` the iCubCluster panel is for now disabled (sorry @randaz81 , @pattacini  and @lornat75 :sweat_smile: ).

@traversaro you owe me a beer :stuck_out_tongue:.

We should discuss about the cross dependency with icub-main( it needs only the `cluster-config.xml`) and about the deployer for windows.
We could keep it as it is for now and try to figure o [... cut]
closed 2019-12-12 by Nicogene