robotology/yarp #732: Connect returns success even if portmonitor fails to connect

drdanz, 2016-04-14:
When 2 ports are connected using a portmonitor, the connection succeeds, but the portmonitor fails to load (for example for a faulty dll), the connect call returns that the connection is ok and the manager show that the ports are connected, even if no data is received.

apaikan, 2016-04-15:
We had a long discussion and try with @drdanz  on this issue. 

- The portmonitor plugin (same as any other carrier modifier) is loaded after the connection is established! This is the way carrier modifiers implemented in YARP. `Portmonitor::configure()` method correctly returns `true/false` but the return value has not been used in YARP for creating connections! 

- A simple and quick solution would be to delegate plugin loading to the PortCore prior establishing the connections and passing a reference to the Carrier! 

- Another complicated solution is to go through all the steps of establishing connections and loading carrier modifiers inside the `Protocol` and take care of the return value of the `Portmonitor::configure()`!

drdanz, 2016-04-15:
The problem is actually not the portmonitor, but *any* modifier carrier that fails.
It can be easily reproduced by doing `yarp connect /foo /bar udp+send.tcp`.