robotology/yarp #733: Windows compile from sources errors (ACE redefinition errors)

JimmyDaSilva, 2016-04-15:
I had previously installed YARP from the precompiled executable and everything worked properly. Now I am trying to compile Yarp from sources on Windows 10 in order to compile the C# bindings. I first uninstalled the precompiled version of yarp, then installed ACE and GSL only from the same precompiled installation, and finally got the sources of yarp from github. CMake configuration was successful but when I tried to build Visual Studio I get many errors about ACE redefining c++ classes/structures. Here is an example:

```
2>C:\Program Files\robotology\ace-6.1.1\ace/os_include/os_time.h(59): error C2011: 'timespec' : redéfinition du type 'struct'
2>  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10586.0\ucrt\time.h(39): note: voir la déclaration de 'timespec'
3>  tinyxmlerror.cpp
2>C:\Program Files\robotology\ace-6.1.1\ace/OS_NS_string.inl(519): warning C4996: 'wcstok': wcstok has been changed to conform with the ISO C standard, adding an extra context parameter. To use the legacy Microsoft wcstok, define _CRT_NON_CONFORMING_WCSTOK.
2>  C:\Program Files (x86)\Windows Kits\10\Include\10.0.10586.0\ucrt\corecrt_wstring.h(254): note: voir la déclaration de 'wcstok'
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(91): error C4430: spécificateur de type manquant - int est pris en compte par défaut. Remarque : C++ ne prend pas en charge int par défaut
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(91): error C2143: erreur de syntaxe : absence de ',' avant '&'
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(109): error C4430: spécificateur de type manquant - int est pris en compte par défaut. Remarque : C++ ne prend pas en charge int par défaut
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(109): error C2143: erreur de syntaxe : absence de ',' avant '&'
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(191): error C2833: 'operator timespec_t' n'est pas un opérateur ou type reconnu
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(191): error C2059: erreur de syntaxe : 'newline'
2>C:\Program Files\robotology\ace-6.1.1\ace/Time_Value.h(191): error C2238: jetons inattendus avant ';'
```

Sorry log is half in French... Could not find a way to change the language even if Visual Studio language is set to English ...

I have attached the full build log as it is quite long. 
[errors.txt](https://github.com/robotology/yarp/files/220879/errors.txt)

I develop primilarly on Linux and not Windows so I may have forgot something simple.
Any idea what is happening?
Thanks your for your help.

pattacini, 2016-04-15:
Hi @JimmyDaSilva 

Which version of Visual Studio are you using? As of now, we don't support the latest VS 2015 yet, given that we still need to provide _ad hoc_ packages (e.g. ACE) for it.

pattacini, 2016-04-15:
I'm on Windows 10 working with VS 2013 and it's fine.

JimmyDaSilva, 2016-04-15:
Yes I am on VS 2015. Going to try VS 2013 then. I will let you know.
Thank you so much @pattacini !

lornat75, 2016-04-15:
Unfortunately I think the problem is that we need to wait for ACE to support VS 2015, right @pattacini ?

JimmyDaSilva, 2016-04-15:
@pattacini Built successfully on VS 2013. Thank you for your help and reactivity!

closed 2016-04-15 by JimmyDaSilva