robotology/yarp #735: Windows linking errors

JimmyDaSilva, 2016-04-18:
Hi all,
I am still having a hard time using yarp on Windows and Visual Studio 2013.
I have set the lib and include directories in my projects, set the environment variables by following the guide for windows...
On a terminal, `yarpserver` runs fine.
But I get error like these:
```
error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl yarp::os::ConstString::~ConstString(void)" (__imp_??1ConstString@os@yarp@@QEAA@XZ)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl yarp::os::ConstString::ConstString(char const *)" (__imp_??0ConstString@os@yarp@@QEAA@PEBD@Z)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl yarp::os::Bottle::Bottle(void)" (__imp_??0Bottle@os@yarp@@QEAA@XZ)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl yarp::os::Port::Port(void)" (__imp_??0Port@os@yarp@@QEAA@XZ)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: __cdecl yarp::os::Network::Network(void)" (__imp_??0Network@os@yarp@@QEAA@XZ)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl yarp::os::Network::~Network(void)" (__imp_??1Network@os@yarp@@UEAA@XZ)
error LNK2001: unresolved external symbol "__declspec(dllimport) public: static bool __cdecl yarp::os::NetworkBase::checkNetwork(double)" (__imp_?checkNetwork@NetworkBase@os@yarp@@SA_NN@Z)
```

@pattacini You said you are using VS2013? Did you ever have this problem before ?

Thank you guys!

drdanz, 2016-04-18:
you are probably missing `target_link_libraries(your_target ${YARP_LIBRARIES})` in your project CMakeLists.txt

JimmyDaSilva, 2016-04-18:
Sorry but there is no CMake involved here. I am compiling with Visual Studio

drdanz, 2016-04-18:
In this case you have to link your project to libYARP_OS and libYARP_init in visual studio project properties.

JimmyDaSilva, 2016-04-18:
I have set the LIBPATH to `robotology\yarp-2.3.64\lib` which contain `YARP_OS.lib` and `YARP_init.lib`
[image]

drdanz, 2016-04-18:
I think that you have to add them in the input tab...

JimmyDaSilva, 2016-04-18:
Nice that works :). 
I don't understand though. Isn't it supposed to load all the libs in the specified path ?

drdanz, 2016-04-18:
No, just imagine what it would happen on a linux system, where all the libraries are in `/usr/lib`...
You are supposed to tell the linker which libraries it should link, and where to find them.

JimmyDaSilva, 2016-04-19:
Sorry to bother you again but I still have issues on VS2013. It builds now but crashes when running.
YARP seems to be missing a path to a DLL. I tried a bunch of stuff like changing the resources path and so on...
```
'KinectHandTracking.exe' (Win32): Loaded 'C:\Users\jimda\Documents\Visual Studio 2013\Projects\KinectHandTracking\x64\Release\KinectHandTracking\AppX\KinectHandTracking.exe'. Symbols loaded.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Cannot find or open the PDB file.
'KinectHandTracking.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Cannot find or open the PDB file.
The thread 0x115c has exited with code -1073741515 (0xc0000135).
The thread 0x2b68 has exited with code -1073741515 (0xc0000135).
The program '[640] KinectHandTracking.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'.
```

[image]

Thanks !

pattacini, 2016-04-20:
Hi @JimmyDaSilva 

From your previous post I imagine you're not using `cmake` to create the VS project. Is that right?

If so, let me point out that this approach can be considered somehow an anti-pattern, meaning that handling directly VS options is very likely to bring about problems - as you're experiencing - because of some missing libraries, include paths and so on so forth, which are difficult also for us to figure out.

On the other hand, using `cmake` is easy, neat, scalable, maintainable and of course intrinsically multi-platform and comes with a very tiny cost.

Would you be willing to switch to the use of `cmake`? Is there any particular requirement not to use it?

JimmyDaSilva, 2016-04-20:
Hi @pattacini. Thank you for you answer.

Yes I am not using `cmake` to create the VS Project.

I know `cmake` is great, even if I can't say I am an experienced user. My main concern is that I need to generate a Windows App. I want to use the Kinect2 SDK to get information and send them over using YARP.

Using the `cmake` example file works fine. The example is generated and runs. But it is not a Windows App but a console application in which I cannot add the Microsoft SDK in.

I tried to copy the properties of the example project but without success.

I am frustrated because it worked one night. Then computer got restarted or I changed something I shouldn't have and it's broken now.

Thank you for your concern :)

traversaro, 2016-04-20:
Fortunatly CMake has support for generating a Windows GUI application in a Windows Store project. 

To generate a Windows GUI executable, just add WIN32 to the add_executable call for your GUI ( https://cmake.org/cmake/help/v3.0/command/add_executable.html ).

To generate a Windows Store project, you can follow this example in CMake configuration : https://cmake.org/cmake/help/v3.1/manual/cmake-toolchains.7.html?highlight=windowsstore#cross-compiling-for-windows-store .

JimmyDaSilva, 2016-04-20:
Going to give this a shot! Thanks for the help!

JimmyDaSilva, 2016-04-20:
I successfully generated a windows Store project including yarp with `cmake`.
Everything builds but I get the same error of missing dependent dll when running.
I only have the `main.cpp` from the example.
When I remove yarp objects the project runs

lornat75, 2016-04-20:
Looking at your error log I do not understand the problem. The "missing pdb" information should be only a warning, which btw should disappear if you run the "release" version of your program. However, the error seems to be related to a missing "dll". Can't you spot anything specific to this problem in the output of your program?

lornat75, 2016-04-20:
ok, I should have checked more carefully. It seems to be something specific to windows apps:

Did yuo try the suggestions at this link:

http://stackoverflow.com/questions/29214446/referencing-native-c-dll-in-c-windows-app-only-works-when-added-as-reference

JimmyDaSilva, 2016-04-20:
More dll gets loaded now, but not enough. I have included `YARP_OS.dll`, `YARP_OSd.dll`, `YARP_init.dll`,`YARP_initd.dll`

```
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Users\jimda\Documents\ProjectKinect2Yarp\build\Debug\AppX\Kinect2HandYarp.exe'. Symbols loaded.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Users\jimda\Documents\ProjectKinect2Yarp\build\Debug\AppX\YARP_OSd.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\profapi.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Windows\System32\ws2_32.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Program Files\WindowsApps\Microsoft.VCLibs.120.00.Debug_12.0.21005.1_x64__8wekyb3d8bbwe\msvcp120d_app.dll'. Cannot find or open the PDB file.
'Kinect2HandYarp.exe' (Win32): Loaded 'C:\Program Files\WindowsApps\Microsoft.VCLibs.120.00.Debug_12.0.21005.1_x64__8wekyb3d8bbwe\msvcr120d_app.dll'. Cannot find or open the PDB file.
The thread 0x2f28 has exited with code -1073741515 (0xc0000135).
The thread 0x2cf4 has exited with code -1073741515 (0xc0000135).
The program '[12060] Kinect2HandYarp.exe' has exited with code -1073741515 (0xc0000135) 'A dependent dll was not found'.
```

lornat75, 2016-04-20:
What is strange is that usually the executable should state which DLL is not found, this helps a lot.

Try adding YARP_devb.dll, YARP_mathd.dll... there could be other DLLs missing (the ones kinect related to the kinect driver, for example).

JimmyDaSilva, 2016-04-20:
I have added all Yarp's dlls to check, same proble.
I haven't included the kinect here yet. It's just the yarp example in the Windows Store format.

Yep very frustrating that VS is not display the missing dll...

JimmyDaSilva, 2016-04-20:
Just checked again. Windows blank app builds and runs.
But if a yarp object is created in the program I get the dll missing problem

JimmyDaSilva, 2016-04-20:
Found it!
I added also ACE.dll by using @lornat75 's link and it runs now :)

Problem is that the communication with the yarpserver does not work ...
```
yarp::os::Port leftHandPositionPort;
leftHandPositionPort.open("/kinect/leftHandPosition:o");
```

This opens a port in a console application, but not the windows store app.
A permission issue ?

lornat75, 2016-04-20:
It could be a a permission issue. Any way to catch the stdout?

JimmyDaSilva, 2016-04-20:
Yep just found a way. 

```
	yarp::os::Network yarp;
	double network_timeout = 10000.0;
	if (!yarp.checkNetwork(network_timeout))
	{
		OutputDebugStringW(L"YARP network is not available\n");
		return;
	}
	OutputDebugStringW(L"YARP network is available\n");
```

Returns `YARP network is not available`

lornat75, 2016-04-20:
can you try testing basic usage of yarp at a normal console, following for example the first example here:

http://www.yarp.it/note_ports.html

JimmyDaSilva, 2016-04-20:
Works fine when running commands via console, like `yarp read /reader`.
The above code works fine with a console application but fails when run by a Windows App

lornat75, 2016-04-20:
Potential issues I see are access problems with the following:

- The file yarp.conf (to see where it is located in your system lunch yarp conf at the command line)
- Sockets in general, if windows apps are executed in a protected environment 

I am just speculating, I know nothing about windows apps.

JimmyDaSilva, 2016-04-20:
yarp.conf location:
`C:\Users\jimda\AppData\Roaming\yarp\config\yarp.conf`

`192.168.99.1 10000 yarp`

Should I change that file to something else ?

Any help will be appreciate but I understand that Windows Apps are quite annoying and not really of your concern.

Thank you for the time you spent on this.

lornat75, 2016-04-20:
Do you have a way to set in the app the environment variable YARP_CONF to C:\Users\jimda\AppData\Roaming\yarp\config ?

JimmyDaSilva, 2016-04-20:
I don't know but I can try tomorrow. You think the problem is the accessibility to the `yarp.conf` file ?

lornat75, 2016-04-20:
It is possible that %APPDATA% is not expanded as in a console application. By setting YARP_CONF you force the location of the yarp.conf file to a directory of your choice (assuming it is accessible from the app). Alternatively you can copy the yarp.conf file to wherever %APPDATA% points to in a windows app.

JimmyDaSilva, 2016-04-20:
I tried running this:
```
	yarp::os::Network yarp;
	bool scanNeeded, serverUsed;
	yarp::os::Contact contact_yarp  = yarp.detectNameServer(true, scanNeeded, serverUsed)
```

Here is the output from the yarpserver:
```
C:\WINDOWS\system32>yarpserver
...
...
Ok.  Ready!
 * NAME_SERVER query "/root"
ASKED NAME_SERVER query /root, GAVE registration name /root ip 192.168.99.1 port 10000 type tcp
 * NAME_SERVER query "/root"
ASKED NAME_SERVER query /root, GAVE registration name /root ip 192.168.99.1 port 10000 type tcp
 * NAME_SERVER query "/root"
ASKED NAME_SERVER query /root, GAVE registration name /root ip 192.168.99.1 port 10000 type tcp
```

So some communication seems to work.
What should I do? Why does `detectNameServer` works but `checkNetwork` fails ?

lornat75, 2016-04-21:
The yarp.conf file is exactly a way to skip detection of nameserver. If the file is not found there could be problems connecting to the nameserver. Did you try my suggestion above?

JimmyDaSilva, 2016-04-22:
I didn't succeed in setting APPDATA for the windows app environment. The APPDATA macro is properly set and available in Visual Studio, but I can't figure out if the App has actually access to this value

lornat75, 2016-04-22:
What about setting YARP_CONF?

Sent from mobile device

JimmyDaSilva, 2016-04-22:
What do you mean? As a global user environment variable ?
Does yarp read that variable somewhere within its fonctions ?

JimmyDaSilva, 2016-04-22:
I have set `YARP_CONF` to the desktop and saved the conf file there. `checkNetwork` stills fails within the Windows App

JimmyDaSilva, 2016-04-27:
@lornat75 I finally decided to use sockets instead.
I am sending the Store App data through a socket, then from a console application I read the data and give it right back to Yarp.
You can choose to close the issue or keep it opened but I don't think I will try working on this issue any time soon.

Thanks again for your time and help on the matter!

lornat75, 2016-04-27:
if you are willing to share the original app code I may give it a try myself to see what goes wrong.

JimmyDaSilva, 2016-04-28:
Here is a minimal VS2013 project of a Windows App reproducing the problem. 

Everything compiles fine.
I had to copy/paste the some DLLs in `YarpApp\YarpApp` so that it runs.
Problem is that the `yarp::os::Network::checkNetwork` fails, so does port creation.
But strangely `yarp::os::Network::detectNameServer` makes some input to the yarpserver terminal:

```
C:\WINDOWS\system32>yarpserver
...
...
Ok.  Ready!
 * NAME_SERVER query "/root"
ASKED NAME_SERVER query /root, GAVE registration name /root ip 192.168.99.1 port 10000 type tcp
 * NAME_SERVER query "/root"
ASKED NAME_SERVER query /root, GAVE registration name /root ip 192.168.99.1 port 10000 type tcp
```

https://drive.google.com/file/d/0B61-Kf77E1hUenF1VFdZV2NJTUE/view?usp=sharing

lornat75, 2016-05-05:
I gave this a try but ran into problems. I think the problems are related to permissions with windows applications but debugging requires more time and knowledge of the windows universal app framework.

closed 2016-05-05 by lornat75