robotology/yarp #738: YARP compilation for Android

apaikan, 2016-04-20:
There are still some problems regarding compiling YARP for Android. Years ago with @alecive  we solved some of these issues but currently @vtikha experienced new issues. We need to elaborate the build system such that: 

- We need to add a specific flag for Android to the YARP build cmake lists
- `lpthread` and `rt` libraries does not exists as separate libraries in android NDK.  When cross compiling for Android, these libraries should no be added to the library list
- `SKIP_ACE` and `CMAKE_DISABLE_FIND_PACKAGE_OpenCV` flags should be enabled by default while compiling for Android

- Unnecessary tools (yarprun, manager, ...) should be disabled and only YARP core libraries need to be compiled 

@drdanz  I can give you hand for a quick fix!

apaikan, 2016-04-20:
By the way @alecive has provided a well documented tutorial for this: 
http://alecive.github.io/blog/2015/08/31/YARP-Cross-Compilation/

alecive, 2016-04-20:
@apaikan :+1: only because there is no +1000
So was the tutorial still working? I was worried it was going to become obsolete or broken over the time.

vtikha, 2016-04-20:
@alecive The tutorial is missing some parts that are needed to compile it successfully. There have been a few changes and I have the list of modifications made.

alecive, 2016-04-20:
@vtikha I would be glad to add them to the tutorial then :smile: 
I just checked, and a quick diff between the current yarp version and my modified files shows that there have been modifications to `yarpOS` in between, so it makes sense.

vtikha, 2016-04-20:
I had sent you the list of changes on the 12th of Feb :wink: I can send it again if needed.

alecive, 2016-04-20:
ahah ops :smile:

alecive, 2016-04-20:
Ok issue tracked there -> https://github.com/alecive/alecive.github.io/issues/4
Did you have any problem with the cmake?

apaikan, 2016-04-20:
@alecive  wait until we apply the necessary modifications into the YARP! I guess your tutorial will become simpler :)  

P.S. @alecive  as a unique and only reference for YARP-Android, your tutorial is a treasure!

apaikan, 2016-04-20:
*Did you have any problem with the cmake?*

Apart form the '-llpthread and -lrt', the rest was fine and compiled smoothly!

vtikha, 2016-04-20:
@apaikan there were other issues with the compilation, not only he ones you mentioned (cmake related)

vtikha, 2016-11-17:
@apaikan can we close this issue? @alecive I suggest we create a documentation within YARP for the compilation with android. If you agree I can take care of that.

drdanz, 2016-11-17:
CMake 3.7 include a lot of new features to support Android, I'm not sure if this can be useful...

alecive, 2016-11-17:
@vtikha Feel free to do whatever you prefer :smile: I can help if needed, although at this point I believe you have a much better picture than me on this :+1:
@drdanz Interesting! I took a look [here](https://cmake.org/cmake/help/v3.7/manual/cmake-toolchains.7.html#cross-compiling-for-android) and it seems to include everything we need, but I don't really have time to work on this :smile: