robotology/yarp #739: "pull" connections do not use the send modifier

drdanz, 2016-04-21:
When using a connection line like this:

```
yarp connect /w /r udp+send.portmonitor+...
```

Everything works correctly and the data is filtered by the portmonitor.
For a connection line like this, instead:

```
yarp connect /w /r mjpeg+send.portmonitor+...
```

The sent data is supposed to go through the portmonitor, and then transferred using the mjpeg protocol, but instead the portmonitor is skipped.

apaikan, 2016-04-25:
Not all the carriers in YARP can be chained! The "mjpeg" carrier implements very basic transport functionalities into the YARP communication infrastructure. Basically it has the carrier modifier in its implementation! 

The portmonitor is just carrier modifier and the users expect to be able to connect it transparently at each side of the connection. In your case (the 'mjpeg' carrier), if the other side of the connection is not a Yarp port, then the there is no place for attaching portmonitor!  

Anyway, it is still possible to attach the portmonitor to the port instead of the connection. And i expect it works even with the mjpeg carrier.

drdanz, 2016-04-26:
Why should it not be able to do it?
The mjpeg is just the "transport" layer protocol (in the same way as `tcp` or `udp`), while the portmonitor is the modifier. Images streamed by the sender are compressed on the sender side and decompressed on the receiver side, but the port sees an image exactly like if it was transferred using `udp` or `tcp`. By the way, `udp` and `tcp` are unfortunate names for representing the protocol spoken by YARP, but are "protocols" that run on top of `udp` and `tcp`, just like `mjpeg`, but with the difference that the transfer is initiated by the sender, while for mjpeg the protocol starts with a request of the receiver.

Internally it should be happening something like this:

```
modifier -> transport --------(network)--------> transport -> modifier
```

Which means that the portmonitor on the receiver side could be used, for example, to filter the images received or to do some other processing.