robotology/yarp #741: yarprobotinterface segfault with Vizzy

plinioMoreno, 2016-04-26:
Around 70% of the times I run yarprobotinterface on [Vizzy](https://github.com/robotology/icub-main/tree/master/app/robots/vizzy), using the same configuration files that I had for robotInterface, I got the following error:

```
[DEBUG]left_arm_calibrator finished calibration of device left_arm_mc 
[INFO]All actions for action level 11 of startup phase finished. 
[Thread 0x7fff4affd700 (LWP 7937) exited]
[INFO]startup phase finished. 
[INFO]run phase starting... 
[DEBUG]robot-interface running happily 
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff69043d5 in malloc_consolidate (
    av=av@entry=0x7ffff6c44760 <main_arena>) at malloc.c:4165
4165	malloc.c: No such file or directory.
```
The program crashes right after finishing the calibration part. This crash never happens with the current version of robotInterface. The yarp version that I have is
```
commit bb889a055aee6f2d57a708a7890183201937cbcf
Author: Marco Randazzo <marco.randazzo@iit.it>
Date:   Fri Apr 1 15:27:50 2016 +0200
```

lornat75, 2016-04-26:
@drdanz can we double check that yarprobotinterface is not "behind" w.r.t to robotinterface?

drdanz, 2016-04-27:
checked, `yarprobotinterface` contains all the commits in iCub `robotInterface`

@plinioMoreno Can we see the whole log and the xml files?

plinioMoreno, 2016-04-27:
@drdanz,

The xml files are [here](https://github.com/robotology/icub-main/tree/master/app/robots/vizzy). 
And the log is [myLog.txt](https://github.com/robotology/yarp/files/238600/myLog.txt).

We have a problem with one motor that causes disabling other two motors due to the joint calibration procedure. Our technician is out, so we will be able to solve that motor problem only the coming week. But even before that problem, the yarprobotinterface was crashing and robotInterface was not crashing.

drdanz, 2016-08-04:
Is this still an issue?

plinioMoreno, 2016-08-10:
I have not done further tests using fresh versions of the yarp master. Feel free to close it in the meantime, and I will re-open it if is still present after executing new tests using fresh versions of yarp master (which will be only in around one month)

drdanz, 2016-08-10:
In this case I'm closing this, thanks.

closed 2016-08-10 by drdanz

plinioMoreno, 2016-09-27:
I run the yarprobotinterface with the fresh master yarp repository, and I did not run into this error anymore.