robotology/yarp #742: Interrupted udp connections are still shown as active in the yarpmanager

lornat75, 2016-04-27:
This happens because udp links are connectionless; when the receiver is killed the sender has no way to realize it and keeps sending data. This is not a problem in command line YARP because `yarp connect` always disconnects before doing new connections. However it affects the manager, because -- to avoid problems to existing connections -- the manager does not re-connect existing connections. This causes apparent udp connections to appear in the manager.

lornat75, 2016-04-27:
More on this: the command: 

`yarp exists /sender /receiver`

is not affected by this, I think the manager could implement the same behavior when checking for existing connections.

@apaikan

drdanz, 2016-04-27:
@lornat75 I think I have a patch for this, but I didn't commit it yet because it had some issues... I will work on this as soon as I have some time.