robotology/yarp #745: robotInterface: corrupted double-linked list after CTRL+C

pattacini, 2016-04-29:
@davidetome commented on [Tue Jan 19 2016](https://github.com/robotology/icub-main/issues/276)

*** glibc detected *** robotInterface: corrupted double-linked list: 0x000000000c75c800 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x75be6)[0x7fa5ad13abe6]
/lib/x86_64-linux-gnu/libc.so.6(+0x7605d)[0x7fa5ad13b05d]
/lib/x86_64-linux-gnu/libc.so.6(+0x77518)[0x7fa5ad13c518]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7fa5ad13f98c]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os4impl14PortCorePacket5resetEv+0x2e)[0x7fa5ae2ad89e]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN26PortWriterBufferBaseHelperD0Ev+0x15b)[0x7fa5ae2c282b]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os20PortWriterBufferBaseD1Ev+0x21)[0x7fa5ae2c1911]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD1Ev+0x447)[0x7fa5765d3dc7]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN4yarp3dev12AnalogServerD0Ev+0x9)[0x7fa5765d4269]
/usr/local/src/robot/icub-main/build-pc104/lib/iCub/skinWrapper.so(_ZN11skinWrapper5closeEv+0x22)[0x7fa5765c3522]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_dev.so.1(_ZN4yarp3dev10PolyDriver9closeMainEv+0x44)[0x7fa5adeb9d34]
robotInterface(_ZN14RobotInterface6Device5closeEv+0x20)[0x4168d0]
robotInterface(_ZN14RobotInterface5Robot7Private12closeDevicesEv+0x46)[0x41ecd6]
robotInterface(_ZN14RobotInterface5Robot10enterPhaseENS_11ActionPhaseE+0x1bee)[0x424ebe]
robotInterface(_ZN14RobotInterface6Module5closeEv+0x55)[0x41a605]
/usr/local/src/robot/yarp/build-pc104/lib/libYARP_OS.so.1(_ZN4yarp2os8RFModule9runModuleEv+0x211)[0x7fa5ae2e2b61]
robotInterface(main+0xe2)[0x413c82]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7fa5ad0e3ead]
robotInterface[0x413fd9]
======= Memory map: ========
00400000-00450000 r-xp 00000000 00:17 12324236                           /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
00650000-00652000 rw-p 00050000 00:17 12324236                           /usr/local/src/robot/icub-main/build-pc104/bin/robotInterface
006d2000-0e1ad000 rw-p 00000000 00:00 0                                  [heap]
7fa5447f9000-7fa5447fa000 ---p 00000000 00:00 0 
7fa5447fa000-7fa544ffa000 rw-p 00000000 00:00 0 
7fa546ffe000-7fa546fff000 ---p 00000000 00:00 0 
7fa546fff000-7fa5477ff000 rw-p 00000000 00:00 0 
7fa548000000-7fa54803b000 rw-p 00000000 00:00 0 
7fa54803b000-7fa54c000000 ---p 00000000 00:00 0 
7fa54d7cb000-7fa54d7cc000 ---p 00000000 00:00 0 
7fa54d7cc000-7fa54dfcc000 rw-p 00000000 00:00 0 
7fa54f7cf000-7fa54f7d0000 ---p 00000000 00:00 0 
7fa54f7d0000-7fa54ffd0000 rw-p 00000000 00:00 0 
7fa5517d3000-7fa5517d4000 ---p 00000000 00:00 0 
7fa5517d4000-7fa551fd4000 rw-p 00000000 00:00 0 
7fa5537d7000-7fa5537d8000 ---p 00000000 00:00 0 
7fa5537d8000-7fa553fd8000 rw-p 00000000 00:00 0 
7fa568000000-7fa5680b5000 rw-p 00000000 00:00 0 
7fa5680b5000-7fa56c000000 ---p 00000000 00:00 0 
7fa56f185000-7fa56f19b000 r-xp 00000000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f19b000-7fa56f39a000 ---p 00016000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f39a000-7fa56f39b000 rw-p 00015000 00:17 12323984                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/parametricCalibrator.so
7fa56f39b000-7fa56f39c000 ---p 00000000 00:00 0 
7fa56f39c000-7fa56fb9c000 rw-p 00000000 00:00 0 
7fa56fb9c000-7fa56fb9d000 ---p 00000000 00:00 0 
7fa56fb9d000-7fa57039d000 rw-p 00000000 00:00 0 
7fa57039d000-7fa57039e000 ---p 00000000 00:00 0 
7fa57039e000-7fa570b9e000 rw-p 00000000 00:00 0 
7fa570b9e000-7fa570baa000 r-xp 00000000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570baa000-7fa570da9000 ---p 0000c000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570da9000-7fa570daa000 rw-p 0000b000 00:17 12324003                   /usr/local/src/robot/icub-main/build-pc104/lib/iCub/xsensmtx.so
7fa570daa000-7fa570dab000 ---p 00000000 00:00 0 
7fa570dab000-7fa5715ab000 rw-p 00000000 00:00 0 
7fa5715ab000-7fa5715ac000 ---p 00000000 00:00 0 
7fa5715ac000-7fa571dac000 rw-p 00000000 00:00 0 
7fa571dac000-7fa571dad000 ---p 00000000 00:00 0 
7fa571dad000-7fa5725ad000 rw-p 00000000 00:00 0 
7fa5725ad000-7fa5725ae000 ---p 00000000 00:00 0 
7fa5725ae000-7fa572dae000 rw-p 00000000 00:00 0 
7fa572dae000-7fa572daf000 ---p 00000000 00:00 0 
7fa572daf000-7fa5735af000 rw-p 00000000 00:00 0 
7fa5735af000-7fa5735b0000 ---p 00000000 00:00 0 
7fa5735b0000-7fa573db0000 rw-p 00000000 00:00 0 
7fa573db0000-7fa573db1000 ---p 00000000 00:00 0 
7fa573db1000-7fa5745b1000 rw-p 00000000 00:00 0 
7fa5745b1000-7fa5745b2000 ---p 00000000 00:00 0 
[... 818 lines cut ...]
7f1ec33a7000-7f1ec33a9000 r-xp 00000000 07:00 10300                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libdl-2.13.so
7f1ec33a9000-7f1ec35a9000 ---p 00002000 07:00 10300                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libdl-2.13.so
7f1ec35a9000-7f1ec35aa000 r--p 00002000 07:00 10300                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libdl-2.13.so
7f1ec35aa000-7f1ec35ab000 rw-p 00003000 07:00 10300                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libdl-2.13.so
7f1ec35ab000-7f1ec371e000 r-xp 00000000 07:00 23107                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/libACE-6.0.3.so
7f1ec371e000-7f1ec391d000 ---p 00173000 07:00 23107                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/libACE-6.0.3.so
7f1ec391d000-7f1ec392e000 r--p 00172000 07:00 23107                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/libACE-6.0.3.so
7f1ec392e000-7f1ec3932000 rw-p 00183000 07:00 23107                      /lib/live/mount/rootfs/filesystem.squashfs/usr/lib/libACE-6.0.3.so
7f1ec3932000-7f1ec3934000 rw-p 00000000 00:00 0 
7f1ec3934000-7f1ec394b000 r-xp 00000000 07:00 10307                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libpthread-2.13.so
7f1ec394b000-7f1ec3b4a000 ---p 00017000 07:00 10307                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libpthread-2.13.so
7f1ec3b4a000-7f1ec3b4b000 r--p 00016000 07:00 10307                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libpthread-2.13.so
7f1ec3b4b000-7f1ec3b4c000 rw-p 00017000 07:00 10307                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libpthread-2.13.so
7f1ec3b4c000-7f1ec3b50000 rw-p 00000000 00:00 0 
7f1ec3b50000-7f1ec3cd1000 r-xp 00000000 07:00 10150                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libc-2.13.so
7f1ec3cd1000-7f1ec3ed1000 ---p 00181000 07:00 10150                      /lib/live/mount/rootfs/filesystem.squashfs/lib/x86_64-linux-gnu/libc-2.13.soAborted

~~~

---

@pattacini commented on [Fri Apr 29 2016](https://github.com/robotology/icub-main/issues/276#issuecomment-215709833)

@randaz81 @barbalberto @drdanz
Please, apply any possible bug fix to **`yarprobotinterface`** instead.

randaz81, 2016-05-03:
A controlboardwrapper which is trying to attach to an already closed motorcontrol device may cause this issue.

pattacini, 2016-05-10:
Ping: what's the current status?
Who can we assign as responsible for this issue?

lornat75, 2016-05-10:
@barbalberto has recently fixed a couple of problems with robotinterface causing segmentation faults -- I am not sure they are the ones referred to in this issue.

drdanz, 2016-08-04:
Is this still an issue?

francesco-romano, 2016-08-04:
We don't have robots to test it now. 
As far as I remember, yes

pattacini, 2016-08-04:
Let me reinforce that I've been still seeing problems upon closure: seg-faults and hang-ups. Don't know if all these problems can be still categorized under this issue. A robot is required to verify the current status any how.

drdanz, 2016-08-04:
@pattacini you are on holiday! Go have fun! :laughing:

drdanz, 2016-11-10:
Does this still happen? I think it was fixed on icub-main `devel` branch by robotology/icub-main@b0e4bd9250d201cc3d40118eddad2bb29b871529, can anyone confirm this?

pattacini, 2016-11-10:
Honestly I'm still catching seg-faults of `yarprobotinterface` every once in a while...

drdanz, 2016-11-10:
@pattacini segmentation faults or "double link or corruption" errors?
Also I'm pretty sure that `yarprobotinterface` is not the problem here, it is very likely to be some buggy device...

pattacini, 2016-11-10:
Definitely, I'm not blaming `yarprobotinterface` that is merely a deployer.
To answer your questions: seg-faults.

However, what is happening after CTRL+C is somehow problematic.

drdanz, 2016-11-10:
Changing component to "Devices" then. It might also be a problem of some devices in icubmod, though.

drdanz, 2016-11-10:
Also might be fixed by #942

drdanz, 2016-11-10:
@pattacini #942 was merged, can you please let me know whether this still happens or not?

pattacini, 2016-11-11:
I'm afraid we have now more serious problems.
iCub does not even start with `yarp/devel` + `icub-main/devel`... 😞 

We just started with @randaz81 to try to nail down what's going wrong.

pattacini, 2016-11-14:
#### Update
Because of an address conflict we had to reboot `PC104`. Since then the problem disappeared. Very weird.

However, `yarprobotinterface` still seg-faults upon closure in at least a couple of different ways.

cc @randaz81

drdanz, 2017-02-16:
@pattacini Any idea about which devices are making it crash?

pattacini, 2017-02-16:
No idea sorry; I had no time to debug 😞 

However, recently, it seems that `yarptobotinterface` is more robust in this respect.

pattacini, 2017-02-17:
Even if it's still potentially risky because we didn't nail down the problem thus far, I'd tend to close this issue for the time being, given that we've not been experiencing other seg-faults.

closed 2017-02-17 by pattacini