robotology/yarp #747: New interface proposal : IMultipleAnalogSensors

traversaro, 2016-05-04:
In the context of developing a reasonable interface to the distributed inertial sensors (accelerometers, gyroscopes, and possibly in the future fully fledged IMUs) mounted on the iCub [1] we encounter several challenges, that we detailed in the following. To overcome this difficulties, we discussed (in different times/places) with @marcoaccame and @barbalberto a possible backward-compatible extension of the `IAnalogSensor`/`IGenericSensor` interface to smoothly support publishing of distributed inertial sensors. In the following we will focus on the iCub use-case that is motivating this discussion, but clearly the proposed new interfaces will be totally robot-independent. 

## How distributed inertial sensors are currently published
Currently the distributed accelerometers are exposed using the `IAnalogSensor` interface. The 
format with which they are published differs from CAN and ETH iCubs : 
 * On CAN iCub, accelerometers for each link are published on the same vector, using the format described in [1].  
 * On ETH iCubs, accelerometers for each body part are published on the same port, but in the measurement vector also metadata (the sensor location, type and timestamps) are published together with the data. 

##  Problems with the current interfaces 
* Tipically the `IAnalogSensor` or `IGenericSensor` are used to expose the readings of just one sensor (that then is published on a YARP port by the `IAnalogWrapper`). The high number of accelerometers present on the robot (~40) make the use of a YARP port for each sensor quite undesirable. 
* If the `IAnalogSensor`/`IGenericSensor` publishes on the same vector (that is then published on a port by the `AnalogWrapper`) readings from multiple sensors, with no way to get at runtime information on **what** is published on the vector . This happens fo the distributed inertial sensors, but also for the classical iCub IMU [2] in which gyroscope, accelerometers, magnetometer and orientation measurements/estimations are published on the same vector, while the structure of the vector is only documented offline. 
* The timestamp is published by a second interface [`IPreciselyTimed`](http://www.yarp.it/classyarp_1_1dev_1_1http://www.yarp.it/classyarp_1_1dev_1_1IPreciselyTimed.html.html) , so it is not possible to publish the different timestamp of the different sensor published on the same port. Furthermore data and timestamp are read (at C++ level) using two different methods, so there is the (small) probability that the user code read the measurement and then the timestamp relative to a different measurement, even if both methods implementation are correctly synchronized using a `Mutex`.  

## Desirable features for the new interfaces 
* Support runtime introspection for identifying which sensor is published in which part of the measurement vector (to match this information with the location of the sensor, as extracted from CAD).
* (Eventually) support publishing a timestamp for each sensor. 
* In general, strictly separate static metadata (sensor location, type) from changing metadata (timestamps) and from data (sensor measurements). 
* Be backward compatible both at C++ level (i.e. be a child of IAnalogSensor) and at the YARP ports level (i.e. do not change the layout of the data in existing (actually used) ports such as IMU, FT (I think instead that changing the layout for the distributed inertial sensor is reasonable. 

## Proposed new interface : IMultipleAnalogSensors
~~~cpp
/*! \file IMultipleAnalogSensors.h interface to multiple analog (i.e. that publish vectors of real numbers) sensors */
namespace yarp {
    namespace dev {
        class IMultipleAnalogSensor;
    }
}

/**
 * @ingroup dev_iface_other
 *
 * A generic interface to sensors (gyro, a/d converters). 
 * Successor to IAnalogSensor, but including support 
 * for publishing multiple sensors with the same object.
 */
class YARP_dev_API yarp::dev::IMultipleAnalogSensor : public yarp::dev::IAnalogSensor
{
public:
    enum AnalogSensorType 
    {
        Accelerometer,
        Gyroscope, 
        Magnetometer,
        SixAxisForceTorque,
        OrientationEuler,
        OrientationQuaternion,
        SkinPatch,
        Unknown
    };

    /**
     *  Structure representing the informations relative to a single sensors.
     */
    struct AnalogSensorInfo 
    { 
         yarp::os::ConstString name;
         AnalogSensorType type;
    };

   /** 
    * Structure representing how a given sensor measurement are published on this interface.
    * The sensor measurements will be published in the elements sensorOffset, sensorOffset+1, ... , sensorOffset+sensorChannels-1 of the vector returned by the read method . 
    */
    struct AnalogSensorOffsetInfo
    { 
         int sensorOffset;
         int sensorChannels;
    };

    virtual ~IMultipleAnalogSensor(){}

    /**
     * Read the total number of the sensors published by this interface.
     * @param[out] nrOfSensors,
     * @return true if the number of sensors was returned, false otherwise.
     */
    virtual bool getNrOfSensors(int &nrOfSensors)=0;

    /**
     * Get the information of a given sensor. 
     * @param[in] sensorIdx a number from 0 to nrOfSensors-1 identifying a given sensor in this interface.  
     * @param[out] info the information relative to the sensor.
     * @return true if all went well, false otherwise.
     */
    virtual bool getSensorInfo(const int sensorIdx, yarp::os::AnalogSensorInfo &info)=0;

    /**
     * Get the offset of a given sensor in the vector returned by the read method. 
     * @param[in] sensorIdx a number from 0 to nrOfSensors-1 identifying a given sensor in this interface. 
     * @param[out] the offset info of the sensor
     * @return true if all went well, false otherwise.
     */
    virtual bool getSensorOffsetInfo(const int sensorIdx, AnalogSensorOffsetInfo &offsetInfo)=0
};
~~~
### Details on the specification 
* The only thing that is meant to change over the lifetime of an an object that exposes the `IMultipleAnalogSensors` interface is the data exposed with the `read` method, and eventually the timestamp. All the other data is assume to be fixed. At the YARP level, this will enable to use the RPC port opened by the `AnalogWrapper` to trasmit only once the `AnalogSensorInfo` and `AnalogSensorOffsetInfo` information. 

* Two separate structure have been  considered to separate the intrinsic information about a sensor (`AnalogSensorInfo`) from the information relative on how the sensor is published by this interface (`AnalogSensorOffsetInfo`) . 

## Missing details 
* In the current proposal, no helper method is used to get the data only relative to a sensor. For user convenience, it could be added a method like: 
~~~
/**
 * Read the sensor measurements for a given sensor.  
 * @param[in] sensorIdx a number from 0 to nrOfSensors-1 identifying a given sensor in this interface. 
 * @param[out] the vector of sensor measurements.
 * @return true if all went well, false otherwise.
 */
virtual int yarp::dev::IMultipleAnalogSensor::readSingleSensor(const int sensorIdx,  yarp::sig::Vector & out) = 0;
~~~

* `yarp::sig::Vector` is currently  used in the `IAnalogSensor` interface, while controlBoard methods tipically used naked `double *` for input/output of vectors. @lornat75 what would you prefer? `double *` are great for compatibility with any kind of vector library, but are awful for error checking (the code can't check if the input/output vector has the correct size.

* Timestamps are currently only published using the [`IPreciselyTimed`](http://www.yarp.it/classyarp_1_1dev_1_1IPreciselyTimed.html) interface, however this interface has the two problems discussed in the section on the current interfaces.
This problem could be solved by having a method that provide a timestamp for each sensor/channel, as in [`IEncodersTimed`](http://www.yarp.it/classyarp_1_1dev_1_1IEncodersTimed.html#ae22a037e958e9955bf1fbc9335270e1e) interface. However I am not sure if this level of complexity is worth, and I don't know (for remotization) if multiple timestamps can be inserted in the envelope of the message trasmitted on the YARP port. @drdanz 

cc @randaz81

## References 
[1] : http://wiki.icub.org/wiki/Distributed_Inertial_sensing
[2] : http://wiki.icub.org/wiki/Inertial_Sensor

traversaro, 2016-05-04:
cc @nunoguedelha @jeljaik

randaz81, 2016-05-04:
I dislike using a single interface to publish data coming for different types of hardware device.
In my view, six axis FT should have a IForceTorque sensor interface, skin should have an ISkinInterface, accelerometers should have an IAccelerometer interface etc.
This comment is not relative to IMultipleAnalogSensor interface but also to the current IAnalogInterface. I always found a very bad solution to expose methods like Calibrate() that have a very specific meaning for a type of sensor and a different one on another type sensor. Using a single interface for multiple sensor types will increase this problem. Btw, ROS uses different types of message for different types of sensors. I find this solution more clean and user-friendly.

marcoaccame, 2016-05-04:
About having one interface dedicated to each hardware device as suggested by @randaz81.

I think it is better. Today it is easier to follow this road because we have different devices for each case: `embObjStrain`, `embObjMais`, `embObjSkin`, `embObjInertials`.

...

About the timestamp of inertial sensors in ETH robots.

**If it is important** that data always have its correct timestamp, I see that the best way is to have an interface similar to the `IEncodersTimed`, which for the single sensor can be:

```
/**
 * Read the sensor measurements for a given sensor plus its timestamp  
 * @param[in] sensorIdx a number from 0 to nrOfSensors-1 identifying a given sensor in this interface. 
 * @param[out] the vector of sensor measurements.
 * @param[timestamp] the timestamp of sensor measurements.
 * @return true if all went well, false otherwise.
 */
virtual int yarp::dev::IMultipleAnalogSensor::readSingleSensorTimed(const int sensorIdx,  yarp::sig::Vector & out, double & timestamp) = 0;
```

The reason is in the following.

In `embObjInertials` we have one function `update()` which is called by the `EthReceiver` periodic thread and imports both data and its timestamp from the UDP socket into the device.

```
bool embObjInertials::update(eOprotID32_t id32, double timestamp, void* rxdata);
```

Now we have the standard `IAnalogSensor::read(yarp::sig::Vector &out)` which publishes a vector containing both data and timestamp in its inside.

Concurrency in write-read is is protected in the two functions `update()` and `read()` using a mutex. The same mutex also guarantees that data and timestamps are always coherent.

If we don't use a single reading function but two different ones are used to retrieve data and timestamp separately, then there is the possibility that `update()` is called in between them, thus creating a de-synch between data and timestamp.  

How often can it happen? Difficult to say. It depends on the priority of the `EthReceiver` thread vs priority of the reading threads, on their rates, on number of inertial sensors attached to the ETH board.

But it can happen.

traversaro, 2016-05-04:
@randaz81 I totally agree that having a YARP device interface / ROS message for each type would be more user friendly. To clarify the rationale behind this proposal is that I was assuming (CAVEAT: this was not validated experimentally) that having single ROS topic / YARP port  (and hence a thread and a single TCP) for each sensor of the robot (for example for the MTB accelerometers there are ~40 sensors in the body of iCub) is not feasible, and hence we need to transmit several sensors readings on the same communication channel (i.e. port/topic).

Given that this sensor are typically heterogeneous (several MTB-like boards mount both a Gyro and an Accelerometer, and some in the future may even mount a complete BBO055 IMU) there is again the need to send multiple sensor readings of different types on the same communication channel. 

Furthermore I fear (but also this is just a pure supposition that having different interfaces for each type of sensors could be more cumbersome to maintain, considering that for each interface you need to maintain (only on the YARP repo) a server device and a client device. This effort is mitigated in ROS by using directly the C++ classes generated from the ROS messages, but in that case you lose the network transparency (i..e. the fact that you can use exactly the same C++ interface both in a device running as part of the yarprobotinterface and on an external module.

barbalberto, 2016-05-04:
I think this interface will not solve basic problems of current interface and in my personal opinion this proposal might be  _even worst_ then what we have now.
Current interface is confusing and just extending it will only increase confusion. If we want to design a new interface it is better to change approach.

> Tipically the IAnalogSensor or IGenericSensor are used to expose the readings of just one sensor (that then is published on a YARP port by the IAnalogWrapper)

Not quite true. This is how we generally use it, but it can export anything since it is a vector of values. The IMU is sending 4 types of data: accelerom, gyro, magnetometer and compass (something you identify with 4 different enums) so you can send different types, same for skinwrapper which uses analogServer to read from a huge number of sensors of the same type (but can be located on different hw boards). 

> If the IAnalogSensor/IGenericSensor publishes on the same vector (that is then published on a port by the AnalogWrapper) readings from multiple sensors, with no way to get at runtime information on what is published on the vector

**This is the real point**, you need to know in advance which data you are receiving from the port and how they are ordered.
The proposal successfully adds this useful information now missing, but I think this can be fixed by adding the methods you propose to the current interface without creating a new one ( just adding a new method is not enough to justify a new interface). 
Anyway I fear this solution may create more confusion and leave a lot of work to the user to create the correct memory to store values and parse them. I foresee a lot of bugs here while using this type of interface.

> Be backward compatible both at C++ level (i.e. be a child of IAnalogSensor) and at the YARP ports level (i.e. do not change the layout of the data in existing (actually used) ports such as IMU, FT (I think instead that changing the layout for the distributed inertial sensor is reasonable.

I do not agree on this point. You can achieve you wishes of compatibility by adding a new method in the current interface as stated before, but a new interface must not have this bond. This will reduce freedom in the design of the interface falling back to something not doing new stuff in the best way and neither be totally back compatible. 

From my experience the best way to go would be create a **brand new** interface, _if needed_, (not inherited from current interfaces) and a new couple of wrapper/client.
Yarp device drivers can implement the new interface while still keeping the current one so that user can still work with previous wrapper if that's enough or decide to use the new wrapper if advanced features are required.
Time has proven that having the IControlMode2 beeing a child of controlmode was a mistake. For example the parent interface cannot be modified niether deprecated/removed because the child is still being used, again imposing constraints on future development.

Another point is that probably each sensor will require specific setting for initialization and run-time configuration, which may be very difficult to collect and fit into a single common interface. So although 'reading' sensors of different type from the same port may be a desirable feature, I think we lose control of them, in the sense that it will be impossible to reach the maximum level of tuning of parameters if we wrap them into a single interface.

### Interface VS data

I think there is a bit of confusion about what's the purpose of an interface and how data shall be sent through a port.
Those are different things and an interface _does not impose_ any particular format on data type, that's the job of the wrapper. Take the stateExt port as an example, the data sent by the port does not match  the `get` methods of any interfaces.

> In general, strictly separate static metadata (sensor location, type) from changing metadata (timestamps) and from data (sensor measurements).

I'm not sure to understand what you mean here. Static metadata that does not change should be handled via rpc like usual. About timestamps  what do you mean by separated? Do you want to use the  envelope or a specific data field? Anyway this is not matter of the interface but a matter of implementation of client/wrapper device.

A self-descriptive type of data is desirable, I'd try an idl generated structure. A positive side effect, will be to allows a human readable printing on terminal.

So to clarify I make some examples of possible interaction betwwen interface(s) and wrapper(s)

- case 1:
Suppose we create different interfaces for each sensor type and a custom client/wrapper for each
iInertialSensor       +  inertialClient/inertialWrapper
iFTSensor           +  FTClient/FTWrapper
iXXXSensor          +  XXXClient/Wrapper

- case 2:
Different interfaces and one client/wrapper that time choose which sensor to use at config  (one type)

- case 3:
Different interfaces and one client/wrapper that at config time choose which sensorS to use (more then one type at the same time)

- case 4:
One generic interface and one client/wrapper to rule them all (one type at time)

- case 5:
One generic interface and one client/wrapper to rule them all (more then one type at time)

Just to give an example, now ROS compatibility can handle a set of precompiled data types and  one of them is choosen by config file (similar to case 2).

Personally I'd go for case 2

traversaro, 2016-05-04:
@barbalberto  
 Good to known that in your experience it easier to just define a new interface instead of modifying/inheriting the old one. The current proposal was explicitly modeled after the new controlmodes interfaces, but if the lesson learned in that case is just to design a brand new interface, I would be happy to just design a new interface.  In that case, the interface can probably be simplified by dropping all the offsetInfo stuff and just provide a `readSingleSensor` method. 

> I'm not sure to understand what you mean here. Static metadata that does not change should be handled via rpc like usual. About timestamps what do you mean by separated? Do you want to use the [...]

Yes, my idea was: Static metadata --> RPC, Dynamic metadata (timestamps) --> envelope, Data ---> port . 
The distinction was not important for the interface, but to ensure backward compatibility with the old device (and its network representation).  

> A self-descriptive type of data is desirable, I'd try an idl generated structure. A positive side effect, will be to allows a human readable printing on terminal. 

I would be happy to go for something similar, but that would mean: 
 * add a streaming port for each device 
 * OR (if the same port is used) break port backward compatibility. 
That seems two things that would be better to avoid, but @lornat75 can definitely give a more informed opinion. 

> Personally I'd go for case 2 

Correct me if I am wrong, but  I fail to see how we can support the use case of the iCub's [Distributed Inertial Sensors](http://wiki.icub.org/wiki/Distributed_Inertial_sensing)  that is the ultimate motivation behind this proposal using any of the case from 1 to 4. For giving you an example, at the moment the arm inertial port publishes 7 accelerometers and 1 gyroscope (and in the future it could host a complete BNO055). Unless we split this measurements in several port/topic (that seems unfeasible to me, but perhaps I am wrong in using this assumption) how can this data be read through a sensor specific interface?

barbalberto, 2016-05-05:
I like you care so much about backward compatibility. I'll share my thought on this topic.

AnalogServer is meant to handle a single sensor. If we want to handle more than one, I'd create a new device (using a new interface) capable of handling more devices. The current analogServer will be a simple use case of the new device and one day it could be deprecated.
This is basically the path followed by the `ControlBoardWrapper2` device. `clientControlBoard` is the firformerst version and implements only a subset of capabilities now available on the new `ControlBoardWrapper2`. Moving from previous to new one is effortless as long as user code uses client/wrapper couple and it does not reads directly from ports, which is discouraged.

Using paired devices you don't need to keep compatibility of data sent through port because the client will handle that for you.

About reading more sensor together I think I've not understood yet _how_ do you think _to group them_ together.

As a personal taste (but it is really personal),  I'd prefer to have one port for each type of sensor, which like:

`/icub/allFT_sensors:o           (no split for parts)`
`/icub/allInertial_sensors:o     (no split for parts)`

and so on. I think there will be not so many ports. The case 2 is enough for this.

If you wanna do the opposite:

`/icub/left_arm/allsensor_no_matter_what`
`/icub/left_leg/allsensor_no_matter_what`
`/icub/right_arm/allsensor_no_matter_what`
`/icub/right_leg/allsensor_no_matter_what`

Then definitely we have to go for case 5. Please consider this will be a very complicated device to implement. Like the one you recently wrote, but with the addictional complication of handling different type of data for each subpart.

traversaro, 2016-07-06:
cc @nunoguedelha

traversaro, 2016-07-06:
I never replied because we had a email thread that postponed the discussion on this, but for completeness I prefer to reply also here, so people not in that email thread can read and build an idea on the issue.

> 
As a personal taste (but it is really personal),  I'd prefer to have one port for each type of sensor, which like: `/icub/allFT_sensors:o           (no split for parts)` `/icub/allInertial_sensors:o     (no split for parts)` and so on. I think there will be not so many ports. The case 2 is enough for this.

The complex part of doing this are mainly two: 
* Each hardware device driver should be connected to each wrapper, as gyro and accelerometers data is usually streamed in the same message. 
* There is not such thing as an "inertial" sensor unfortunately! In particular in our case the sensor types streamed by a group of MTB boards readed by the same MTB is a an arbitrary grouping of gyroscope and accelerometers (not in a even number!) , see http://wiki.icub.org/wiki/Distributed_Inertial_sensing . 

> Then definitely we have to go for case 5. Please consider this will be a very complicated device to implement. Like the one you recently wrote, but with the addictional complication of handling different type of data for each subpart. 

I personally don't think it would be so complex, if implemented correctly and as long as the methods to support are only the one documented in the proposed interfaces. The maintenance effort would be much bigger in having a different interface for each type. 
Clearly the issue is quite different if we want to support methods specific for each type. 

One possible solution is to scale down this proposal to only cover combinations of "inertial" sensors (i.e. accelerometers, gyroscopes, magnetometers, orientation estimations) to efficiently expose just the "inertial" sensors contained in the EMS and MTB boards, to quickly reach our goal to have a nice interface to this sensors.

traversaro, 2017-06-19:
This proposal lost momentum, as the need for more typeful interfaces for "analog"-like sensors emerged. The "distributed inertial" use-case should be addressed by https://github.com/robotology/yarp/issues/1273 , while a proper solution for how to expose the STRAIN2 boards that will measure contain both an F/T, a IMU and two temperature sensors still needs to be discussed. Anyhow, for now I think it is better to close this issue.

closed 2017-06-19 by traversaro