robotology/yarp #749: Python binding build failure

lewisanderson, 2016-05-06:
Hi,

I am encountering a build error with python bindings on yarp. I am using Ubuntu 14.0.4 and followed the build instructions here: http://www.yarp.it/install_yarp_linux.html. I enabled CREATE_PYTHON (along with the other recommended flags).

I tried cloning an older version of this repo (42c6fa4e635cc765c82a2a618010e2e03fd40e53 from roughly a year ago), and it built successfully. 

Therefore I suspect this is a regression caused by some recent change.

[ 99%] Building CXX object bindings/CMakeFiles/_yarp_python.dir/yarpPYTHON_wrap.cxx.o
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Bottle_specialize(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:32469:13: error: ‘class yarp::os::Bottle’ has no member named ‘specialize’
     (arg1)->specialize(arg2);
             ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Bottle_setNested(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:32529:13: error: ‘class yarp::os::Bottle’ has no member named ‘setNested’
     (arg1)->setNested(arg2);
             ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Image_read(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50930:3: error: ‘ConnectionReader’ is not a member of ‘yarp’
   yarp::ConnectionReader *arg2 = 0 ;
   ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50930:3: note: suggested alternatives:
In file included from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/PortReader.h:11:0,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/Portable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/BinPortable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/all.h:19,
                 from /home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:4137:
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:31:29: note:   ‘yarp::os::ConnectionReader’
 class YARP_OS_API yarp::os::ConnectionReader {
                             ^
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:31:29: note:   ‘yarp::os::ConnectionReader’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50930:27: error: ‘arg2’ was not declared in this scope
   yarp::ConnectionReader *arg2 = 0 ;
                           ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:28: error: expected type-specifier
   arg2 = reinterpret_cast< yarp::ConnectionReader * >(argp2);
                            ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:28: error: expected ‘>’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:28: error: expected ‘(’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:28: error: ‘ConnectionReader’ is not a member of ‘yarp’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:28: note: suggested alternatives:
In file included from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/PortReader.h:11:0,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/Portable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/BinPortable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/all.h:19,
                 from /home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:4137:
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:31:29: note:   ‘yarp::os::ConnectionReader’
 class YARP_OS_API yarp::os::ConnectionReader {
                             ^
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionReader.h:31:29: note:   ‘yarp::os::ConnectionReader’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:53: error: expected primary-expression before ‘>’ token
   arg2 = reinterpret_cast< yarp::ConnectionReader * >(argp2);
                                                     ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50952:61: error: expected ‘)’ before ‘;’ token
   arg2 = reinterpret_cast< yarp::ConnectionReader * >(argp2);
                                                             ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Image_write(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50968:3: error: ‘ConnectionWriter’ is not a member of ‘yarp’
   yarp::ConnectionWriter *arg2 = 0 ;
   ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:50968:3: note: suggested alternatives:
In file included from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/PortWriter.h:10:0,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/Portable.h:11,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/BinPortable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/all.h:19,
                 from /home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:4137:
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionWriter.h:30:29: note:   ‘yarp::os::ConnectionWriter’
 class YARP_OS_API yarp::os::ConnectionWriter {
                             ^
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionWriter.h:30:29: note:   ‘yarp::os::ConnectionWriter’
[... 91 lines cut ...]
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/BinPortable.h:10,
                 from /home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/all.h:19,
                 from /home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:4137:
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionWriter.h:30:29: note:   ‘yarp::os::ConnectionWriter’
 class YARP_OS_API yarp::os::ConnectionWriter {
                             ^
/home/lewis/git/yarp/src/libYARP_OS/include/yarp/os/ConnectionWriter.h:30:29: note:   ‘yarp::os::ConnectionWriter’
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:56263:53: error: expected primary-expression before ‘>’ token
   arg2 = reinterpret_cast< yarp::ConnectionWriter * >(argp2);
                                                     ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:56263:61: error: expected ‘)’ before ‘;’ token
   arg2 = reinterpret_cast< yarp::ConnectionWriter * >(argp2);
                                                             ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_IGazeControl_getSaccadesStatus(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:77190:28: error: ‘class yarp::dev::IGazeControl’ has no member named ‘getSaccadesStatus’
     result = (bool)(arg1)->getSaccadesStatus(arg2);

                            ^
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_IGazeControl_setSaccadesStatus(PyObject*, PyObject*)’:
/home/lewis/git/yarp/build/bindings/yarpPYTHON_wrap.cxx:78434:28: error: ‘class yarp::dev::IGazeControl’ has no member named ‘setSaccadesStatus’
     result = (bool)(arg1)->setSaccadesStatus(arg2);
                            ^
make[2]: *** [bindings/CMakeFiles/_yarp_python.dir/yarpPYTHON_wrap.cxx.o] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make: *** [all] Error 2

Tobias-Fischer, 2016-06-15:
Hi,
The following works fine for me:
`cd $YARP_ROOT/bindings`
`mkdir build_python`
`cd build_pyton`
`ccmake ..` and setting `CREATE_PYTHON=ON`
`make`

Hope that helps.

drdanz, 2016-06-17:
I cannot reproduce this on a clean build...
Please remove `yarpPYTHON_wrap.cxx` and try again, sometimes it is not regenerated when it should.

drdanz, 2016-08-04:
No feedback received in 2 months. Closing this

closed 2016-08-04 by drdanz