robotology/yarp #754: Yarp 2.3.63 with java from sources installation problem

vickyvou, 2016-05-11:
Dear all,
i am trying to install the 2.3.63 version (specifically required for a project) from the sources but we also need to compile it with java bindings. The computer has Windows 8. I have installed swig (SWIG_DIR is set to the corresponding directory) and the java jdk 1.8.0_91. To install the dependencies, i have downloaded yarp_2.3.63_v10_x86 (using VS10). (I have followed the same procedure for the latest version of yarp and there it works without a problem in another machine with windows 10). However, when i am trying to compile yarp through VS, i get the following errors. Any idea what it might be?

Thank you very much

1>------ Build started: Project: ZERO_CHECK, Configuration: Debug Win32 ------
2>------ Build started: Project: yarpcar, Configuration: Debug Win32 ------
2>  add_yarpcar_plugins.cpp
2>  yarpcar.vcxproj -> D:\code\yarp\build\lib\Debug\yarpcard.dll
3>------ Build started: Project: jyarp_classes, Configuration: Debug Win32 ------
3>  Building Custom Rule D:/code/yarp/bindings/CMakeLists.txt
3>  CMake does not need to re-run because D:\code\yarp\build\bindings\CMakeFiles\generate.stamp is up-to-date.
3>  Generating ../src/Example1.class, ../src/LoadYarp.class, ../src/YarpImageHelper.class, ../yarp/Bottle.class, ../yarp/BottleCallback.class, ../yarp/BufferedPortBottle.class, ../yarp/BufferedPortImageFloat.class, ../yarp/BufferedPortImageInt.class, ../yarp/BufferedPortImageMono.class, ../yarp/BufferedPortImageRgb.class, ../yarp/BufferedPortImageRgba.class, ../yarp/BufferedPortImageRgbFloat.class, ../yarp/BufferedPortProperty.class, ../yarp/BufferedPortSound.class, ../yarp/BVector.class, ../yarp/CartesianEvent.class, ../yarp/CartesianEventParameters.class, ../yarp/CartesianEventVariables.class, ../yarp/charArray.class, ../yarp/ConnectionReader.class, ../yarp/ConnectionWriter.class, ../yarp/Contact.class, ../yarp/Contactable.class, ../yarp/ContactStyle.class, ../yarp/DeviceDriver.class, ../yarp/DeviceResponder.class, ../yarp/DriverCreator.class, ../yarp/Drivers.class, ../yarp/DummyConnector.class, ../yarp/DVector.class, ../yarp/FlexImage.class, ../yarp/GazeEvent.class, ../yarp/GazeEventParameters.class, ../yarp/GazeEventVariables.class, ../yarp/IAmplifierControl.class, ../yarp/IAmplifierControlRaw.class, ../yarp/IAudioVisualGrabber.class, ../yarp/IAudioVisualStream.class, ../yarp/IAxisInfo.class, ../yarp/ICalibrator.class, ../yarp/ICartesianControl.class, ../yarp/IConfig.class, ../yarp/IControlCalibration.class, ../yarp/IControlCalibration2.class, ../yarp/IControlCalibration2Raw.class, ../yarp/IControlCalibrationRaw.class, ../yarp/IControlDebug.class, ../yarp/IControlLimits.class, ../yarp/IControlLimitsRaw.class, ../yarp/IControlMode.class, ../yarp/IControlModeRaw.class, ../yarp/IEncoders.class, ../yarp/IEncodersRaw.class, ../yarp/IFrameGrabber.class, ../yarp/IFrameGrabberControls.class, ../yarp/IFrameGrabberControlsDC1394.class, ../yarp/IFrameGrabberImage.class, ../yarp/IFrameGrabberImageRaw.class, ../yarp/IFrameGrabberRgb.class, ../yarp/IFrameWriterAudioVisual.class, ../yarp/IFrameWriterImage.class, ../yarp/IGazeControl.class, ../yarp/IImpedanceControl.class, ../yarp/IImpedanceControlRaw.class, ../yarp/Image.class, ../yarp/ImageFloat.class, ../yarp/ImageInt.class, ../yarp/ImageMono.class, ../yarp/ImageRgb.class, ../yarp/ImageRgba.class, ../yarp/ImageRgbFloat.class, ../yarp/IOpenLoopControl.class, ../yarp/IOpenLoopControlRaw.class, ../yarp/IPidControl.class, ../yarp/IPidControlRaw.class, ../yarp/IPositionControl.class, ../yarp/IPositionControlRaw.class, ../yarp/ITorqueControl.class, ../yarp/ITorqueControlRaw.class, ../yarp/IVelocityControl.class, ../yarp/IVelocityControlRaw.class, ../yarp/NameStore.class, ../yarp/Network.class, ../yarp/NetworkBase.class, ../yarp/Pid.class, ../yarp/PidVector.class, ../yarp/PixelBgr.class, ../yarp/PixelBgra.class, ../yarp/PixelHsv.class, ../yarp/PixelHsvFloat.class, ../yarp/PixelRgb.class, ../yarp/PixelRgba.class, ../yarp/PixelRgbFloat.class, ../yarp/PixelRgbInt.class, ../yarp/PixelRgbSigned.class, ../yarp/PolyDriver.class, ../yarp/Port.class, ../yarp/Portable.class, ../yarp/PortablePairBase.class, ../yarp/PortReader.class, ../yarp/PortReaderBufferBase.class, ../yarp/PortReaderBufferBaseCreator.class, ../yarp/PortReaderCreator.class, ../yarp/PortReport.class, ../yarp/PortWriter.class, ../yarp/PortWriterBufferBase.class, ../yarp/PortWriterBufferManager.class, ../yarp/PortWriterWrapper.class, ../yarp/Property.class, ../yarp/PropertyCallback.class, ../yarp/Random.class, ../yarp/RateThread.class, ../yarp/RateThreadWrapper.class, ../yarp/ResourceFinder.class, ../yarp/RFModule.class, ../yarp/RpcClient.class, ../yarp/RpcServer.class, ../yarp/Searchable.class, ../yarp/SearchMonitor.class, ../yarp/SearchReport.class, ../yarp/Semaphore.class, ../yarp/Sound.class, ../yarp/Stamp.class, ../yarp/Stamped.class, ../yarp/StubDriverCreator.class, ../yarp/SVector.class, ../yarp/SWIGTYPE_p_bool.class, ../yarp/SWIGTYPE_p_Bytes.class, ../yarp/SWIGTYPE_p_Clock.class, ../yarp/SWIGTYPE_p_double.class, ../yarp/SWIGTYPE_p_float.class, ../yarp/SWIGTYPE_p_f_p_q_const__char_v_______int.class, ../yarp/SWIGTYPE_p_ICalibrator.class, ../yarp/SWIGTYPE_p_int.class, ../yarp/SWIGTYPE_p_NameStore.class, ../yarp/SWIGTYPE_p_NestedContact.class, ../yarp/SWIGTYPE_p_Pid.class, ../yarp/SWIGTYPE_p_p_char.class, ../yarp/SWIGTYPE_p_p_yarp__os__Value.class, ../yarp/SWIGTYPE_p_ResourceFinderOptions.class, ../yarp/SWIGTYPE_p_Runnable.class, ../yarp/SWIGTYPE_p_Searchable.class, ../yarp/SWIGTYPE_p_size_t.class, ../yarp/SWIGTYPE_p_std__string.class, ../yarp/SWIGTYPE_p_Type.class, ../yarp/SWIGTYPE_p_unsigned_char.class, ../yarp/SWIGTYPE_p_unsigned_int.class, ../yarp/SWIGTYPE_p_void.class, ../yarp/SWIGTYPE_p_yarp__os__InputStream.class, ../yarp/SWIGTYPE_p_yarp__os__NestedContact.class, ../yarp/SWIGTYPE_p_yarp__os__OutputStream.class, ../yarp/SWIGTYPE_p_yarp__os__PortInfo.class, ../yarp/SWIGTYPE_p_yarp__os__SizedWriter.class, ../yarp/SWIGTYPE_p_yarp__os__TypedReaderCallbackT_yarp__sig__Sound_t.class, ../yarp/SWIGTYPE_p_yarp__sig__ImageOfT_yarp__sig__PixelBgr_t.class, ../yarp/SWIGTYPE_p_yarp__sig__Matrix.class, ../yarp/Things.class, ../yarp/Thread.class, ../yarp/Time.class, ../yarp/TypedReaderBottle.class, ../yarp/TypedReaderCallbackImageFloat.class, ../yarp/TypedReaderCallbackImageInt.class, ../yarp/TypedReaderCallbackImageMono.class, ../yarp/TypedReaderCallbackImageRgb.class, ../yarp/TypedReaderCallbackImageRgba.class, ../yarp/TypedReaderCallbackImageRgbFloat.class, ../yarp/TypedReaderImageFloat.class, ../yarp/TypedReaderImageInt.class, ../yarp/TypedReaderImageMono.class, ../yarp/TypedReaderImageRgb.class, ../yarp/TypedReaderImageRgba.class, ../yarp/TypedReaderImageRgbFloat.class, ../yarp/TypedReaderProperty.class, ../yarp/TypedReaderSound.class, ../yarp/UnbufferedContactable.class, ../yarp/Value.class, ../yarp/Vector.class, ../yarp/VectorBase.class, ../yarp/Vocab.class, ../yarp/yarp.class, ../yarp/yarpConstants.class, ../yarp/yarpJNI.class, ../yarp/YarpVocabPixelTypesEnum.class
3>CUSTOMBUILD : warning : [options] bootstrap class path not set in conjunction with -source 1.3
3>CUSTOMBUILD : warning : [options] source value 1.3 is obsolete and will be removed in a future release
3>CUSTOMBUILD : warning : [options] target value 1.3 is obsolete and will be removed in a future release
3>CUSTOMBUILD : warning : [options] To suppress warnings about obsolete options, use -Xlint:-options.
3>  yarp\SWIGTYPE_p_bool.java:14: error: annotations are not supported in -source 1.3
3>    protected SWIGTYPE_p_bool(long cPtr, @SuppressWarnings("unused") boolean futureUse) {
3>                                          ^
3>    (use -source 5 or higher to enable annotations)
3>  yarp\SWIGTYPE_p_Bytes.java:14: error: annotations are not supported in -source 1.3
3>    protected SWIGTYPE_p_Bytes(long cPtr, @SuppressWarnings("unused") boolean futureUse) {
3>                                           ^
3>    (use -source 5 or higher to enable annotations)
3>  yarp\SWIGTYPE_p_Clock.java:14: error: annotations are not supported in -source 1.3
3>    protected SWIGTYPE_p_Clock(long cPtr, @SuppressWarnings("unused") boolean futureUse) {
3>                                           ^
3>    (use -source 5 or higher to enable annotations)
3>  yarp\SWIGTYPE_p_double.java:14: error: annotations are not supported in -source 1.3
3>    protected SWIGTYPE_p_double(long cPtr, @SuppressWarnings("unused") boolean futureUse) {
3>                                            ^
[... cut ...]
3>    protected SWIGTYPE_p_yarp__sig__Matrix(long cPtr, @SuppressWarnings("unused") boolean futureUse) {
3>                                                       ^
3>    (use -source 5 or higher to enable annotations)
3>  30 errors
3>  4 warnings
3>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\Microsoft.CppCommon.targets(151,5): error MSB6006: "cmd.exe" exited with code 1.
4>------ Build started: Project: ALL_BUILD, Configuration: Debug Win32 ------
========== Build: 3 succeeded, 1 failed, 32 up-to-date, 0 skipped ==========

drdanz, 2016-05-12:
This is something that is probably related to some recent version of swig and that IIRC was fixed in master since july.
The problem is the `-source 1.3` in java command line.
The easiest way to fix it should be to edit the file bindings/CMakeLists.txt and find out the part where it prepares for java bindings and replace `1.3` with `1.6`.

Let us know if this works!

vickyvou, 2016-05-12:
Yes! Thank you it now compiled without a problem, however now i get an error (while trying to execute yarp server) that it is missing the YARP_initd.dll and i should try to reinstall the program to correct this problem. Any ideas?

Once again,
thank you very much!

drdanz, 2016-05-12:
Perhaps the location of YARP_initd.dll is not in your `PATH` environment variable? It should be set by the installer, but I think that this can be disabled

vickyvou, 2016-05-12:
i was about to update. All works, i had forgotten to add the Lib! thank you!

drdanz, 2016-05-13:
Thanks for the update! Closing the issue, feel free to reopen it if you still have issues

closed 2016-05-13 by drdanz