robotology/yarp #758: [OSX] yarpmanager++ CMake generation error

claudiofantacci, 2016-05-16:
Trying to generate makefiles after enabling YARPMANAGER_PP I get the following error:

```
CMake Error at conf/YarpOSXUtilities.cmake:74 (add_executable):
   Cannot find source file:

     Resources/AppIcon.icns

   Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
   .hxx .in .txx
 Call Stack (most recent call first):
   src/yarpmanager++/CMakeLists.txt:140 (yarp_osx_duplicate_and_add_bundle)
```

It should be something related to `__APPLE__` systems only while calling `yarp_osx_duplicate_and_add_bundle`.

The problem is related to the AppIcon.icns path, it should be **src-manager/Resources/AppIcon.icns**.
Maybe @apaikan can look into it :-)

apaikan, 2016-05-16:
@claudiofantacci :
It seems there is a missing resource file which is probably discarded during the original commit. 

Would you please comment `yarp_osx_duplicate_and_add_bundle` and let me know the rests is fine?

Ill copy the icon file and commit it. Apparently you are the only one who investigates yarpmanager++ on Mac ;)

claudiofantacci, 2016-05-16:
I added src-manager in front of "Resource/AppIcon.icns" in yarp_osx_duplicate_and_add_bundle function call and the compilation process went out smoothly.

I think the problem relies only on the missing prefix folder in the fanction call, the icon files is actually present in the repository.

PR #778 (2016-05-27): Fix the cmake path to MacOS yarpmanager++ icon
It was missing part of the relative path from the yarpmanager++ source folder to the AppIcon.icns location.
The compilation process was failing on MacOSX systems if CREATE_YARPMANAGER_PP was set to ON.

Fixes #758
<a href='#crh-start'></a><a href='#crh-data-%7B%7D'></a>
[image]
[image]&nbsp;[image]
<a href='#crh-end'></a>
commit 91d76854 (2016-05-27): Fix the cmake path to MacOS yarpmanager++ icon

It was missing part of the relative path from the yarpmanager++ source folder to the AppIcon.icns location.
The compilation process was failing on MacOSX systems if CREATE_YARPMANAGER_PP was set to ON.

Fixes #758
closed 2016-05-27 by drdanz