robotology/yarp #767: reading ROS topic from Yarp

tkelestemur, 2016-05-24:
Hi, I am trying to read a ROS topic that is publishing geometry_msgs/Pose type message.

I am running these in order:
`roscore`
`yarp server --ros` 
`yarpidl_rosmsg --name /typ@/yarpidl`
`rosrun icub_ros joint_transformer`  this node is publishing to topic called`icub/jointPose` from a ros bag file.  
`rosPoseListener` -> yarp code that is trying to read ros topic. (the code is given below)

when I run `yarp read /icub/jointPose@/yarp/listener` I am getting what is publishing to topic but I cannot get the data with this code

```
#include "geometry_msgs_Pose.h"
#include <stdio.h>
#include <yarp/os/all.h>

using namespace yarp::os;
#include <iostream>
using namespace std;

int main(int argc, char *argv[]) {
    Network yarp;
    Node node("/icubSim/poseListener");
    Port port;
    port.setReadOnly();
    if (!port.open("icub/jointPose")) {
        fprintf(stderr,"Failed to open port\n");
        return 1;
    }
    // cout << "Port oppend!" <<endl;
    // yarp.connect("icub/jointPose", "/icubSim/poseListener");
    while (true) {
        geometry_msgs_Pose jointPose;
        // Bottle msg;
        if (!port.read(jointPose)) {
            fprintf(stderr,"Failed to read msg\n");
            continue;
        }

        cout << "x position: " << jointPose.position.x  <<endl;
        // printf("Got [%d]\n",jointPose.position.x);

    }

    return 0;
}
```
When I run the I get this: `x position: 4.24399e-314` again and againg which doesn't make sense.

Any ideas?
I am using latest version of Yarp with ROS Kinetic.

barbalberto, 2016-05-24:
Please use the yarp:.os::Subscriber instead of a plain yarp::port: http://www.yarp.it/classyarp_1_1os_1_1Subscriber.html

It'll take care of lower details for you.

```
Node node("/icubSim/poseListener");
yarp::os::Subscriber<geometry_msgs_Pose>  mySubscriber;  // this must be done after the node has been created
mySubscriber.topic(""myTopicName");
geometry_msgs_Pose*                     rosInData  = mySubscriber.read( );
```

Hope this helps, if you still have problem feel free to ask.

lornat75, 2016-05-25:
I would also give a check that you are compiling generate the type geometry_msgs_Pose correctly using yarpidl_rosmsg.

barbalberto, 2016-05-25:
Just noticed, topic should start with '/' character, otherwise I guess yarp will let you open the port but maybe the connection with ROS may fail

tkelestemur, 2016-05-25:
@barbalberto Thank you very much! It now works without problems. The best part is I don't need to run `yarpidl_rosmsg --name /typ@/yarpidl` anymore. I guess these classes weren't documented because I couldn't see anything about them in ROS-Yarp [tutorial](http://www.yarp.it/yarp_with_ros.html).

FYI, it doesn't matter whether the topic name starts with '/' or not.

barbalberto, 2016-05-25:
Happy to know it works. You are right, we have to update the documentation... sorry about that.
Thanks for the info about the '/'.

lornat75, 2016-05-25:
Hi @tkelestemur the yarpidl_rosmsg --name /typ@/yarpidl is used to dynamically interpret ROS messages (i.e. from `yarp read`). In your case you import a ROS type in YARP so there is no need for the dynamic conversion.

The publisher subscriber functions are mentioned here:

http://www.yarp.it/rostypes_tutorial_continued.html

Maybe we could make this more evident.

However the code you wrote is legitimate so it would be worth checking why it does not work (homework for us).

randaz81, 2016-05-25:
btw, just for curiosity, what is the purpose of joint_transformer/icub_ros/jointPose ? I'm asking just because the name of your module resembles a bridge... are you aware that robotStatepublisher already broadcasts the pose of the icub frames (without the need of implementing additional stuff)?

tkelestemur, 2016-05-25:
@lornat75 thank you for the info. I haven't seen that tutorial. Now I am trying to use this with RateThread class but I get segmentation error. You may help this one too. The is below.

```
// -*- mode:C++; tab-width:4; c-basic-offset:4; indent-tabs-mode:nil -*-

#include <yarp/os/all.h>
#include <yarp/dev/ControlBoardInterfaces.h>
#include <yarp/dev/CartesianControl.h>
#include <yarp/dev/PolyDriver.h>
#include <yarp/sig/Vector.h>

#include "geometry_msgs_Pose.h"

#include <stdio.h>
#include <string>
#include <iostream>

using namespace yarp::os;
using namespace yarp::dev;
using namespace yarp::sig;

using namespace std;

class ControlThread: public RateThread
{

    PolyDriver dd;
    ICartesianControl *icart;

    Vector cRot, cOrt; // current rotation and orientation of robot's joint
    Vector dRot, dOrt; // desired rotation and orientation of robot's joint

    yarp::os::Subscriber<geometry_msgs_Pose> poseSub;
    // geometry_msgs_Pose* jointPose;

public:
    ControlThread(int period):RateThread(period){}

    bool threadInit()
    {
        //initialize here variables
        printf("ControlThread:starting\n");

        Property options("(device cartesiancontrollerclient)");
        options.put("remote","/icubSim/cartesianController/right_arm");
        options.put("local","/cartesian_client/right_arm");

        dd.open(options);

        if (dd.isValid()) {
           dd.view(icart);

           if (!icart){
              // std::cout << "driver is available!" << std::endl; // debug
              return false;
           }

        }

        // get the torso dofs
        Vector newDof, curDof;
        icart->getDOF(curDof);
        newDof=curDof;

        // enable the torso yaw and pitch
        // disable the torso roll
        newDof[0]=0;
        newDof[1]=0;
        newDof[2]=0;

        // impose some restriction on the torso pitch
        limitTorsoPitch();

        // send the request for dofs reconfiguration
        icart->setDOF(newDof,curDof);

        icart->setTrajTime(1.0);

        dRot.resize(3);
        dOrt.resize(4);

        // ROS initialization
        Node node("/icubSim/poseSub");
        poseSub.topic("/icub/jointPose");

        return true;
    }

    void threadRelease()
    {
        printf("ControlThread:stopping the robot\n");
        icart->stopControl();
        dd.close();

        printf("Done, goodbye from ControlThread\n");
    }

    void run()
    {

      printICubPoseStatus();
      // getHumanJointPose();

      geometry_msgs_Pose *jointPose = poseSub.read();
      cout << "x position: " << jointPose->position.x  <<endl; // debug
      dOrt = cOrt;
      // dRot[0] = -0.1; // desired x coordinate
      // dRot[1] = 0.1; // desired y coordinate
      // dRot[2] = 0.1; // desired z coordinate

      // icart->goToPose(dRot,dOrt);
      // icart->goToPoseSync(X_desired,O_desired); // send request and wait for reply
      // icart->waitMotionDone(0.04); // wait until the motion is done and ping at each 0.04 seconds
    }

    // void getHumanJointPose() {
    //   geometry_msgs_Pose* jointPose  = poseSub.read();
    //   // dRot[0] = -jointPose->position.z;
    //   // dRot[1] = -jointPose->position.x;
    //   // dRot[2] = jointPose->position.y;
    //   // cout << "x position: " << jointPose->position.x  <<endl; // debug
    //   std::cout << "debug" << std::endl;
    // }

    void limitTorsoPitch()
    {
        int axis=0; // pitch joint
        double min, max;
        int MAX_TORSO_PITCH = 30.0;
        // sometimes it may be helpful to reduce
        // the range of variability of the joints;
        // for example here we don't want the torso
        // to lean out more than 30 degrees forward

        // we keep the lower limit
        icart->getLimits(axis,&min,&max);
        icart->setLimits(axis,min,MAX_TORSO_PITCH);
    }

    void printICubPoseStatus(){
      icart->getPose(cRot, cOrt);
      cout << "robot palm rotation (xyz)[m] = " << cRot.toString().c_str() << endl;
      // cout << "Current Orientation (O)[m] = " << O_current.toString().c_str() << endl;

    }

};

int main(int argc, char *argv[])
{
    Network yarp;

    if (!yarp.checkNetwork())
    {
        printf("No yarp network, quitting\n");
        return 1;
    }

    ControlThread ctrlThread(400); //period is 40ms

    ctrlThread.start();
    int RUN_TIME = 4; // seconds
    bool done=false;
    double startTime=Time::now();
    while(!done)
    {
        if ((Time::now()-startTime)>RUN_TIME)
            done=true;
    }

    ctrlThread.stop();

    return 0;
}

```

tkelestemur, 2016-05-25:
@randaz81 I am using that to get joint rotation and orientation coming from a ros topic publishe by kinect node.

lornat75, 2016-05-25:
@tkelestemur can you post this into a new issue?

tkelestemur, 2016-05-25:
@lornat75 just oppend a new issue.

closed 2016-05-26 by lornat75