robotology/yarp #770: using yarp::os::Subscriber with RateThread

tkelestemur, 2016-05-25:
I am trying to use `yarp::os::Subscriber` with RateThread to control robots right arm. 
The code is in this [link](https://github.com/tkelestemur/icub_ros/blob/master/iCubSim/jointFollower.cpp). When I run, I get segmentation fault. I guess I need to change something with pointer declarations.

barbalberto, 2016-05-25:
Have you tried to run it with gdb to see _where_ the segfault happens?

tkelestemur, 2016-05-25:
@barbalberto I get segmentation fault with this line `cout << "x position: " << jointPose->position.x  <<endl; // debug` when I comment it out I don't get segmentation fault.

barbalberto, 2016-05-25:
From documentation here:
http://www.yarp.it/classyarp_1_1os_1_1Subscriber.html#ad7dc6cefd66c56d14e1cf1c42036ac4e

``` Read a message from the port.  Waits by default. May return NULL if the port status has changed.```

It says read function can return NULL, so it is good practise to check if pointer is valid before using it. I guess in your case it is not valid.

barbalberto, 2016-05-26:
A bit more of insights of what may be happening.
With yarp only, you can decide the timing of connections, like:
```
port.open("/local");
ret = Network::connect("/local","/remote");
if(ret)
   port.read();  --> at this time we know the connection is ok
```

When connecting yarp to ROS, is the roscore in charge of making connections, therefore if we do

```
subscriber.topic("/topic");
subscriber.read();
```
maybe the connection is not established yet, so we can get a `NULL` pointer as a result.

Also, @ line 84 the node is instantiated inside the thread_init function. It means that as soon as the function exits the node is destroyed. 
It is better if its scope is bigger then the scope of the topic.

tkelestemur, 2016-05-26:
@barbalberto I guess you are right. I have tried putting all ros-related declerations in thread run function like this:
```
    void run()
    {

      Node node("/icubSim/poseSub");
      yarp::os::Subscriber<geometry_msgs_Pose> poseSub;
      poseSub.topic("/icub/jointPose");

      geometry_msgs_Pose *jointPose = poseSub.read();
      cout << "x position: " << jointPose->position.x  <<endl; // debug

    }
```

In this way, it works but every time run function called by thread it recreates the node and topic which is not convenient. I will go without RateThread but we should think a way to use ros subscriber and publisher with RateThread class.

barbalberto, 2016-05-26:
You can declare the a Node pointer as a member of `ControlThread` class, then you can do the `new` in the thread init. This way the scope is preserved and you can initialize the node when you think it is right.

```

class ControlThread: public RateThread
{

    PolyDriver dd;
    ...
    yarp::os::Node node;

    public:
    bool threadInit()
    {
        ...
        // ROS initialization
        node = new yarp::os::Node("/icubSim/poseSub");
        return true;
    }

```

Just remember to delete the node while quitting.

tkelestemur, 2016-05-26:
I've tried but I get conversion error:
```
/home/tarik/ros_ws/src/icub_ros/iCubSim/src/jointFollower.cpp:84:14: error: no match for ‘operator=’ (operand types are ‘yarp::os::Node’ and ‘yarp::os::Node*’)
         node = new yarp::os::Node("/icubSim/poseSub");
              ^
In file included from /usr/local/include/yarp/os/all.h:30:0,
                 from /home/tarik/ros_ws/src/icub_ros/iCubSim/src/jointFollower.cpp:3:
/usr/local/include/yarp/os/Node.h:18:29: note: candidate: yarp::os::Node& yarp::os::Node::operator=(const yarp::os::Node&)
 class YARP_OS_API yarp::os::Node : public Contactables {
                             ^
/usr/local/include/yarp/os/Node.h:18:29: note:   no known conversion for argument 1 from ‘yarp::os::Node*’ to ‘const yarp::os::Node&’

```

barbalberto, 2016-05-26:
It looks your variable ```node``` is not a pointer.
Are you using
```yarp::os::Node *node```   ?
or
```yarp::os::Node node```   ?

tkelestemur, 2016-05-26:
I fixed it to `yarp::os::Node *node` it works now but still it doesn't solve my problem. Topic recreated every time when run function is called. So I also put these declarations to ControlThread class:

```
yarp::os::Subscriber<geometry_msgs_Pose> poseSub;
geometry_msgs_Pose *jointPose;
```

and read topic messeges like this:

```
jointPose = poseSub.read();
cout << "x position: " << jointPose->position.x  <<endl; // debug
```
Final version code is [here](https://github.com/tkelestemur/icub_ros/blob/master/iCubSim/src/jointFollower.cpp).

@barbalberto It works without issues now. Thank you for your great support! 

However, I'm little lost most probably because of my lack of C++ knowledge :)

barbalberto, 2016-05-26:
Glad it works now. Yes this whole issue was about C++ code and not actually yarp-related.
I'd advise you to strenghten you c++ knowledge before going deeper into more complex application.

closed 2016-05-26 by barbalberto