robotology/yarp #781: yarp::os::SystemInfo : missing implementation for Mac OS

apaikan, 2016-05-30:
The class `yarp::os::SystemInfo` does not provide any system information for Mac Os. 

**Source of the problem:**
pre-processor macro `__linux__` has not been defined on Mac OS. Probably adding `__APPLE__` to all the relevant functions will solve  the problem! 

CC:  @claudiofantacci  (thanks!)

francesco-romano, 2016-06-01:
I started adding some info for memory and Cpu.
It is just a "skeleton"...

claudiofantacci, 2016-06-01:
Cool!
I didn't think it would have differ so much from the linux implementation.
We might also want to meet up to coordinate ourself to make the job done. Let me know.

francesco-romano, 2016-06-01:
Well.. I'm not sure that in linux we cannot do the same (more or less), as `sysctl` exists also on linux I think.
The difference here is that in linux we are parsing some files, while here we are using system calls.

francesco-romano, 2016-06-06:
@apaikan this is the branch: https://github.com/robotology/yarp/tree/fix/osx_system_info
It is based on `master`, so when we finish I think we can merge it in master (as bug fix) and devel

francesco-romano, 2016-06-13:
@apaikan @claudiofantacci 
I've just implemented some other methods.

If you want take a look at it..
BTW: description of the structures should really be added. Some structure are not easy to understand (e.g. CpuLoad)

claudiofantacci, 2016-06-13:
I didn't have the chance to work on it. Sorry about that. Seems fine to me.
Maybe @apaikan can do some test with his latest modules trying to parse OSX-related info.

francesco-romano, 2016-06-14:
Ok. I open a PR (https://github.com/robotology/yarp/pull/799) to track this issue, otherwise it gets lost.

francesco-romano, 2016-08-05:
Given that the PR has been merged I'd close this issue.
We can open it again if something is missing.

closed 2016-08-05 by francesco-romano