robotology/yarp #782: timestamp/envelope problems in portmonitor (only in recv mode) ?

randaz81, 2016-05-31:
I'm connecting two ports using a portmonitor plugin.
In **send** mode the timestamp/envelope is ok.
In **recv** mode the timestamp/envelope seems to be missing (e.g. in following snippet of code, taken from **camCalibWithPose** time is always=0). 

```c++
void HeadEncoderPort::onRead(yarp::os::Bottle &h_encs)
{
    yarp::os::Stamp s;
    this->getEnvelope(s);
    double time = s.getTime();
    if(time !=0)
   {
        _prtImgIn->setHeadEncoders(time, h_encs);
    }
}
```

However the command **yarp read** able to display the envelope! @drdanz  @apaikan

drdanz, 2016-05-31:
I had this same problem some time ago, and I think I tested this with @apaikan, but with yarp read it disappeared, so I never went further with the testing.
At this point I believe that the issue might be in the type of port used, since `yarp read` uses a `yarp::os::PortReader`, and `camCalibWithPose` uses `yarp::os::BufferedPort`

drdanz, 2016-06-20:
See also #729

drdanz, 2016-09-22:
Duplicate of #909

drdanz, 2016-09-26:
Fixed in `devel`

commit f7290e5c (2016-09-26): PortMonitor,Bayer: Set parent connection reader in modifyIncomingData

Fixes #909
Fixes #782
Fixes #729
closed 2016-09-26 by drdanz