robotology/yarp #786: Unit Quaternion handling functions in yarp::math

traversaro, 2016-05-31:
Related issues: https://github.com/robotology/yarp/issues/635 , https://github.com/robotology/idyntree/issues/168 .

@francesco-romano recently implemented functions for converting between rotation matrices and unit quaternions in iDynTree, and we added some tests to verify that our implementation was compatible (modulo the real/imaginary serialization issues) with the libraries iDynTree needs to interact with, namely YARP, Eigen and KDL. 

We wrote some tests, and apparently iDynTree/KDL/Eigen implementations seems to be coherent, while in YARP for a given Rotation matrix the `dcm2quat` method returns the inverse (i.e. the conjugate) of the quaternion returned by the other libraries. 
However the YARP implementation seems to be self consistent, i.e. R == quat2dcm(dcm2quat(R)), so it seems to be a problem of the assumed quaternion convention. Unfortunately there is no "natural" way to map a Rotation matrix to a Unit Quaternion, and multiple conventions are equally valid as long as a the action of a Quaternion on a 3D vector is coherently defined. However given that  the quaternion methods were added for interoperability (see https://github.com/robotology/yarp/issues/635) it is important to properly define the convention used. 

@pattacini in https://github.com/robotology/yarp/issues/635 you referred to MATLAB as the source for those function.. in particular which functions did you use as a reference?

pattacini, 2016-05-31:
Hi @traversaro 

Look for example at [**`n=quat2dcm(q)`**](http://it.mathworks.com/help/aerotbx/ug/quat2dcm.html), where it's said:

> Additionally, q has its scalar number as the first column

traversaro, 2016-06-01:
Yes, I think that the key part is:
> The direction cosine matrix performs the coordinate transformation of a vector in inertial axes to a vector in body axes. 

In fact, Matlab itself provides a [`quat2rotm`](http://it.mathworks.com/help/robotics/ref/quat2rotm.html) that is coherent with the "robotics" convention: 

~~~
dcmAero = quat2dcm([1 0 1 0])

dcmAero =

         0         0   -1.0000
         0    1.0000         0
    1.0000         0         0

dcmRobotics = quat2rotm([1 0 1 0])

dcmRobotics =

    0.0000         0    1.0000
         0    1.0000         0
   -1.0000         0    0.0000
~~~

I just checked, and the `quat2rotm` definition is coherent with the definition of quaternions associated with a rotation matrix as given in the Springer Handbook of Robotics and in A Mathematical Introduction to Robotic Manipulation (both also use the real/imaginary serialization).

traversaro, 2016-06-02:
For reference, I did not tested the tf2 quaternion --> rotation functions, but they appear to have been copied from the Bullet library: 
https://github.com/ros/geometry2/blob/c0f1905bfca22695a387c59ceffdd8c8a99498af/tf2/include/tf2/LinearMath/Matrix3x3.h#L146
However I think they are compatible with KDL function, because the Python version of tf2 uses KDL bindings: 
https://github.com/ros/geometry2/blob/c0f1905bfca22695a387c59ceffdd8c8a99498af/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py#L50 .

I think that if we want to use this function for ROS compatibility we need to use implement the `quat2rotm` function (to maintain the nice C++/Matlab parallelism).

traversaro, 2016-06-09:
cc @aerydna

traversaro, 2016-07-20:
Just discussed f2f with @randaz81 : implementeting the `rotm2quat` etc etc function would solve the problem, but will leave the YARP math API in a bit on inconsistent state, with some function using the `dcm` term and some other using `rotm` . 
To clarify things, the idea is to adopt the relevant function names from the Matlab Robotics System Toolbox : http://it.mathworks.com/help/robotics/coordinate-system-transformations.html .
In most cases, this will just mean deprecating the old function and rename it to one with the proper names.

pattacini, 2016-07-20:
Hold on... I was not present in the discussion :smile:

pattacini, 2016-07-22:
Indeed, I'd go for `dcm2quat` rather than `rotm2quat`, sticking to its implementation, obviously. This is to wave a flag in favor of those who want to avoid confusion between functions termed with `dcm` and functions using `rotm` :smile: 

The reason is that the `dcm2axis` and `axis2dcm` are available since almost 8 years in our code and for this reason their use has become very spread in many of our repos and documentation as well (i.e. wikis).

I really reckon it'd be an unnecessary hassle to revise the whole code just for changing a function name, while we could more profitably introduce `dcm2quat` and `quat2dcm` instead.

As side note, MATLAB convention is also not unique throughout their toolboxes Aerospace (released years ago) vs. Robotics (more recent): so you can find `dcm2quat` as well as `rotm2quat`. This is to say that there's really no strong motivation to drop our names: ours remains a fair choice.

pattacini, 2016-07-22:
So the problem is that the `dcm2quat` outputs a different outcome from `rotm2quat`. In this case I'd suggest to implement both or just stick to `rotm2quat`, the service you really need.

There's no need to change `dcm2axis` then.

drdanz, 2016-11-10:
I think this should be fixed by 0d6826e, @traversaro @pattacini can you confirm this?

randaz81, 2016-11-16:
The commit mentioned above has been integrated in PR #959 which introduces the class `yarp::math::Quaternion`. The two functions `rotationmatrix2quaternion()` and `quaternion2roattionmatrix()` are now class methods ( respectively: `Quaternion::fromRotationMatrix()` and `Quaternion::toRotationMatrix()`)

randaz81, 2017-01-10:
fixed by class `yarp::math::Quaternion`, closing.

closed 2017-01-10 by randaz81