robotology/yarp #789: [yarplogger] Not registering log messages

rlober, 2016-06-07:
cc: @jeljaik, @ahoarau

First time I have encountered this with yarplogger.

I am running a `yarpserver` on our cluster and the log messages from `yarprobotinterface` are not being registered by `yarplogger`. `yarplogger` is registered on the network:

```
icub@icub-desktop:/usr/local/src/robot/source/icub-main/build-pc104/share/iCub/robots/iCubParis02$ yarp name list
registration name /iCubParis02 ip 10.0.0.1 port 10000 type tcp
registration name /icub-desktop ip 10.0.0.147 port 10002 type tcp
registration name /pc104 ip 10.0.0.2 port 10004 type tcp
registration name /yarplogger ip 10.0.0.147 port 10059 type tcp
registration name fallback ip 224.2.1.1 port 10000 type mcast
*** end of message
```

The logger is started: 
```
[INFO]  [07/06/2016 13:41:07] Logger started
```

But no registered processes are appearing. 

I made a quick test to check if it is truly a logger issue:

```cpp
#include <iostream>
#include <yarp/os/Log.h>
#include <yarp/os/LogStream.h>

int main(int argc, char const *argv[]) {

  yarp::os::Log yLog;

  yLog.info() << "Information";
  yLog.warning() << "Warning";
  yLog.error() << "Error";

  return 0;
```

**output**
```
icub@icub-desktop:~/Desktop/yLogTest/build$ ./yarpy 
[INFO]Information 
[WARNING]Warning 
[ERROR]Error 
icub@icub-desktop:~/Desktop/yLogTest/build$ 
```

And same issue on `yarplogger` - no registered processes or messages.

Any idea if I am missing something? For information, we recently updated yarp,

```
icub@icub-desktop:~/Desktop/yLogTest/build$ yarp version
YARP version 2.3.65
```

Is there a configuration file somewhere that I need to adapt to our setup?

Thanks in advance,
Ryan

claudiofantacci, 2016-06-07:
Hello @rlober ,

since you didn't mention it explicitly, did you run `yarpserver` with `--log` option?

claudiofantacci, 2016-06-07:
Hi @rlober ,

in order to log info, warning, error, etc. messages into the `yarplogger` you have to run `yarprun` beforehand with the `--log` option, e.g. `yarprun --server /myyarprun --log`.
Then, each program must be run on the node `/myyarprun` and the messages will be automatically delivered to the `yarplogger`.

traversaro, 2016-06-07:
Quoting from https://github.com/robotology/yarp/issues/392#issuecomment-72917112 (sorry for the documentation not being in a proper place :( ) : 

> you can use the logger also if you start the module from the command line. In this case, you just need to set the environment variable `YARP_FORWARD_LOG_ENABLE=1` on your machine.

Beware that if you have a machine with `YARP_FORWARD_LOG_ENABLE=1` and on it you launch a program with the `yarprun` (i.e. for example through the `yarpmanager`) then you will see two different programs in the `yarplogger` windows.

randaz81, 2016-06-07:
...but the two logs will be different, i.e. the log obtained with yarprun --log just contains the yDebug(), yError() etc. while the log obtained with YARP_FORWARD_LOG_ENABLE=1 contains all the previous PLUS all the printf, cout etc. It is NOT recommended because it is very resource consuming!

rlober, 2016-06-07:
Well looking through the cmake options we only see `ENABLE_YARPRUN_LOG`

claudiofantacci, 2016-06-07:
@rlober is an environmental variable you can set on your local machine which is uncorrelated with CMake builds. As an example, on Linux machines you would have an `export YARP_FORWARD_LOG_ENABLE=1` into the user's .bash_rc file.

Following @randaz81 explanation I would suggest you to run `yarprun` with the `--log` option, then run  your programs on the yarprun node using `yarpmanager`/`yarpmanager++` (the latter has been recently added to YARP but needs to be enabled from CMake).

randaz81, 2016-06-07:
that's completely unrelated, since, as far as i remember, ENABLE_YARPRUN_LOG enables the internal log of yarprun and should be always off unless you are debugging yarprun core!

rlober, 2016-06-07:
@claudiofantacci @randaz81 

Gotcha, did not see the word "environment". Nevertheless I believe we will stick to the recommended way of doing things. `yarprun` was launched through `icub-cluster.py` so I can only assume that those flags are indeed used. We just ran `yarprobotinterface` through `yarpmanger` and got our log. 

I think this can be closed but I have one parting question: If I launch a binary from console which uses the `Log` utility how do I get it to register with `yarplogger`? Must I use, `export YARP_FORWARD_LOG_ENABLE=1`?

traversaro, 2016-06-07:
You can write: 
~~~
export YARP_FORWARD_LOG_ENABLE=1
./yarpy 
~~~

Or even just:
~~~
YARP_FORWARD_LOG_ENABLE=1 ./yarpy 
~~~

rlober, 2016-06-07:
Thanks @traversaro 

Closing.

closed 2016-06-07 by rlober