robotology/yarp #793: cannot disconnect mjpeg connection?

randaz81, 2016-06-08:
The following command:
> yarp disconnect /a /b

seems to have no effect on a mjpeg connection. @vtikha @drdanz @barbalberto

drdanz, 2016-06-08:
mjpeg connection is reversed. Try

```
yarp disconnect /b /a
```

drdanz, 2016-06-17:
@randaz81 This is a known behaviour, but I cannot find documentation about it.

This is an issue that deserves to be investigated, because in the long term, I would like to be able to close it properly.

```
yarpdev --device test_grabber
```

```
yarpview --name /view
```

```
yarp connect /grabber /view mjpeg
```

```
yarp admin rpc /grabber
>>list out
Response: null
>>list out null
Response: (from "/grabber") (to null) (carrier mjpeg) (push 0)
```

```
yarp admin rpc /view
>>list in
Response: "/grabber" anon_rpc
>>list in /grabber
Response: (from "/grabber") (to "/view") (carrier mjpeg) (push 0)
```

```
yarpview --name /view2
```

```
yarp connect /grabber /view2 mjpeg
```

```
yarp admin rpc /grabber
>>list out
Response: null null
>>list out null
Response: (from "/grabber") (to null) (carrier mjpeg) (push 0) (from "/grabber") (to null) (carrier mjpeg) (push 0)
```

So even if we could change the disconnect for reversed connections, at the moment the sender does not have the information required to perform it