robotology/yarp #795: Bug in iKinCartesianSolver / motorControlAdvanced

pattacini, 2016-06-08:
@serena-ivaldi commented on [Wed Jun 08 2016](https://github.com/robotology/icub-tutorials/issues/5)

Hello,
we just made a fresh install of YARP, icub-main and icub-tutorials and we tested the tutorial motorControlAdvanced. 
To launch it, we follow the instructions to set the simulator first. So we launch

iCub_SIM
simCartesianControl
iKinCartesianSolver --context simCartesianControl --part left_arm

The problem is that this command does not seem to work for iKinCartesianSolver on iCubSim.
It looks for the file cartesianSolver.ini but not in the proper context. It does not seem to look for simCartesianControl.
It does:

icub@precision:~$ iKinCartesianSolver --context simCartesianControl --part left_arm
||| clearing context
||| adding context [cartesianSolver]
||| configuring
||| added context simCartesianControl
||| default config file specified as cartesianSolver.ini
||| checking [/home/icub/cartesianSolver.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubNancy01] (robot YARP_CONFIG_HOME)
...

And at the end it finds:
found /home/icub/software/share/iCub/robots/iCubNancy01/cartesianSolver.ini

This file is configured for icub, so it does not work for icubSim.

By the way, it gives an unusual error also about an extended Torso state:

[ERROR]Problem connecting to /icub/torso/state:o from /cartesianSolver/left_arm/torso/state:i, is the remote device available?
[ERROR]*** Extended port /icub/torso/stateExt:o was not found on the controlBoardWrapper I'm connecting to. Falling back to compatibility behaviour
[WARNING]Updating to newer yarp and the usage of controlBoardWrapper2 is suggested***
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open

How can we fix this?
Thanks in advance

---

@pattacini commented on [Wed Jun 08 2016](https://github.com/robotology/icub-tutorials/issues/5#issuecomment-224679868)

Hi @serena-ivaldi 

Regarding the context search, it's very likely that **`YARP_DATA_DIRS`** does not point to all the required paths. Could you check this out?

A simple way to test if a specific file is found throughout the contexts is to launch:

```sh
> yarp resource --context simCartesianControl --find cartesianSolver.ini
```

For the second problem I'd suggest you to open up a ticket on dedicated tracking systems: e.g. `icub-support` or `yarp`. 

Some random thoughts:
- Did you update the robot software too?
- Did you check that you're launching the freshest binaries?
- Isn't there any port conflict?
- Last resort: did you try a complete power cycle of the whole system?

---

@serena-ivaldi commented on [Wed Jun 08 2016](https://github.com/robotology/icub-tutorials/issues/5#issuecomment-224682190)

Hi Ugo,

thanks for quick reply. 
If I launch yarp resource, then it does not find that context:

icub@precision:~$ yarp resource --context simCartesianControl --find cartesianSolver.ini
||| configuring
||| added context simCartesianControl
||| default config file specified as cartesianSolver.ini
||| checking [/home/icub/cartesianSolver.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubNancy01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubNancy01] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/iCubNancy01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/iCubNancy01] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/iCubNancy01] (robot YARP_CONFIG_DIRS)
||| checking [robots/iCubNancy01] (robot YARP_DATA_DIRS)
||| checking [/home/icub/software/share/yarp/robots/iCubNancy01] (robot YARP_DATA_DIRS)
||| checking [/home/icub/software/share/iCub/robots/iCubNancy01] (robot YARP_DATA_DIRS)
||| found /home/icub/software/share/iCub/robots/iCubNancy01
||| checking [config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/software/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/software/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/icub/software/share/iCub/robots/iCubNancy01/cartesianSolver.ini] (robot)
||| found /home/icub/software/share/iCub/robots/iCubNancy01/cartesianSolver.ini
||| finding file [from]
"/home/icub/software/share/iCub/robots/iCubNancy01/cartesianSolver.ini"

The context is in /home/icub/software/share/iCub/contexts/simCartesianControl which is included in YARP_DATA_DIRS:
export YARP_DATA_DIRS=$YARP_DATA_DIRS:/home/icub/software/share/yarp:/home/icub/software/share/iCub

It just seem to ignore the context. I made it work by manually setting icubSim into the file /home/icub/software/share/iCub/robots/iCubNancy01/cartesianSolver.ini
By this trick, the other error about the extended torso port disappears.

About the other things, we are installing a new machine, so it's a brand new installation from zero (started around 4PM this afternoon)

---

@pattacini commented on [Wed Jun 08 2016](https://github.com/robotology/icub-tutorials/issues/5#issuecomment-224683475)

So, it's a general problem of yarp installation.
Moving this issue then.

pattacini, 2016-06-08:
Actually, yarp doesn't go seek into `/home/icub/software/share/iCub/context`.
@drdanz @traversaro any clue?

drdanz, 2016-06-09:
@serena-ivaldi `robots` take precedence over `contexts`, and since the requested file was found for the current robot, the search stops and the file that was found is returned.
If you are using the simulator and not `iCubNancy01` you should either unset the `YARP_ROBOT_NAME` environment variable, or set it to `iCubSim` (or `iCubGazeboSim` depending on the simulator you are using)

traversaro, 2016-06-09:
Just a note: for the time being the `YARP_ROBOT_NAME` of the iCub simulated in Gazebo has been `icubGazeboSim`.
I don't like it, but until we change it (see https://github.com/robotology/icub-gazebo/issues/16) it is better to stick to it.

pattacini, 2016-06-11:
I think we can close this.

closed 2016-06-11 by pattacini