robotology/yarp #798: yarp.it uses an invalid security certificate

drdanz, 2016-06-14:
Not a big issue, but the browsers complain that https://www.yarp.it/ and https://yarp.it/ use an invalid security certificate.

* The certificate is not trusted because it is self-signed.
* The certificate is only valid for icub.org
* The certificate expired on 15/06/13 12:18.

I suggest at least to replace the certificate with at least a valid self-signed one, or even better, a signed one using [Let's Encrypt](https://letsencrypt.org/)

cc: @mbrunettini

mbrunettini, 2016-06-17:
To use HTTPS on yarp.it is a non-sense. There is nothing worth to be protected by SSL.

closed 2016-06-17 by mbrunettini

drdanz, 2016-06-17:
Then we should disable it instead...

mbrunettini, 2016-06-17:
It’s not currently possible due to the webserver configuration.