robotology/yarp #807: YARP does not compile on PC104 with image 1.8

jeljaik, 2016-06-20:
On `iCubParis01` we have been forced to use the icub-live image 1.8, but we are having problems compiling the latest `yarp` on PC104. This is the error we get:

```
/usr/local/src/robot/source/yarp/src/libYARP_OS/src/BottleImpl.cpp: In member function 'virtual void yarp::os::impl::StoreInt64::fromString(const yarp::os::ConstString&)':
/usr/local/src/robot/source/yarp/src/libYARP_OS/src/BottleImpl.cpp:722: error: 'strtoll' is not a member of 'ACE_OS'
make[2]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/BottleImpl.cpp.o] Error 1
make[1]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/all] Error 2
make: *** [all] Error 2
```
Additional info ... 

On PC104
```
icub@pc104:~$ dpkg -l | grep libace
ii  libace-dev                              5.4.7-12                             C++ network programming framework developmen
ii  libace5.4.7c2a                          5.4.7-12                             C++ network programming framework
```

ahoarau, 2016-06-21:
can you add `#include <stdlib.h>` on top of `/usr/local/src/robot/source/yarp/src/libYARP_OS/src/BottleImpl.cpp` ?

lornat75, 2016-06-27:
Hi, looks like strtoll is available in ACE version 5.7. One workaround could be do uninstall libace from the system and compile ace 5.7 from sources.

closed 2016-07-06 by drdanz

jeljaik, 2016-07-19:
Just wanted to conclude this issue by saying that in the end we replaced the PC104 for `iCubParis01`, updated all the related software as well as the firmware of the robot and our robot is not functional. Thanks for the support.