robotology/yarp #808: Qt 5.7 & Yarp failed link on Windows 10 (also iCub)

gregoire-pointeau, 2016-06-21:
Hello,
My installation was working on windows 10 until an automatic update last Friday (17 June).
Since I have the following problem when I launch any GUI (`yarpmanager`, `iCubGui`...)
I have the following error message: 
for `yarpmanager`:
> `?setSelectionModel@QListWidget@@UAEXPAVQItemSelectionModel@@@Z process entry point not found in the library link library C:\Robot\yarp\build\bin\Release\yarpmanager.exe`

for `iCubGui`:
> `?getProcAddress@QOpenGLContext@@QBEP6AXXZPBD@Z process entry point not found in the library link library C:\Robot\robotology\Qt\5.7\msvc2013\bin\Qt5OpenGL.dll`

I reinstalled Qt, re-pull everything, and rebuilt several times everything.

My configuration:
- Windows 10
- MVS 12 2013
- Cmake 3.5.2
- Qt 5.7

`Qt5_DIR: C:\Robot\robotology\Qt\5.7\msvc2013\lib\cmake`

INCLUDE has:
`C:\Robot\robotology\Qt\5.7\msvc2013\include`

PATH has:
`C:\Robot\robotology\Qt\5.7\msvc2013\lib`
`C:\Robot\robotology\Qt\5.7\msvc2013\bin`

Does anyone had a similar problem using windows 10?

Thanks

gregoire-pointeau, 2016-07-26:
Any update on it anyone ?

drdanz, 2016-07-26:
@gregoire-pointeau I'm sorry, I'm not a Windows user, perhaps @randaz81, @pattacini or @mbrunettini saw something similar? Are you sure that you don't have more than one qt5 installation in your path? I've seen strange behaviours on windows with recent versions of CMake that include qt5 dlls in its path.
Anyway it looks to me something related to your setup, not a bug in yarp, therefore I'm closing this, please reopen it if you find out that the bug is actually in YARP, or open a new one in the robotology/QA if you need more support with this issue.

closed 2016-07-26 by drdanz