robotology/yarp #809: yarp_add_plugin

AbuMussabRaja, 2016-06-23:
CMake Error at CMakeLists.txt:18 (yarp_add_plugin):
  Unknown CMake command "yarp_add_plugin".

Trying to add OpenCVGrabber and above is the error from CMake,

Ref: 
https://github.com/robotology/yarp/tree/master/src/modules/opencv

drdanz, 2016-06-23:
Did you enable the flag in YARP main build or by chance are you trying to build it alone? It doesn't seem like a possibile error in yarp build...

AbuMussabRaja, 2016-06-24:
Dear Daniele, TBH i am fairly new to yarp and icub and working on my fyp, when ever i get some bug and go to my adviser he says post on their forum :p so thats what i did.
Okay what i need to do is to stream a video on the iCub Simulator screen, (over a yarp port) and i read some where that i need OpenCVGrabber for that, so i was trying to add that :p i installed pre build binaries and couldnt find opencvgrabber or ffmpeg. :( 
i would be so so so so so thankful to you if you could point me to some direction where to look instead of simulator readmes and stuff :(

drdanz, 2016-06-27:
Sorry, neither the OpenCV grabber, nor the ffmpeg grabber come with the binary releases, you will have to rebuild yarp in order to use them, unless someone wants to build them...
Which visual studio version are you using by the way? And is it 32 or 64 bit?

AbuMussabRaja, 2016-06-27:
Thank you so much for your reply.
i am using VS11, 2012 and 32 bit. tried building it once from sources and hell of errors just came to ruin my day :p

drdanz, 2016-07-06:
Sorry, but at the moment the only way to do it at the moment is to rebuild yarp and enable the carriers, you can find detailed instructions in YARP documentation and iCub wiki.
Unless some windows using VS11 32bit wants to step in, build these carriers, and send you the dll there is nothing I can do at the moment. Perhaps we can consider including it in a future binary release...

closed 2016-07-06 by drdanz