robotology/yarp #812: yarpscope: min and max values in plots

gsaponaro, 2016-06-29:
Hi @drdanz @iron76, I've been using **[yarpscope](http://www.yarp.it/yarpscope.html)** successfully (thanks!) and I have a couple of comments:

* suggestion - in the documentation, initially it says that the ```--min``` and ```--max``` command-line parameters (Simple Mode) affect the X axis, while about the XML Mode at the end of the page it says that ```minval``` and ```maxval``` are the minimum and maximum values on the Y axes. I think that it should refer to Y axis in both cases because X is time, right?
* issue - setting Y ranges to be different than the defaults has no effect for me. I tried both with the Simple Mode command-line parameters and with the XML minval/maxval tags like in the examples. I confirmed that the XML is correctly loaded and parsed by yarpscope - changes in the other fields affect the plot, but these Y ranges do not. For example, trying to force the Y axis between 0.0 and 1.0 (the range of the signal which I want to plot) has no effect when yarpscope is started:

[image]

drdanz, 2016-07-04:
@gsaponaro Thanks for the documentation patch.
About the second issue, when I run ``yarpscope --min 0.0 --max 1.0`` I get a window with the correct limits.
On which platform are you running it?
Are you using the Qt5 version, or still using the GTK2 version?
Can you show me the command line you are using?
Perhaps there is some rescaling involving when data is received?
Have you tried starting ``yarpscope`` without automatic connection, and connect it later?

gsaponaro, 2016-07-05:
Hi @drdanz, thanks for the help.

My setup: Ubuntu 14.04, yarp 2.3.66, Qt5 GUIs.

I have a fairly complex yarpscope XML with many subplots, but I prepared a minimal example just for the sake of this issue.

To reproduce the problem, try running **yarpscope** in XML mode with [this file](https://gist.github.com/gsaponaro/7bc7a5fddb06b4f9bca7a855adba0337) that contains one plot with ```minval="0.0"``` and ```maxval="1.0"```:

```yarpscope --xml giovanni_yarpscope_debug2.xml```

Note that data is not yet received at this point (the port does not even exist), but I see a window with Y from 0.0 to 5.0 (where did this number come from?), like this:

[image]

When I create ```/some_port_with_data_between_zero_and_one:o``` and write some numbers in there, **yarpscope** plots the values, but Y is still at 5.0 (which is expected since there was no rescaling).

drdanz, 2016-07-06:
@gsaponaro I found the bug... the values were read as int instead of float, see #819

gsaponaro, 2016-07-06:
Cool, works like a charm now. :+1:

PR #819 (2016-07-06): yarpscope: Fix reading minval and maxval from xml
Fixes #812

cc @gsaponaro
<a href='#crh-start'></a><a href='#crh-data-%7B%22processed%22%3A%20%5B%22https%3A//github.com/robotology/yarp/pull/819%23issuecomment-230738534%22%5D%2C%20%22comments%22%3A%20%7B%22General%20Comment%22%3A%20%7B%22html_url%22%3A%20%22https%3A//github.com/robotology/yarp/pull/819%23issuecomment-230738534%22%2C%20%22comments%22%3A%20%5B%7B%22body%22%3A%20%22Confirmed%20working%2C%20thanks.%22%2C%20%22created_at%22%3A%20%222016-07-06T10%3A41%3A32Z%22%2C%20%22user%22%3A%20%7B%22avatar_url%22%3A%20%22https%3A//avatars.githu [... cut]
commit 05d9630d (2016-07-06): yarpscope: Fix reading minval and maxval from xml

Fixes #812
closed 2016-07-06 by gsaponaro