robotology/yarp #815: yarprobotinterface: if it doesn't find yarp server,  doesn't print any error and doesn't exit

valegagge, 2016-06-30:


drdanz, 2016-07-04:
@valegagge when I run `yarprobotinterface` without yarp server running I receive this error.

```
[FATAL]Sorry YARP network does not seem to be available, is the yarp server available? 
yarprobotinterface(_ZN4yarp2os9LogStreamD1Ev+0x1aa) [0x421f88]
yarprobotinterface(main+0x94) [0x421b61]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf5) [0x7f9aeb51ef45]
yarprobotinterface() [0x421a09]
```

Are you using some particular configuration? On which platform?

valegagge, 2016-07-05:
I noticed this issue on pc104 with a bad configuration file of yarpserver.

I launched yarp server on pc-desktop and then I launched yarprobotInterface (on pc104). It run correctly but it opened unnamed ports.
The problem was due to an error in file yarp.conf file. (wrong ip address)

drdanz, 2016-07-08:
So this is probably not an issue in `yarprobotinterface`, but in `yarp::os::Network::checkNetwork()` that for some reason finds a network available even if it is not the right one.
Any other YARP program would have the same behaviour.

@valegagge Can you add some details about how to reproduce it?

francesco-romano, 2016-07-19:
I remember also @alecive complaining about this stuff last year.

claudiofantacci, 2016-07-19:
The behavior I'm experiencing is different. Invoking checkNetwork() blocks the program when ```yarpserver``` is not running, without any possibility to recover.
OS: macOS El Capitan 10.11.5.

traversaro, 2016-07-19:
> The behavior I'm experiencing is different. Invoking checkNetwork() blocks the program when yarpserver is not running, without any possibility to recover.
OS: macOS El Capitan 10.11.5.

That typically happens if you have saved a yarpserver (in the file that you can get from `yarp conf` on a network (such as `192.168.1.1`) to which you are not connected at the moment. Adding a timeout (as in https://github.com/robotology/gazebo-yarp-plugins/commit/dc8196c5c234cdf425cf3bc54a35d4c248e0fc2e) typically solves the issue.

claudiofantacci, 2016-07-19:
Indeed @traversaro that's my case.

claudiofantacci, 2016-11-17:
Is this still an issue?
Is there any way to replicate the issue?
@valegagge @traversaro

traversaro, 2016-11-17:
I never experienced this issue.

claudiofantacci, 2016-11-17:
Ok, let's wait @valegagge feedback. Then we might consider closing this issue.

closed 2020-03-04 by no-response[bot]