robotology/yarp #822: Python bindings for an existing C++ code

clement-moulin-frier, 2016-07-12:
Hello,

I have existing C++ code using `yarp` and I would like to generate Python bindings for it. 
For example, let's consider that I want to create bindings for [that code](https://github.com/robotology/yarp/blob/master/example/rfmodule/main.cpp) (from the repo `example` directory), such that I can then construct an instance of `MyModule` directly in Python. Note that this is different from using the existing Yarp Python bindings to re-code `MyModule` directly in Python. I don't want to re-code my existing code using the provided Yarp Python bindings, but instead to bind my existing C++ code to Python.

I guess that it should be feasible by e.g. adapting the [CMake you use for the provided bindings
](https://github.com/robotology/yarp/blob/master/bindings/CMakeLists.txt), so that it can generate bindings for my specific C++ code using the same swig configuration (`yarp.i` etc...).

However that CMake file is quite complex and I have no idea how I could adapt it to my need. Maybe you have already tried to do such a thing and could point me to an easy solution?

Thanks a lot

Tobias-Fischer, 2016-07-14:
Hi @clement-moulin-frier,
I think it's quite easy if you go for the hack-ish way. You first have to separate header and source, so in your example you would copy the class in its own header file (and disregard the `main` as you don't need it). Then you copy the `.h` and `.cpp` file in a convenient location, e.g. `yarp/src/libYARP_OS/include/yarp/os` and `yarp/src/libYARP_OS/src` respectively, and add them to the `libYARP_OS/CMakeLists.txt` file. Then you `make` and `make install` as usually.
All what's left now is adding the new header file to the `yarp.i` file, e.g.
```
%{
/* Includes the header in the wrapper code */
#include <yarp/os/MyModule.h>
%}
 
/* Parse the header file to generate wrappers */
%include <yarp/os/MyModule.h>
```
Now you recompile the bindings and you're ready to go.

Obviously, a much nicer way of doing things is not putting your files in the `libYARP_OS` directory, you can achieve the same things in a custom directory if you write an according `CMakeLists.txt` file and include this from one of the `CMakeLists.txt` files of yarp. All you need is that the header can be found within swig, and the source file needs to be compiled obviously.

Best, Tobi

drdanz, 2016-07-15:
Thanks @Tobias-Fischer for replying.
A less hack-ish option would be to create your own swig module, have a look at the swig tutorials around...

Closing this, feel free to reopen if you need more help

closed 2016-07-15 by drdanz

clement-moulin-frier, 2016-07-15:
Thanks for the rapid answers guys. I'm first trying the hack-ish option provided by @Tobias-Fischer , then I'll try to do it the clean way. I'll keep you informed here if I find a generic and clean solution.

@Tobias-Fischer : Since we are working on the same project it's actually worth explaining you what I have in mind (I do it here but we can then continue the discussion elsewhere, e.g. Slack). 
The idea is to bind the functionalities provided in the [`wrdac` libraries](https://github.com/robotology/wysiwyd/tree/master/main/src/libraries/wrdac) in order to be able to use them in an interactive way through a Python shell. 
A relevant example is the `opcClient`: it would be very convenient to be able to manipulate the `OPC` directly from a Python shell (adding and removing objets, etc...), instead of relying on the `opcPopulater` (which requires coding a function for each specific manipulation one wants to do), or relying on the rough syntax of the `OPC` rpc commands.

Any opinion on this? (we can continue discussing it in another place, e.g. Slack)