robotology/yarp #826: Yarp nodes not properly detected within ROS

Tobias-Fischer, 2016-07-14:
Hi,
All is working fine with the yarp to ROS communication, however there is an annoying issue when using `roswtf`. Yarp seems to use the IP+port as identifier for a node rather than the name:
```
WARNING The following nodes are unexpectedly connected:
 * unknown (http://10.0.0.139:10007/)->/rviz_node (/laser)
ERROR The following nodes should be connected but aren't:
 * /cer_laser->/rviz_node (/laser)
```
So the identifier `(http://10.0.0.139:10007/)` should really be `/cer_laser`.

I'd be happy to help resolving this issue, however I have no clue on how to fix this.

Thanks, Tobi

drdanz, 2016-07-14:
YARP port do not implement all the [ROS slave API](http://wiki.ros.org/ROS/Slave_API) yet.
`roswtf` is probably calling some of the methods not yet implemented.
The fix will require some coding in YARP PortCore and perhaps in the xmlrpc and tcpros carriers and I'm not sure if it is going to be easy...
If you want to help, a good start would be to find out exactly what methods are called by `roswtf`...`wireshark` and the link above can help you in this!

Tobias-Fischer, 2016-07-15:
Hi @drdanz,
Thanks for the hint! You were right, we need to implement `getBusInfo` from the slave API. I think I nearly finished a first implementation, however there is one piece of information I don't know how to access.
How do I know which host / port a `NodeItem` is connected to? I can get the address of the `NodeItem` itself using `item.contactable->where().toURI()`, however I need the address of the items it is connected to.
Do you have any clue how to achieve this @drdanz?

Thanks, Tobi

Tobias-Fischer, 2016-07-15:
Hi,
I got it all working nicely. I'll create a pull request next week.

Best, Tobi

drdanz, 2016-08-02:
Thanks a lot @Tobias-Fischer for working on this.

closed 2016-08-02 by drdanz