robotology/yarp #832: yarpmotorinterface silently suppresses groups in which one parameters list is not ending with a parenthesis

traversaro, 2016-07-19:
Example: 
~~~
  <device name="wholebodydynamics" type="wholebodydynamics">
        <param name="modelFile">model.urdf</param>
        <param name="fixedFrameGravity">(0,0,-9.81)</param>
   
        <group name="FOO">
            <param name="bar">(1.0,2.0,
                                                 3.0,4.0</param>
        </group>
  </device>
~~~

In this case the `FOO` group is simply not passed to the `open` of the device, without any printed warning or error. If the parser encounter an error of this kind, it should fail as soon as possible to simplify debugging of malformed configuration files. 

@DanielePucci @fjandrad

DanielePucci, 2016-07-21:
-.-
~~~
without any printed warning or error
~~~

pattacini, 2016-12-13:
I'm having big problems launching the `cartesiancontrollerserver` components along with the simulator.

If I type:
```
$ yarprobotinterface --context simCartesianControl --config no_legs.xml
```

I get:
```
[ERROR]SolverNameToConnect option is missing
```

The option is there.

@drdanz has this error something to do with this patch?

Nicogene, 2017-05-05:
Which is the status of this issue? Has been fixed by #995?

traversaro, 2017-05-05:
I think so, as far as I remember.

closed 2017-05-05 by traversaro