robotology/yarp #836: Yarp crahes when connecting to ports

pattacini, 2016-07-29:
@KevinAStein commented on [Fri Jul 29 2016](https://github.com/robotology/QA/issues/129)

Hi,
I have a program using yarp ports as described in http://www.yarp.it/rpc_ports.html
It used to work fine since lately we started using the ```devel``` branch of yarp because @traversaro developed a floating base estimation that requires that branch. Now every time I try to connect to a RPC or buffered port it results in a segmentation fault. Neither ```yarp rpc /portname``` nor ```yarp read ... /portname``` work.
 Can somebody help me with this?
Thanks

 @hu-yue 

---

@traversaro commented on [Fri Jul 29 2016](https://github.com/robotology/QA/issues/129#issuecomment-236137747)

There have been recently some fixes related to memory access in the `devel` branch, are you using the latest commit in the `devel` branch?

---

@traversaro commented on [Fri Jul 29 2016](https://github.com/robotology/QA/issues/129#issuecomment-236138147)

By the description of the problem, it could be something related to having two different version of YARP in the system. Can you check that you have (for example) a unique `libYARP_sig.so` (or at a maximum two, one for the `build` folder and one for the install folder)?

---

@pattacini commented on [Fri Jul 29 2016](https://github.com/robotology/QA/issues/129#issuecomment-236139263)

This is a yarp related issue. Moving to yarp repository

KevinAStein, 2016-07-29:
I checked and made sure that there is only one in the build/lib and one in the install. I still get the error. Other programs are working fine, though and I can not see a difference in the code.

traversaro, 2016-07-29:
You have this problem on every RPC port or just in one opened by a specific program?

KevinAStein, 2016-07-29:
I have the problem on every port by that program.

drdanz, 2016-07-29:
Are you using YARP on just one machine or on more than one? If more than one, are all of them on devel and on the same commit?

KevinAStein, 2016-08-01:
I am running it on the same computer and it works fine using the master branch but crashes using the devel branch

traversaro, 2016-08-01:
The program that is giving problems is the new experimental floating base estimator?

KevinAStein, 2016-08-01:
no, it is code for generating walking patterns

KevinAStein, 2016-08-01:
Hi,
I guess I found what causes the issue:
In my code I had a flag to either run it with or without yarp and therefore put ``` Network yarp; ``` inside an if statement. For some reason this works fine using the master branch but not in the devel branch!

So if I run the code attached it does not register the port with yarpserver and runs the while loop without waiting for an answer. 

```
#include <yarp/os/all.h>
#include <stdio.h>
using namespace yarp::os;
int main(int argc, char *argv[]) {
    if (argc<2) {
        fprintf(stderr, "Please supply a port name for the server\n");
        return 1;
    }
    const char *name = argv[1];
    RpcServer port;  

    bool use_yarp = true; 
    if (use_yarp){  
      Network yarp;     
    }

    port.open(name);
    while (true) {
        printf("Waiting for a message...\n");
        Bottle cmd;
        Bottle response;
        port.read(cmd,true);
        printf("Got message: %s\n", cmd.toString().c_str());
        response.addString("you");
        response.addString("said");
        response.append(cmd);
        printf("Sending reply: %s\n", response.toString().c_str());
        port.reply(response);
    }
}
```

pattacini, 2016-08-01:
@KevinAStein when yarp is not initialized, the behavior is unpredictable: it's not a matter of master/devel branch then.

When the scope where you instantiate the yarp network ends, then the network itself gets deallocated, causing the problem you're experiencing with your code.

Therefore, the fix consists in instantiating yarp network differently, e.g. soon afterwards `main()` begins.

barbalberto, 2016-08-01:
That code seems bugged to me in both cases...
If `use_yarp` is `false`, then how can ports to be opened and used?

If `use_yarp` is `true`, the Network object is destroyed right after the if, so it is like it was not initialized at all. 
If this code DOES work, then there is a bug somewere because it should not.

Also, ALWAYS check the result od port.open. That's the best way to detect if something is going wrong with the network

hu-yue, 2016-08-01:
Ok, then I think there's a bug in the master branch of YARP, because it was working if `Network yarp` is inside the if statement, with the `master` branch of YARP, which also means then that the behaviour is the correct one with the `devel` branch.

pattacini, 2016-08-01:
@hu-yue as I said, when the network is not initialized, the final outcome is very variable and mostly unpredictable. I'm afraid, indeed, that no changes between `master` and `devel` occurred in this respect.

By the way, @drdanz or @barbalberto might add a word on this.

drdanz, 2016-08-01:
This statement
```
    if (use_yarp){  
      Network yarp;     
    }
```
Calls both `yarp::os::Network` constructor and destructor. The constructor calls `Network::init()`, and the destructor calls `Network::fini()`. After `Network::fini()`, you should not be able to create port (like before calling `Network::init`), I'm not sure why it allows you to do so.

An alternative would be to call `init()` and `fini()` directly:

```
    if (use_yarp){
        yarp::os::Network::init();
    }
```
and later

```
    if (use_yarp){
        yarp::os::Network::fini();
    }
```
Or eventually to use a pointer to a `Network` and create/destroy it using `new` and `delete`.

francesco-romano, 2016-08-01:
Totally agree with @pattacini. The behaviour is undefined. The solution by @drdanz  is what I usually use if I cannot create a network object (explicitly call `init` and `fini`)

KevinAStein, 2016-08-01:
so I figured that my original code opens the ports in the if statement:

```  
if (tests.use_yarp){    
    Network yarp;
    if (!yarp.checkNetwork()){
      cout << "ERROR: yarp.checkNetwork() failed."  << endl;
    }
    else if (tests.PG_command_in.open("/PatternGenerator/command:i") &&
        tests.PG_data_out.open("/PatternGenerator/data:o") &&
        tests.PG_time_in.open("/PatternGenerator/time:i") &&
        tests.PG_zmp_in.open("/PatternGenerator/zmp:i") &&
        tests.PG_com_in.open("/PatternGenerator/com:i")) {
      cout << "all Ports open" << endl;
    }
    else{
      cout << " [ERROR] using yarp" << endl;
    }
  }
```
and then run the while loop where I receive the commands:

```
  if (tests.use_yarp){ 
    while (true){
      cout << "waiting for command:" << endl;
      Bottle cmd;
      Bottle response;
      tests.PG_command_in.read(cmd, true);
      .......
   }
}
```

this, for some reasons, works with the master branch but does not with the devel branch.
However I will correct my code with the suggestions, thanks for your help!

drdanz, 2016-08-01:
Glad we helped, I'm not sure why does it work on master, but that was an odd behaviour anyway.
Closing this issue, feel free to reopen it if you need more help on this.

closed 2016-08-01 by drdanz