robotology/yarp #837: Problems displaying fonts in 4k monitors

lornat75, 2016-07-29:
All guis have problems displaying fonts in high-res monitors. 

The problems is probably this one:
http://doc.qt.io/qt-5/highdpi.html

lornat75, 2016-07-31:
Checking the documentation more carefully we need to ensure that the line:

qputenv("QT_DEVICE_PIXEL_RATIO",QByteArray("2"));

does not cause problems to systems with low-resolution. More importantly this feature is deprecated in qt 5.6, so we need to make it version specific and remove it once we stop supporting qt 5.4 and (probably) 5.5.

lornat75, 2016-07-31:
These commits have been reverted because did mess up the visualization on normal screens.

lornat75, 2016-07-31:
Looks like the safest solution at the moment is to leave to the user to set the environment variable:

QT_DEVICE_PIXEL_RATIO = 2

This can be done at the system level (recommended) in the control panel or on each terminal.

lornat75, 2016-07-31:
Proper fix should be done using new features in qt 5.6. 

@mbrunettini maybe in the next release of the binaries we should replae qt .5.4 with qt 5.6 .

@drdanz, @aitek4iit  what do you think?

pattacini, 2016-07-31:
Quick update: together with @lornat75 we've tested that `QT_DEVICE_PIXEL_RATIO = auto` works for both high-res and normal systems.

lornat75, 2016-07-31:
@pattacini I have updated the pull request, would you mind give it a try?

pattacini, 2016-07-31:
@lornat75 it works nicely; just merged!

mbrunettini, 2016-08-01:
Ok, I will update to Qt 5.6 where it is supported.

aitek4iit, 2016-08-01:
Yes the AUTO flag should do the trick for both resolution and I think Qt 5.6 wolud manage better the high dpi resolutions beacuse in the 5.4 version this feature was only a beta.

mbrunettini, 2016-08-01:
QT 5.6 is only avavilabe on windows with the following versions:
-VS 2013 (V12) 32 and 64 bit
-VS 2015 (V14) 32 and 64 bit

claudiofantacci, 2016-12-19:
I'm writing here concerning about `QT_DEVICE_PIXEL_RATIO` qtputenv option.

Under macOS, QT version 5.7.0 (from brew), launching `yaprdataplayer`, `yaprview`, `yarpmotorgui`, the following warning is raised:
```
Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use: 
    QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors. 
    QT_SCREEN_SCALE_FACTORS to set per-screen factors. 
    QT_SCALE_FACTOR to set the application global scale factor.
```

Probably `QT_AUTO_SCREEN_SCALE_FACTOR` set to `1` (boolean true) will make the job done.

drdanz, 2016-12-20:
@claudiofantacci Opened a new issue, see @1038

Nicogene, 2017-05-08:
Changes introduced in #1200 should solve, for now I close this issue, if someone encounter some problems I will re-open it

closed 2017-05-08 by Nicogene