robotology/yarp #867: Fix remaining travis builds with integration tests due to timeouts

Tobias-Fischer, 2016-08-12:
Hi,
Some of the travis builds (https://travis-ci.org/robotology/yarp/) which perform integration tests are sometimes timing out. I think there are ways around this, and it would be nice to always have the integration tests tested.
Ways around:
https://docs.travis-ci.com/user/common-build-problems/#My-builds-are-timing-out
-> prefix the command which does not produce outputs for a long time with `travis_wait` or `travis_wait X` where X are the minutes to wait

I've also seen the cache archiving timing out, which can also be circumvented (https://docs.travis-ci.com/user/caching/#Setting-the-timeout):
```
cache:
  timeout: 1000
```
Best, Tobi

PS: Is there already an issue regarding `thrift` with `XCode`?

drdanz, 2016-08-16:
> PS: Is there already an issue regarding thrift with XCode (https://travis-ci.org/robotology/yarp/jobs/151730452)?

I don't think there is one, and I don't think it is just XCode, but any multi-config CMake Generator (i.e. Visual Studio as well)

Tobias-Fischer, 2016-08-18:
The tests still fail sometimes, now due to a timeout in `ctest`. It should be fairly easy to increase that timeout. Also, splitting the test in its shared and static components should half the test time ;) (well, the total time will be the same, obviously). I'll do it once I have some time to play around :P

Tobias-Fischer, 2016-08-21:
Can close this after https://github.com/robotology/yarp/commit/69b59f5d724f3f75b4779867f43f743c60edae41

closed 2016-08-21 by Tobias-Fischer