robotology/yarp #870: Tools missing documentation and man-pages

drdanz, 2016-08-16:
Several tools and GUIs are missing documentation and/or a man page:

- [ ] `yarpbatterygui`
- [x] `yarplaserscannergui`
- [ ] `yarplogger`
- [ ] `yarplogger-console`
- [ ] `yarpmanager++`
- [ ] `yarpmanager-console`
- [ ] `yarprobotinterface`
- [ ] `yarpidl_rosmsg`
- [ ] `yarpidl_thrift`
- [ ] `yarpros`
- [ ] `yarpserver`

drdanz, 2016-08-16:
`yarplaserscannergui` was in the source code, I moved it in the right place

drdanz, 2016-08-16:
`yarpserver` documentation was in `yarp server` documentation, I moved it in the right place with some small updates.

pattacini, 2016-08-16:
For `yarplogger` see also #638.

randaz81, 2017-05-12:
http://www.yarp.it/yarprobotinterface.html is missing.  😭

jgvictores, 2021-02-15:
> http://www.yarp.it/yarprobotinterface.html is missing. 😭

Some info at: <http://wiki.icub.org/wiki/Robot_configuration> ([perma](https://web.archive.org/web/20200221154145/http://wiki.icub.org/wiki/Robot_configuration))