robotology/yarp #872: Yarp IDL not working for multi-config cmake generators

Tobias-Fischer, 2016-08-16:
Hi,
As can be seen in https://travis-ci.org/robotology/yarp/jobs/151730452, the idl demos fail when XCode is used as there are cmake errors (e.g. `YARPIDL_thrift_LOCATION-NOTFOUND`).
@drdanz thinks this might happen on any multi-config cmake generator (https://github.com/robotology/yarp/issues/867).
Just reporting this here, I am not an expert on IDL/thrift, so it would be great if someone else could detail why this might happen. Not sure whether https://github.com/robotology/yarp/issues/487 or especially https://github.com/robotology/yarp/issues/368 might be related.

Best, Tobi

drdanz, 2016-08-16:
Actually I think that the IDL works, the issue is in CMake that does not look for the IDL tools in the current config directory.

Tobias-Fischer, 2016-08-17:
I've had a quick look, and this can easily be resolved. All what needs to be done is removing the `if (MSVC)` statement in https://github.com/robotology/yarp/blob/master/conf/YarpDescribe.cmake#L26-L28

Is there any reason why the `Debug` and `Release` folders shouldn't be added to the `YARP_IDL_BINARY_HINT` in Linux? If there is no good reason, I suggest removing the `if` statement, and we're done :).

drdanz, 2016-08-17:
AHA! Good catch @Tobias-Fischer!

Something like this should fix it, then:

```diff
-set(YARP_IDL_BINARY_HINT ${CMAKE_BINARY_DIR}/bin)
-if (MSVC)
-    set(YARP_IDL_BINARY_HINT -"${YARP_IDL_BINARY_HINT};${YARP_IDL_BINARY_HINT}/Debug;${YARP_IDL_BINARY_HINT}/Release")
-endif(MSVC)
+set(YARP_IDL_BINARY_HINT ${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR})
+foreach(_config ${CMAKE_CONFIGURATION_TYPES})
+  list(APPEND YARP_IDL_BINARY_HINT ${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}/${_config}"
+endforeach()
```

The `YARP_IDL_TO_DIR` has a few issues and I hope to remove it in the future... Here's another, when you are building in Debug, you probably want to link the Debug version of YARP, and you probably want to use the Debug version of `yarpidl_thrift`, and this will find the Release version instead. Anyway, let's just fix the build for now!
@Tobias-Fischer do you want to make a pull request for this or should I?

Tobias-Fischer, 2016-08-18:
I think it's all fixed, just waiting for Travis now. I also deleted the allowed failure from Travis. Regarding your Release vs. Debug issue: Wouldn't it be sufficient to do the following?
```
set(YARP_IDL_BINARY_HINT ${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR})
set(YARP_IDL_BINARY_HINT "${YARP_IDL_BINARY_HINT};${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}/${CMAKE_BUILD_TYPE}")
foreach(_config ${CMAKE_CONFIGURATION_TYPES})
  set(YARP_IDL_BINARY_HINT "${YARP_IDL_BINARY_HINT};${CMAKE_BINARY_DIR}/${CMAKE_INSTALL_BINDIR}/${_config}")
endforeach()
```
In this way, it always first looks in the path corresponding to the `CMAKE_BUILD_TYPE`.

drdanz, 2016-08-18:
Unfortunately that won't work, because `CMAKE_BUILD_TYPE` and `CMAKE_CONFIGURATION_TYPES` are a different thing, the first is for single configuration generators like the makefile one, the other is for multi configuration generators like xcode and msvc. On these the configuration can be changed from the IDE (i.e. you can build both Debug and Release in the same build folder), but the Config file is generated only once when you run CMake, and in that moment `CMAKE_BUILD_TYPE` is usually undefined.
Anyway `YARP_IDL_TO_DIR` is broken and has to be removed at some point, so I wouldn't waste time on trying to fix it...

Tobias-Fischer, 2016-08-18:
Ahhh I see. Thanks for the info!

drdanz, 2016-08-18:
@Tobias-Fischer Pull request merged. Thanks!

commit d4463aa9 (2016-08-18): Don't allow failure for Xcode in Travis anymore.

This fixes #872
commit 274a263b (2016-08-18): Add configuration types to idl binary hint. Fixes #872
closed 2016-08-18 by drdanz