robotology/yarp #878: Announce new releases

traversaro, 2016-08-19:
Classically open source projects announce their new releases using dedicated low volume mailing lists, so anyone interested in being informed on the new releases can just join the dedicated mailing list and forget about it, and then received an email announcing the new release. 

For yarp new releases, we have starting to rely on the  [`robotology/QA`](https://github.com/robotology/QA) repository to make announcements related to iCub Facility-mantained software, under the label [`announcements`](https://github.com/robotology/QA/issues?q=label%3Aannouncement). 

This is great for people subscribed to the `QA` repo, but it is a bit inconvenient for people that are interested only on `announcements`, as there is no way to just suscribe to `announcement` posts. 

For this reason I think it would be convenient to modify the release instructions (that I cannot find at the moment) to send a link to the Release Notes (or even to the related `robotology/QA` issue)
on the `rc-hackers` mailing list, as proposed by @lornat75  in https://github.com/robotology/QA/issues/118#issuecomment-226155395 .

cc @lornat75 @drdanz @mbrunettini @pattacini

drdanz, 2016-08-19:
+1 for using a mailing list for announcements

pattacini, 2016-08-19:
We're already emailing developers for the announcements, plus posting the same announcements on [Q&A](https://github.com/robotology/QA). I think we only need to write this on the stone of the release instructions as suggested by @traversaro.

drdanz, 2016-08-19:
Ah, I see... Release instructions are in [scripts/admin/README_RELEASE.md](https://github.com/robotology/yarp/blob/master/scripts/admin/README_RELEASE.md)

drdanz, 2016-11-10:
I think we can close this now.

closed 2016-11-10 by drdanz