robotology/yarp #880: Random crashes in unit tests

drdanz, 2016-08-19:
There are a few tests that sometimes fail with random crashes. I'll use this issue to keep track of them

- [ ] `OS::PortTest`
  - 2016/08/08 - `master` - [2442.26](https://travis-ci.org/robotology/yarp/jobs/150567776) - `***Exception: SegFault  3.67 sec`
- [ ] `OS::NodeTest`
  - 2016/08/05 - `devel` - [2438.6](https://travis-ci.org/robotology/yarp/jobs/150047588) - `***Exception: Other  0.40 sec`
- [ ] `OS::PublisherTest`
  - 2016/08/07 - `devel` - [2502.6](https://travis-ci.org/robotology/yarp/jobs/152940446) - `***Exception: Other  0.80 sec`
- [ ] `dev::PolyDriverTest::Valgrind::MemCheck`
  - 2016/08/18 - `master` - [2521.28](https://travis-ci.org/robotology/yarp/jobs/153223497) - `***Timeout 1500.00 sec`
- [ ] `dev::ControlBoardRemapperTest`
  - 2016/08/19 - `devel` - [2533.10](https://travis-ci.org/robotology/yarp/jobs/153301421) - `***Exception: SegFault  1.30 sec`
  - 2016/08/07 - `devel` - [2502.22](https://travis-ci.org/robotology/yarp/jobs/152940464) - `***Timeout 1500.00 sec`
- [ ] `dev::FrameTransformClientTest`
  - 2016/08/16 - `devel` - [2489.29](https://travis-ci.org/robotology/yarp/jobs/152676268) - `***Failed    6.10 sec`

Tobias-Fischer, 2016-08-21:
- [ ] `integration::rpc`
  - 2016/08/19 - `master` - [2535.19](https://travis-ci.org/robotology/yarp/jobs/153322232) - `***Failed   12.97 sec` (`Cannot make connection`)

traversaro, 2016-09-08:
* [ ] `dev::PolyDriverTest::Valgrind::MemCheck` - [2625.26](https://travis-ci.org/robotology/yarp/jobs/158236933) - `***Timeout 1500.00 sec`

drdanz, 2016-10-03:
The `OS::NodeTest` seems to be failing quite often recently on builds 11 (gcc) and 15 (clang), i.e. macOS, `TRAVIS_WITH_ACE=false`, `TRAVIS_WITH_CXX11=true`. Interesting fact, it happens both on `master` and `devel`, therefore it should not be related to the latest changes for ROS...

drdanz, 2016-10-03:
> The OS::NodeTest seems to be failing quite often recently on builds 11 (gcc) and 15 (clang), i.e. macOS, TRAVIS_WITH_ACE=false, TRAVIS_WITH_CXX11=false. Interesting fact, it happens both on master and devel, therefore it should not be related to the latest changes for ROS...

I just saw it on a build 16 (clang, TRAVIS_WITH_ACE=false, TRAVIS_WITH_CXX11=true), but I had to restart the build since it is on a pull request)

At this point I think it might be something on macOS + SKIP_ACE enabled, I've never seen this only recently and quite often.
Perhaps it is related to one of these commits that modified macOS stuff:

    59c42ae Fixed MCAST on macOS
    8c72b50 Fixed UDP os macOS
    e075ff2 Updated OSX System Info
    7673218 Added draft of ProcessInfo
    11b0de7 Initial implementation of SystemInfo for OSX
    75c8318 YARP_OS: Fix PlatformThread for OSX, no ACE, no c++11

@traversaro @francesco-romano Can you please try to riproduce this on a Mac and see if reverting one or more of these commits it disappears?

traversaro, 2016-10-04:
Brief recap ( on my machine ): 
In master & devel, it is possible to reproduce the failure with approx 20 runs with `ctest -R NodeTest --output-on-failure --repeat-until-fail 100`. Going back in time. 
* 75c8318 Fails in the same way. Commits before it fail to compile. 
* v2.3.65 , v2.3.66 Fail to compile. 
* v2.3.64 Fails in the same way.

drdanz, 2016-10-04:
Maybe we should move the macOS + SKIP_ACE builds to the allowed failures, this crash happens very often...

drdanz, 2016-10-12:
Ok, this is getting *REALLY* annoying, I moved the macOS + SKIP_ACE builds to the allowed failures, I will re-enable them later if we find a fix. Anyway I don't think that the combo macOS + SKIP_ACE is really useful, we might also consider disabling these builds completely...

traversaro, 2016-10-12:
Yes, given the amount of work require by that, we could simply declare macOS + SKIP_ACE "unsupported".

aerydna, 2016-11-09:
the FrameTransformClientTest failure (the race condition one) should be fixed by the commit 20f8c792c0288878862159b0ac10c32055c529e6. I'm investigating on the memcheck one and they could have the same cause (and so this could be fixed as well).

drdanz, 2016-11-10:
@aerydna unfortunately the memory leak is not fixed yet, see [2910.21](https://travis-ci.org/robotology/yarp/jobs/174561371) and [2910.22](https://travis-ci.org/robotology/yarp/jobs/174561372)

Tobias-Fischer, 2016-11-10:
Just a thought: `m_transform_storage` is not initialized as `nullptr` in the `FrameTransformClient` constructor. Not sure whether this could cause trouble ..

traversaro, 2017-09-23:
Have we stopped updating this page, or have the random crash decreased over time? Most of them never occurred in 2017, apparently.

drdanz, 2017-09-25:
I'm afraid we just stopped updating the page... 💩

traversaro, 2017-10-28:
The `OS::RateThreadTest` segfault just happened on a build, updated the page.

traversaro, 2017-12-01:
Added https://github.com/robotology/yarp/issues/1478 .

traversaro, 2018-08-22:
New entry:  `dev::MultipleAnalogSensorsInterfacesTest::Valgrind::MemCheck`, with 2018/08/22 - `master` - [6040.19](https://travis-ci.org/robotology/yarp/jobs/419104010) - `***Failed    3.19 sec` .

closed 2025-01-22 by randaz81