robotology/yarp #881: Non-deterministic crash in ControlBoardRemapperTest

traversaro, 2016-08-20:
Local failure after ~400 trials, on Ubuntu 16.04 : 

~~~
Test #246: dev::ControlBoardRemapperTest ....***Exception: Other  2.14 sec
yarp: 0 | ControlBoardRemapperTest: 
test the controlboard remapper
yarp(c19bf740): Scanning. I'm scanning. I hope you like scanning too.
yarp(c19bf740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/yarp/plugins.
yarp(c19bf740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/iCub/plugins.
yarp(c19bf740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/codyco/plugins.
yarp(c19bf740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/external/YARP/share/yarp/plugins.
yarp(c19bf740): Plugin [name: fakeMotionControl] [dll: yarp_fakeMotionControl] [fn: fakeMotionControl]
yarp(c19bf740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/install/lib/iCub/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(c19bf740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/install/lib...
The rest of the test output was removed since it exceeds the threshold of 1024 bytes.
yarp: 0 | ControlBoardRemapperTest: 
test the controlboard remapper
yarp(cea91740): Scanning. I'm scanning. I hope you like scanning too.
yarp(cea91740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/yarp/plugins.
yarp(cea91740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/iCub/plugins.
yarp(cea91740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/install/share/codyco/plugins.
yarp(cea91740): Loading configuration files related to plugins from /home/pegua/src/codyco-superbuild/build/external/YARP/share/yarp/plugins.
yarp(cea91740): Plugin [name: fakeMotionControl] [dll: yarp_fakeMotionControl] [fn: fakeMotionControl]
yarp(cea91740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/install/lib/iCub/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/install/lib/codyco/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Found plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Plugin [name: ] [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Found plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
[WARNING]AxisMap  parameter not found 
[INFO]FakeMotionControl: Using default AxisMap 
[WARNING]AxisType  parameter not found 
[INFO]FakeMotionControl: Using default AxisType (revolute) 
[WARNING]Encoder  parameter not found 
[INFO]Using default Encoder 
[INFO]created device <fakeMotionControl>. See C++ class yarp::dev::FakeMotionControl for documentation.
yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard open reported successful] passed ok
yarp: 0 | ControlBoardRemapperTest:   [interface position correctly opened] passed ok
yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard seems functional] passed ok
[INFO]/testRemapperRobot/a : no ROS initialization required 
[INFO]/testRemapperRobot/a  initting YARP initialization 
yarp(cea91740): register contact /testRemapperRobot/a/rpc:i
yarp: Port /testRemapperRobot/a/rpc:i active at tcp://127.0.0.1:10002/
yarp(cea91740): register contact /testRemapperRobot/a/command:i
yarp: Port /testRemapperRobot/a/command:i active at tcp://127.0.0.1:10012/
yarp(cea91740): register contact /testRemapperRobot/a/state:o
yarp: Port /testRemapperRobot/a/state:o active at tcp://127.0.0.1:10022/
yarp(cea91740): register contact /testRemapperRobot/a/stateExt:o
yarp: Port /testRemapperRobot/a/stateExt:o active at tcp://127.0.0.1:10032/
[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.
yarp: 0 | ControlBoardRemapperTest:   [controlboardwrapper2 open reported successful] passed ok
yarp: 0 | ControlBoardRemapperTest:   [interface for multiple wrapper correctly opened for the controlboardwrapper2] passed ok
yarp: 0 | ControlBoardRemapperTest:   [controlboardwrapper2 attached successfully to the device] passed ok
yarp(cea91740): Plugin [name: ] [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Trying plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(cea91740): Found plugin [dll: /home/pegua/src/codyco-superbuild/build/external/YARP/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
[WARNING]AxisMap  parameter not found 
[INFO]FakeMotionControl: Using default AxisMap 
[WARNING]AxisType  parameter not found 
[INFO]FakeMotionControl: Using default AxisType (revolute) 
[WARNING]Encoder  parameter not found 
[INFO]Using default Encoder 
[INFO]created device <fakeMotionControl>. See C++ class yarp::dev::FakeMotionControl for documentation.
yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard open reported successful] passed ok
yarp: 0 | ControlBoardRemapperTest:   [interface position correctly opened] passed ok
yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard seems functional] passed ok
[INFO]/testRemapperRobot/b : no ROS initialization required 
[INFO]/testRemapperRobot/b  initting YARP initialization 
yarp(cea91740): register contact /testRemapperRobot/b/rpc:i
[... 1063 lines cut ...]
yarp(cea91740): asking tcp://testRemapperRobot/a/state:o: [list] [out] "/test/remoteControlBoardRemapper/testRemapperRobot/a/state:i"
yarp(6effd700): TextCarrier::expectSenderSpecifier
yarp(cea91740): ** asking tcp://testRemapperRobot/a/state:o: [del] "/test/remoteControlBoardRemapper/testRemapperRobot/a/state:i"
yarp(4d7fa700): TextCarrier::expectSenderSpecifier
yarp(c96bc700): output for route /testRemapperRobot/a/state:o->udp->/test/remoteControlBoardRemapper/testRemapperRobot/a/state:i asking other side to close by out-of-band means
yarp(c96bc700): query name /test/remoteControlBoardRemapper/testRemapperRobot/a/state:i
yarp(37fff700): TextCarrier::expectSenderSpecifier
yarp(37fff700): Port command (admin->text->/test/remoteControlBoardRemapper/testRemapperRobot/a/state:i): /test/remoteControlBoardRemapper/testRemapperRobot/a/state:i should remove input: ~/testRemapperRobot/a/state:o
yarp: Removing output from /testRemapperRobot/a/state:o to /test/remoteControlBoardRemapper/testRemapperRobot/a/state:i
yarp(4cff9700): Total message length: 410
yarp(cea91740): working on connection /testRemapperRobot/a/stateExt:o to /test/remoteControlBoardRemapper/testRemapperRobot/a/stateExt:i (disconnect)
yarp(cea91740): query name /testRemapperRobot/a/stateExt:o
yarp(cea91740): query name /test/remoteControlBoardRemapper/testRemapperRobot/a/stateExt:i
yarp(cea91740): asking tcp://testRemapperRobot/a/stateExt:o: [list] [out] "/test/remoteControlBoardRemapper/testRemapperRobot/a/stateExt:i"
yarp(6d7fa700): TextCarrier::expectSenderSpecifier
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

0% tests passed, 1 tests failed out of 1

Total Test time (real) = 498.74 sec

The following tests FAILED:
	246 - dev::ControlBoardRemapperTest (OTHER_FAULT)
~~~

traversaro, 2016-08-20:
Crash in Travis (from #880 ) : 
~~~
58/68 Test #58: dev::ControlBoardRemapperTest ......***Exception: SegFault  1.30 sec
yarp: 0 | ControlBoardRemapperTest: 
test the controlboard remapper
yarp(a49ff0bf): Scanning. I'm scanning. I hope you like scanning too.
yarp(a49ff0bf): Loading configuration files related to plugins from /Users/travis/build/robotology/yarp/build/share/yarp/plugins.
yarp(a49ff0bf): Plugin [name: fakeMotionControl] [dll: yarp_fakeMotionControl] [fn: fakeMotionControl]
yarp(a49ff0bf): Trying plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(a49ff0bf): Found plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(a49ff0bf): Plugin [name: ] [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(a49ff0bf): Trying plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
yarp(a49ff0bf): Found plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]
[WARNING]AxisMap  parameter not found 
[INFO]FakeMotionControl: Using default AxisMap 

[WARNING]AxisType  parameter not found 

[INFO]FakeMotionControl: Using default AxisType (revolute) 

[WARNING]Encoder  parameter not found 

[INFO]Using default Encoder 

[INFO]created device <fakeMotionControl>. See C++ class yarp::dev::FakeMotionControl for documentation.

yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard open reported successful] passed ok

yarp: 0 | ControlBoardRemapperTest:   [interface position correctly opened] passed ok

yarp: 0 | ControlBoardRemapperTest:   [fakeMotionControlBoard seems functional] passed ok

[INFO]/testRemapperRobot/a : no ROS initialization required 

[INFO]/testRemapperRobot/a  initting YARP initialization 

yarp(a49ff0bf): register contact /testRemapperRobot/a/rpc:i

yarp: Port /testRemapperRobot/a/rpc:i active at tcp://127.0.0.1:10002/

yarp(a49ff0bf): register contact /testRemapperRobot/a/command:i

yarp: Port /testRemapperRobot/a/command:i active at tcp://127.0.0.1:10012/

yarp(a49ff0bf): register contact /testRemapperRobot/a/state:o

yarp: Port /testRemapperRobot/a/state:o active at tcp://127.0.0.1:10022/

yarp(a49ff0bf): register contact /testRemapperRobot/a/stateExt:o

yarp: Port /testRemapperRobot/a/stateExt:o active at tcp://127.0.0.1:10032/

[INFO]created wrapper <controlboardwrapper2>. See C++ class yarp::dev::ControlBoardWrapper2 for documentation.

yarp: 0 | ControlBoardRemapperTest:   [controlboardwrapper2 open reported successful] passed ok

yarp: 0 | ControlBoardRemapperTest:   [interface for multiple wrapper correctly opened for the controlboardwrapper2] passed ok

yarp: 0 | ControlBoardRemapperTest:   [controlboardwrapper2 attached successfully to the device] passed ok

yarp(a49ff0bf): Plugin [name: ] [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]

yarp(a49ff0bf): Trying plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]

yarp(a49ff0bf): Found plugin [dll: /Users/travis/build/robotology/yarp/build/lib/yarp/yarp_fakeMotionControl.so] [fn: fakeMotionControl]

[WARNING]AxisMap  parameter not found 

[INFO]FakeMotionControl: Using default AxisMap 

[... 1343 lines cut ...]

yarp(a49ff0bf): asking tcp://testRemapperRobot/b/stateExt:o: [list] [out] "/test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i"

yarp: Removing input from /testRemapperRobot/b/stateExt:o to /test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i

yarp(e26e8e61): TextCarrier::expectSenderSpecifier

yarp(a49ff0bf): ** asking tcp://testRemapperRobot/b/stateExt:o: [del] "/test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i"

yarp(6aa5a574): TextCarrier::expectSenderSpecifier

yarp(78209f73): output for route /testRemapperRobot/b/stateExt:o->udp->/test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i asking other side to close by out-of-band means

yarp(78209f73): query name /test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i

yarp(cd97e3f9): TextCarrier::expectSenderSpecifier

yarp(cd97e3f9): Port command (admin->text->/test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i): /test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i should remove input: ~/testRemapperRobot/b/stateExt:o

yarp: Removing output from /testRemapperRobot/b/stateExt:o to /test/remoteControlBoardRemapper/testRemapperRobot/b/stateExt:i

yarp: 1 | ControlBoardRemapperTest:   FAILURE in file /Users/travis/build/robotology/yarp/tests/libYARP_dev/ControlBoardRemapperTest.cpp:302 [remotecontrolboardremapper open reported successful, testing it] ok (0) == true (1)

yarp: 1 | ControlBoardRemapperTest:   FAILURE in file /Users/travis/build/robotology/yarp/tests/libYARP_dev/ControlBoardRemapperTest.cpp:75 [interface position correctly opened] ok (0) == true (1)
~~~

traversaro, 2016-08-20:
Error in https://s3.amazonaws.com/archive.travis-ci.org/jobs/152940464/log.txt : log too big .

traversaro, 2016-12-21:
https://travis-ci.org/robotology/yarp/jobs/185471895 : `***Timeout 1500.00 sec`.

traversaro, 2018-04-10:
By the way, I am quite sure this is not directly related to the `ControlBoardRemapper`, it is just that the `controlboardwrapper` and the `remote_controlboard` do not have any test specific for just testing them.

randaz81, 2023-11-23:
Probably fixed. Feel free to reopen if needed.

closed 2023-11-23 by randaz81