robotology/yarp #882: Keep formating of responses in RPC

gregoire-pointeau, 2016-08-22:
Hi,
I have a silly "useless" question: it is possible to display the formating of a response through an RPC correctly?
Let me explain. I have a module A with different functions inside and a `rpc` port let's say: `/A/rpc`
I open a terminal and connect it to `/A/rpc` and I send a command expecting an answer.
The answer is created has follow:
`string answer = "part 1 + \n" + "part 2"`
In A, `print answer` gives:
> part1
part2

while in my console, the response is:
>console: yarp rpc /A/rpc
query
Response: "part1 \n part2"

This is not really important but I would like to have an interface a bit more formalized if possible.

Thanks in advance for your help.

pattacini, 2016-08-22:
Hi @gregoire-pointeau 

One possibility would be formatting your reply like this:

```cpp
yarp::os::Bottle reply;
reply.addVocab(Vocab::encode("many"));
reply.addString("part 1 \n part 2");
```

gregoire-pointeau, 2016-08-22:
Perfect ! thanks Ugo !

closed 2016-08-22 by gregoire-pointeau