robotology/yarp #885: Introducing user defined (environment) variables in application description files

nunoguedelha, 2016-08-24:
It would be very useful to add more flexibility to the way we can parametrise an application (defined by a **yarpmanager** application description file), by introducing user defined environment variables which scope would be limited to the application, as it is done in the `icub-tests` xml scripts, where such parameters are wrapped by the tag `<environment>`.

Currently, the closer we have to that is the `<prefix>` mechanism, which allows to set a prefix to all the ports opened by one or a some of the modules run by the application. For instance :
```
<!-- Embedded application -->
   <application>
      <name>Application_Name</name>
      <prefix>icubGazeboSim</prefix>
   </application>
   ...
```
In some cases this is not enough. Let's assume we write an application which defines connections between ports that have different prefixes (using the `external` parameter doesn't fully solve the problem), or opens a port for which the variable part in the name is not a prefix...

Even if other solutions exist, using a user defined variable here would be much simpler and efficient, like:
```
<!-- Embedded application -->
   <application>
      <name>Application_Name</name>
      <environment>--robotname icubGazeboSim --context leftLeg</environment >
   </application>
   ...
```

nunoguedelha, 2016-08-24:
@apaikan , as per our last discussion.

apaikan, 2016-12-07:
It would be very useful to also have the system variable into the xml file. For example: 

- From bash:
```bash
export BLADE_1=/icub20
```
- Within the application xml file:
```xml
<application>
   <variables>
       <variable name="VAR1"> 200 </variable>
       <variable name="VAR2"> 300 </variable>
       ...
   </variables>
   <module>
        <name>yarpview</name>
        <parameters>--x ${VAR1} --y ${VAR2}</parameters>
        <node>${BLADE_1}</node>
        ...
   </module>
</application>
```

For the above example, `yarpmanager` will substituted the variables with the user-defined values and/or with the system environment variables.

traversaro, 2016-12-07:
In CMake to avoid confusion we CMake variable and system environment variables have two different syntaxes (`${VAR}` and `$ENV{VAR}`) to avoid confusions.

apaikan, 2016-12-07:
@traversaro  Isn't better to have only one type of variables (e.g. `${VAR}`) which can be overwritten in the XML file if needed?

aerydna, 2016-12-19:
fixed by #1031

PR #1031 (2017-02-07): Yarpmanager impro
fixes #885 
waiting an @apaikan check to quickly extend to other xmlparsing classes (module resources etc).

`$ENV{foo}` will access to the environment variable "foo"

`<var name="bar">baz</var>` will create a baz-valued, bar variable
then
`${bar}` will access to the "bar" variable and so will give you the value "baz"
closed 2017-02-22 by apaikan