robotology/yarp #888: rpc server send formatted string

rlober, 2016-08-30:
Hello all,

I have an `RpcServer` running and when a client is connected and requests the string `help` I put a formatted string in a bottle and then send that in the reply. Unfortunately when I try to use this functionality from the command line, i.e.
```
yarp rpc /server_port/rpc:i
>> help
```
The output is,
```
Response: "The help string with all escape sequences e.g. '\n' as plain text..."
```
I assume this is because the `yarp rpc` tool just automatically calls the `toString()` method on the reply?

Is there any way to send the string so that the user gets the reply with the formatting? Do I need to define a datatype and overload the `toString()` method?

Here is the relevant code: https://github.com/robotology/codyco-modules/blob/jtc_mods/src/devices/jointTorqueControl/JointTorqueControl.cpp#L85-L109

Thanks in advance!

pattacini, 2016-08-30:
Hi @rlober 

I'd point you to https://github.com/robotology/QA/issues/132.
Let's comment further therein, if you'll have other doubts.

Thus, closing here.

closed 2016-08-30 by pattacini

rlober, 2016-08-30:
@pattacini  Beautiful! Thanks man.

for the lazy:

**solution**
``` 
std::string bigString = // Format big string;
replyBottle.addVocab(yarp::os::Vocab::encode("many"));  // The magical command.
replyBottle.addString(bigString);
```

Output goes from: 
```
Response: "\n===============\nValid Commands\n===============\nNote: Omit the brackets [] when writing your commands.\n\n[save] [location] - saves all current parameters to a txt file at the specified [location] (defaults to ./).\n[get] [joint_list] - returns a list of strings with the joint labels.\n[get] [no_joints] - returns the number of joints in the part.\n[get] [params] [joint_index] - get all parameter values for a joint index.\n[get] [param_id] [joint_index] - get the parameter value for a joint index.\n[set] [joint_index] [param_id] [param_value] [param_id] [param_value] ... - set the parameter value(s) for a joint index.\n\n  Valid [param_id] values:\n  --> kp - proportional gain (Joint Torque Loop)\n  --> ki - integral gain (Joint Torque Loop)\n  --> kd - proportional gain (Joint Torque Loop)\n  --> max_int - ??? (Joint Torque Loop)\n  --> max_pwm - ??? (Joint Torque Loop)\n  --> kv - proportional gain ??? (Motor Parameters)\n  --> kcp - ??? (Motor Parameters)\n  --> kcn - ??? (Motor Parameters)\n  --> coulombVelThr - ??? (Motor Parameters)\n  --> kff - feed-forward gain for gravity compensation torque (Motor Parameters)\n  --> frictionCompensation - coefficient of friction compensation (0-1) (Motor Parameters)\n"
```

to...

```
Responses:
  
===============
Valid Commands
===============
Note: Omit the brackets [] when writing your commands.

[save] [location] - saves all current parameters to a txt file at the specified [location] (defaults to ./).
[get] [joint_list] - returns a list of strings with the joint labels.
[get] [no_joints] - returns the number of joints in the part.
[get] [params] [joint_index] - get all parameter values for a joint index.
[get] [param_id] [joint_index] - get the parameter value for a joint index.
[set] [joint_index] [param_id] [param_value] [param_id] [param_value] ... - set the parameter value(s) for a joint index.

  Valid [param_id] values:
  --> kp - proportional gain (Joint Torque Loop)
  --> ki - integral gain (Joint Torque Loop)
  --> kd - proportional gain (Joint Torque Loop)
  --> max_int - ??? (Joint Torque Loop)
  --> max_pwm - ??? (Joint Torque Loop)
  --> kv - proportional gain ??? (Motor Parameters)
  --> kcp - ??? (Motor Parameters)
  --> kcn - ??? (Motor Parameters)
  --> coulombVelThr - ??? (Motor Parameters)
  --> kff - feed-forward gain for gravity compensation torque (Motor Parameters)
  --> frictionCompensation - coefficient of friction compensation (0-1) (Motor Parameters)
```

pattacini, 2016-08-30:
Btw @rlober did you know that with [**`Thrift`**](http://www.yarp.it/tutorials.html) you'll have rpc services documentation _automagically_ done for you? :wink:

rlober, 2016-08-30:
@pattacini very interesting! I will have to play with this tool.