robotology/yarp #899: Unify DgramTwoWayStream code for linux and macOS

francesco-romano, 2016-09-02:
After #862 the code in linux and macOS are now (completely) different as the mac code does not run on linux.
In particular, even if the connection is successfully added, both the sender and the receiver hangs.

We should investigate this issue as having duplicate code is not ideal (especially if there is no reason why they should be different!)

francesco-romano, 2017-09-25:
Debugging this issue with @Nicogene, we found out two interesting issues in Linux (behaving differently from BSD-like systems):

- the value returned by `getsockopt` is "doubled" for some unknown reasons (see https://linux.die.net/man/7/socket)
- the value returned by Linux for the socket buffer size is too big compared to the maximum UDP datagram size allowed by IPv4 protocol, eg. 65k (and something). This will cause an `EMSGSIZE` error when sending. For example, see comparison between macOS:
```
makas-pro:build makaveli$ sysctl -a | grep udp
net.inet.udp.maxdgram: 9216
net.inet.udp.recvspace: 196724
```
and Linux
```
net.core.rmem_default = 212992
net.core.wmem_default = 212992
```
Note that these two last options are probably generic for all sockets (TCP and UDP) while specific UDP options are the following:
```
net.ipv4.udp_mem = 185607    247476    371214
net.ipv4.udp_rmem_min = 4096
net.ipv4.udp_wmem_min = 4096
```
but they do not seem to be read from the C-level API

Nicogene, 2017-09-25:
@lornat75 @drdanz  do you have some time for a F2F discussion about `udp`?
We also discovered this: https://stackoverflow.com/questions/9853099/how-to-solve-sending-udp-packet-using-sendto-got-message-too-long#comment12560885_9853210.

PR #1366 (2017-10-09): Unification of DGramTwoWayStream
This PR unifies the code in `DGramTwoWayStream` and deletes a useless `while()` in `flush()`, the packet is already divided in packets in the `write()` function in order to be sent through datagrams.
It fixes #899.

TODO:
- [x] Tests on Linux (with/without ACE) 
- [x] Tests on macOs (with/without ACE) 
- [x] Tests on Windows (with ACE) 

All tests passed, ready to be merged.
closed 2017-10-09 by drdanz