robotology/yarp #903: YARP master compilation fails (tinyXML related)

kt10aan, 2016-09-06:
Hello,

YARP from master (https://github.com/robotology/yarp/commit/b447d15718383f1fd97ea6175e421ff1d083f647) fails to compile. The problem is related to tinyXML and in ``yarprobotinterface``.
Please note that I am using the tinyXML from the ``extern`` folder, I have not installed the ``libtinyxml2-dev`` package.
System is ``Ubuntu 16.04``.

Here is the output of make:

```
[ 67%] Linking CXX executable ../../bin/yarprobotinterface
CMakeFiles/yarprobotinterface.dir/XMLReader.cpp.o: In function `RobotInterface::XMLReader::Private::readActionTag(TiXmlElement*)':
XMLReader.cpp:(.text+0x515e): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
XMLReader.cpp:(.text+0x5679): undefined reference to `TiXmlAttributeSet::Find(std::string const&) const'
collect2: error: ld returned 1 exit status
src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/build.make:332: recipe for target 'bin/yarprobotinterface' failed
make[2]: *** [bin/yarprobotinterface] Error 1
CMakeFiles/Makefile2:2524: recipe for target 'src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all' failed
make[1]: *** [src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all] Error 2
Makefile:149: recipe for target 'all' failed                                                                                                                                                                      
make: *** [all] Error 2
```

traversaro, 2016-09-06:
After uninstalling `libtinyxml-dev`, I am unable to reproduce the issue. 
Can you post your `CMakeCache.txt` and the output of `make VERBOSE=1` ?

kt10aan, 2016-09-08:
Here is my ``CMakeCache.txt`` and the ``make VERBOSE`` output:

[CMakeCache.txt](https://github.com/robotology/yarp/files/461599/CMakeCache.txt)

```
kostas@terra:~/robotcub/yarp/build$ make VERBOSE=1
/usr/bin/cmake -H/home/kostas/robotcub/yarp -B/home/kostas/robotcub/yarp/build --check-build-system CMakeFiles/Makefile.cmake 0                                                                                   
/usr/bin/cmake -E cmake_progress_start /home/kostas/robotcub/yarp/build/CMakeFiles /home/kostas/robotcub/yarp/build/CMakeFiles/progress.marks                                                                     
make -f CMakeFiles/Makefile2 all                                                                                                                                                                                  
make[1]: Entering directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                    
make -f extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/build.make extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/depend                                                         
make[2]: Entering directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                    
cd /home/kostas/robotcub/yarp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kostas/robotcub/yarp /home/kostas/robotcub/yarp/extern/qcustomplot /home/kostas/robotcub/yarp/build /home/kostas/robotcub/yarp/build/extern/qcustomplot /home/kostas/robotcub/yarp/build/extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/DependInfo.cmake --color=                                                    
make[2]: Leaving directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                     
make -f extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/build.make extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/build                                                          
make[2]: Entering directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                    
[  0%] Automatic moc for target YARP_priv_qcustomplot                                                                                                                                                             
cd /home/kostas/robotcub/yarp/build/extern/qcustomplot && /usr/bin/cmake -E cmake_autogen /home/kostas/robotcub/yarp/build/extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot_automoc.dir/ Release               
AUTOGEN: Checking /home/kostas/robotcub/yarp/extern/qcustomplot/qcustomplot/qcustomplot.cpp                                                                                                                       
AUTOGEN: Checking /home/kostas/robotcub/yarp/extern/qcustomplot/qcustomplot/qcustomplot.h                                                                                                                         
make[2]: Leaving directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                     
[  0%] Built target YARP_priv_qcustomplot_automoc                                                                                                                                                                 
make -f extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/build.make extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/depend                                                                         
make[2]: Entering directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                    
cd /home/kostas/robotcub/yarp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/kostas/robotcub/yarp /home/kostas/robotcub/yarp/extern/qcustomplot /home/kostas/robotcub/yarp/build /home/kostas/robotcub/yarp/build/extern/qcustomplot /home/kostas/robotcub/yarp/build/extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/DependInfo.cmake --color=                                                            
make[2]: Leaving directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                     
make -f extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/build.make extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/build                                                                          
make[2]: Entering directory '/home/kostas/robotcub/yarp/build'                                                                                                                                                    
make[2]: Nothing to be done for 'extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/build'.                                                                                                                  
[... cut ...]
src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/build.make:332: recipe for target 'bin/yarprobotinterface' failed
make[2]: *** [bin/yarprobotinterface] Error 1
make[2]: Leaving directory '/home/kostas/robotcub/yarp/build'
CMakeFiles/Makefile2:2524: recipe for target 'src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all' failed
make[1]: *** [src/yarprobotinterface/CMakeFiles/yarprobotinterface.dir/all] Error 2
make[1]: Leaving directory '/home/kostas/robotcub/yarp/build'
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

drdanz, 2016-09-21:
@kt10aan `-ltinyxml` in the link command means that you are actually linking TinyXML installed somewhere on your system.
Also the CMakeCache.txt shows that you are using it:

```

//Additional compiler flags for TinyXML
TinyXML_DEFINITIONS:STRING=

//The directory containing a CMake configuration file for TinyXML.
TinyXML_DIR:PATH=TinyXML_DIR-NOTFOUND

//TinyXML include directory
TinyXML_INCLUDE_DIRS:PATH=

//Path to a library.
TinyXML_TINYXML_LIBRARY_DEBUG:FILEPATH=TinyXML_TINYXML_LIBRARY_DEBUG-NOTFOUND

//Path to a library.
TinyXML_TINYXML_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libtinyxml.so

//Use system-installed TinyXML, rather than a private copy (recommended)
YARP_USE_SYSTEM_TINYXML:BOOL=ON

//Use package TinyXML
YARP_USE_TINYXML:BOOL=ON

//Details about finding TinyXML
FIND_PACKAGE_MESSAGE_DETAILS_TinyXML:INTERNAL=[/usr/lib/x86_64-linux-gnu/libtinyxml.so][v(2.6)]
//ADVANCED property for variable: TinyXML_DEFINITIONS
TinyXML_DEFINITIONS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: TinyXML_DIR
TinyXML_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: TinyXML_INCLUDE_DIRS
TinyXML_INCLUDE_DIRS-ADVANCED:INTERNAL=1
//ADVANCED property for variable: TinyXML_TINYXML_LIBRARY_DEBUG
TinyXML_TINYXML_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
//ADVANCED property for variable: TinyXML_TINYXML_LIBRARY_RELEASE
TinyXML_TINYXML_LIBRARY_RELEASE-ADVANCED:INTERNAL=1
```

If you uninstalled TinyXML, try running `cmake . -U*TinyXML* -U*TINYXML*` (that will remove all TinyXML related variables from your CMakeCache, and try again.

Otherwise you can just set `YARP_USE_SYSTEM_TINYXML=OFF` that will ensure that you are using the `extern` copy of the library.

closed 2016-09-28 by drdanz