robotology/yarp #904: Windows 8 java compilation error

vickyvou, 2016-09-06:
Hello,
i am trying to compile YAPR to now a Windows 8 machine with JAVA bindings. During compilation i get the following error:
`1>------ Build started: Project: ZERO_CHECK, Configuration: Release Win32 ------
2>------ Build started: Project: yarpcar, Configuration: Release Win32 ------
2>  add_yarpcar_plugins.cpp
2>     Creating library D:/yarp-2.3.63/build/lib/Release/yarpcar.lib and object D:/yarp-2.3.63/build/lib/Release/yarpcar.exp
2>  yarpcar.vcxproj -> D:\yarp-2.3.63\build\lib\Release\yarpcar.dll
3>------ Build started: Project: YARP_init, Configuration: Release Win32 ------
3>     Creating library D:/yarp-2.3.63/build/lib/Release/YARP_init.lib and object D:/yarp-2.3.63/build/lib/Release/YARP_init.exp
3>  YARP_init.vcxproj -> D:\yarp-2.3.63\build\lib\Release\YARP_init.dll
4>------ Build started: Project: jyarp, Configuration: Release Win32 ------
5>------ Build started: Project: gyarpmanager, Configuration: Release Win32 ------
6>------ Build started: Project: yarpview, Configuration: Release Win32 ------
7>------ Build started: Project: testModule, Configuration: Release Win32 ------
8>------ Build started: Project: yarp-config, Configuration: Release Win32 ------
9>------ Build started: Project: yarpdev, Configuration: Release Win32 ------
10>------ Build started: Project: yarpidl_rosmsg, Configuration: Release Win32 ------
11>------ Build started: Project: harness_os, Configuration: Release Win32 ------
6>  yarpview.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpview.exe
8>  yarp-config.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarp-config.exe
12>------ Build started: Project: harness_dev, Configuration: Release Win32 ------
13>------ Build started: Project: harness_math, Configuration: Release Win32 ------
7>  testModule.vcxproj -> D:\yarp-2.3.63\build\bin\Release\testModule.exe
4>     Creating library D:/yarp-2.3.63/build/lib/Release/jyarp.lib and object D:/yarp-2.3.63/build/lib/Release/jyarp.exp
5>  gyarpmanager.vcxproj -> D:\yarp-2.3.63\build\bin\Release\gyarpmanager.exe
12>  harness_dev.vcxproj -> D:\yarp-2.3.63\build\bin\Release\harness_dev.exe
4>  jyarp.vcxproj -> D:\yarp-2.3.63\build\lib\Release\jyarp.dll
10>  yarpidl_rosmsg.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpidl_rosmsg.exe
14>------ Build started: Project: harness_sig, Configuration: Release Win32 ------
9>  yarpdev.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpdev.exe
15>------ Build started: Project: yarp, Configuration: Release Win32 ------
13>  harness_math.vcxproj -> D:\yarp-2.3.63\build\bin\Release\harness_math.exe
16>------ Build started: Project: yarpmanager, Configuration: Release Win32 ------
11>  harness_os.vcxproj -> D:\yarp-2.3.63\build\bin\Release\harness_os.exe
17>------ Build started: Project: yarprun, Configuration: Release Win32 ------
14>  harness_sig.vcxproj -> D:\yarp-2.3.63\build\bin\Release\harness_sig.exe
18>------ Build started: Project: yarpscope, Configuration: Release Win32 ------
17>  yarprun.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarprun.exe
15>  yarp.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarp.exe
19>------ Build started: Project: jyarp_classes, Configuration: Release Win32 ------
20>------ Build started: Project: yarpserver, Configuration: Release Win32 ------
21>------ Build started: Project: yarpserver2, Configuration: Release Win32 ------
22>------ Build started: Project: yarpserver3, Configuration: Release Win32 ------
16>  yarpmanager.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpmanager.exe
19>  Building Custom Rule D:/yarp-2.3.63/bindings/CMakeLists.txt
19>  CMake does not need to re-run because D:\yarp-2.3.63\build\bindings\CMakeFiles\generate.stamp is up-to-date.
19>  Generating ../src/Example1.class, ../src/LoadYarp.class, ../src/YarpImageHelper.class, ../yarp/Bottle.class, ../yarp/BottleCallback.class, ../yarp/BufferedPortBottle.class, ../yarp/BufferedPortImageFloat.class, ../yarp/BufferedPortImageInt.class, ../yarp/BufferedPortImageMono.class, ../yarp/BufferedPortImageRgb.class, ../yarp/BufferedPortImageRgba.class, ../yarp/BufferedPortImageRgbFloat.class, ../yarp/BufferedPortProperty.class, ../yarp/BufferedPortSound.class, ../yarp/BVector.class, ../yarp/CartesianEvent.class, ../yarp/CartesianEventParameters.class, ../yarp/CartesianEventVariables.class, ../yarp/charArray.class, ../yarp/ConnectionReader.class, ../yarp/ConnectionWriter.class, ../yarp/Contact.class, ../yarp/Contactable.class, ../yarp/ContactStyle.class, ../yarp/DeviceDriver.class, ../yarp/DeviceResponder.class, ../yarp/DriverCreator.class, ../yarp/Drivers.class, ../yarp/DummyConnector.class, ../yarp/DVector.class, ../yarp/FlexImage.class, ../yarp/GazeEvent.class, ../yarp/GazeEventParameters.class, ../yarp/GazeEventVariables.class, ../yarp/IAmplifierControl.class, ../yarp/IAmplifierControlRaw.class, ../yarp/IAudioVisualGrabber.class, ../yarp/IAudioVisualStream.class, ../yarp/IAxisInfo.class, ../yarp/ICalibrator.class, ../yarp/ICartesianControl.class, ../yarp/IConfig.class, ../yarp/IControlCalibration.class, ../yarp/IControlCalibration2.class, ../yarp/IControlCalibration2Raw.class, ../yarp/IControlCalibrationRaw.class, ../yarp/IControlDebug.class, ../yarp/IControlLimits.class, ../yarp/IControlLimitsRaw.class, ../yarp/IControlMode.class, ../yarp/IControlModeRaw.class, ../yarp/IEncoders.class, ../yarp/IEncodersRaw.class, ../yarp/IFrameGrabber.class, ../yarp/IFrameGrabberControls.class, ../yarp/IFrameGrabberControlsDC1394.class, ../yarp/IFrameGrabberImage.class, ../yarp/IFrameGrabberImageRaw.class, ../yarp/IFrameGrabberRgb.class, ../yarp/IFrameWriterAudioVisual.class, ../yarp/IFrameWriterImage.class, ../yarp/IGazeControl.class, ../yarp/IImpedanceControl.class, ../yarp/IImpedanceControlRaw.class, ../yarp/Image.class, ../yarp/ImageFloat.class, ../yarp/ImageInt.class, ../yarp/ImageMono.class, ../yarp/ImageRgb.class, ../yarp/ImageRgba.class, ../yarp/ImageRgbFloat.class, ../yarp/IOpenLoopControl.class, ../yarp/IOpenLoopControlRaw.class, ../yarp/IPidControl.class, ../yarp/IPidControlRaw.class, ../yarp/IPositionControl.class, ../yarp/IPositionControlRaw.class, ../yarp/ITorqueControl.class, ../yarp/ITorqueControlRaw.class, ../yarp/IVelocityControl.class, ../yarp/IVelocityControlRaw.class, ../yarp/NameStore.class, ../yarp/Network.class, ../yarp/NetworkBase.class, ../yarp/Pid.class, ../yarp/PidVector.class, ../yarp/PixelBgr.class, ../yarp/PixelBgra.class, ../yarp/PixelHsv.class, ../yarp/PixelHsvFloat.class, ../yarp/PixelRgb.class, ../yarp/PixelRgba.class, ../yarp/PixelRgbFloat.class, ../yarp/PixelRgbInt.class, ../yarp/PixelRgbSigned.class, ../yarp/PolyDriver.class, ../yarp/Port.class, ../yarp/Portable.class, ../yarp/PortablePairBase.class, ../yarp/PortReader.class, ../yarp/PortReaderBufferBase.class, ../yarp/PortReaderBufferBaseCreator.class, ../yarp/PortReaderCreator.class, ../yarp/PortReport.class, ../yarp/PortWriter.class, ../yarp/PortWriterBufferBase.class, ../yarp/PortWriterBufferManager.class, ../yarp/PortWriterWrapper.class, ../yarp/Property.class, ../yarp/PropertyCallback.class, ../yarp/Random.class, ../yarp/RateThread.class, ../yarp/RateThreadWrapper.class, ../yarp/ResourceFinder.class, ../yarp/RFModule.class, ../yarp/RpcClient.class, ../yarp/RpcServer.class, ../yarp/Searchable.class, ../yarp/SearchMonitor.class, ../yarp/SearchReport.class, ../yarp/Semaphore.class, ../yarp/Sound.class, ../yarp/Stamp.class, ../yarp/Stamped.class, ../yarp/StubDriverCreator.class, ../yarp/SVector.class, ../yarp/SWIGTYPE_p_bool.class, ../yarp/SWIGTYPE_p_Bytes.class, ../yarp/SWIGTYPE_p_Clock.class, ../yarp/SWIGTYPE_p_double.class, ../yarp/SWIGTYPE_p_float.class, ../yarp/SWIGTYPE_p_f_p_q_const__char_v_______int.class, ../yarp/SWIGTYPE_p_ICalibrator.class, ../yarp/SWIGTYPE_p_int.class, ../yarp/SWIGTYPE_p_NameStore.class, ../yarp/SWIGTYPE_p_NestedContact.class, ../yarp/SWIGTYPE_p_Pid.class, ../yarp/SWIGTYPE_p_p_char.class, ../yarp/SWIGTYPE_p_p_yarp__os__Value.class, ../yarp/SWIGTYPE_p_ResourceFinderOptions.class, ../yarp/SWIGTYPE_p_Runnable.class, ../yarp/SWIGTYPE_p_Searchable.class, ../yarp/SWIGTYPE_p_size_t.class, ../yarp/SWIGTYPE_p_std__string.class, ../yarp/SWIGTYPE_p_Type.class, ../yarp/SWIGTYPE_p_unsigned_char.class, ../yarp/SWIGTYPE_p_unsigned_int.class, ../yarp/SWIGTYPE_p_void.class, ../yarp/SWIGTYPE_p_yarp__os__InputStream.class, ../yarp/SWIGTYPE_p_yarp__os__NestedContact.class, ../yarp/SWIGTYPE_p_yarp__os__OutputStream.class, ../yarp/SWIGTYPE_p_yarp__os__PortInfo.class, ../yarp/SWIGTYPE_p_yarp__os__SizedWriter.class, ../yarp/SWIGTYPE_p_yarp__os__TypedRead [... cut ...]
19>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V110\Microsoft.CppCommon.targets(172,5): error MSB6006: "cmd.exe" exited with code 9009.
21>  yarpserver2.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpserver2.exe
20>  yarpserver.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpserver.exe
22>  yarpserver3.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpserver3.exe
18>  yarpscope.vcxproj -> D:\yarp-2.3.63\build\bin\Release\yarpscope.exe
23>------ Build started: Project: ALL_BUILD, Configuration: Release Win32 ------
========== Build: 22 succeeded, 1 failed, 13 up-to-date, 0 skipped ==========`

any ideas what it could be?
thanks!

francesco-romano, 2016-09-06:
I imagined you installed the Java JDK (not the JRE), and it is properly configured?

vickyvou, 2016-09-06:
Yes, i have installed jdk1.8.0_101. Machine is 64b and the jdk is for 64b
(not sure if this is relevant). JAVA_HOME is set to C:\Program
Files\Java\jdk1.8.0_101.

francesco-romano, 2016-09-07:
Ok.
What happen if you try to run `javac` in the command prompt?

vickyvou, 2016-09-07:
ok, yeah found out why. The bin was not in the path. Thank you for your
help, it all works now!

vtikha, 2016-09-07:
👍

closed 2016-09-07 by vtikha