robotology/yarp #908: Yarp for C# (sender) and Java(receiver) - robot control

abolfazlzaraki, 2016-09-16:
Hello everybody,

I am going to make a network between C# and Java code as sender and receiver of an "String". I decided to use Yarp since it supports different platforms. Does anybody have a sample code or an instruction that helps on that?

Thanks in advance.

Tobias-Fischer, 2016-09-18:
Hi,
First you need to compile the bindings for C# and Java, see http://www.yarp.it/yarp_swig.html
Then, there is example code for [Java](https://github.com/robotology/yarp/blob/master/bindings/src/Example1.java) and [C#](https://github.com/robotology/yarp/blob/master/bindings/example.cs)

For sending strings around, just adapt the very simple example [here](http://www.yarp.it/yarp_code_examples.html).

Hope that helps.

drdanz, 2016-09-21:
@abolfazlzaraki Please let us know if @Tobias-Fischer reply is enough and if we can close this issue, or if you need more help.

abolfazlzaraki, 2016-09-21:
Thanks @Tobias-Fischer. It works for me :)

closed 2016-09-21 by abolfazlzaraki