robotology/yarp #909: Bayer carrier provides no envelope / timestamp

gsaponaro, 2016-09-20:
When reading some data (e.g., image) using the Bayer carrier, the timestamp is always zero. However, when reading the same image using simply the UDP carrier, the timestamp is valid.

This problem possibly affects the mjpeg carrier as well.

See also: https://github.com/robotology/QA/issues/139

/cc @ale-git @lornat75 @barbalberto

drdanz, 2016-09-21:
mjpeg was fixed in 153601a8f0779b410d8e40f8dacb659444093ab1.

The bayer carrier is just a modifying carrier and the envelope is handled by the actual carrier. Anyway, this works for me:

```
$ yarpdev --verbose --device test_grabber --height 3 --width 4 --stamp
```

```
$ yarp read /r envelope
```

```
$ yarp connect /grabber /r udp+recv.bayer
yarp: Port /r active at tcp://192.168.99.3:10006/
yarp: Receiving input from /grabber to /r using udp+recv.bayer
1763 1474447.53142 [mat] [rgb] (3 36 1 4 3) {4 0 8 4 3 4 46 4 0 88 2 0 44 30 8 44 5 4 65 23 0 86 0 0 85 85 8 85 58 4 85 85 0 85 42 0}
1764 1474447.53142 [mat] [rgb] (3 36 1 4 3) {4 0 8 4 3 4 46 4 0 88 2 0 2 1 8 2 5 4 23 2 0 44 0 0 0 0 8 0 1 4 0 0 0 0 0 0}
1765 1474447.53142 [mat] [rgb] (3 36 1 4 3) {4 0 8 4 3 46 46 4 85 88 44 85 2 1 8 2 5 46 23 23 85 44 85 85 0 0 8 0 1 46 0 0 85 0 42 85}
1766 1474447.53142 [mat] [rgb] (3 36 1 4 3) {4 0 1 4 2 0 46 4 0 88 2 0 44 29 1 44 3 0 65 23 0 86 0 0 85 85 1 85 57 0 85 85 0 85 42 0}
```

Perhaps the image you are using does not contain the timestamp?

gsaponaro, 2016-09-21:
I get an envelope with your ```yarp read /r envelope``` method on the console, but not with this kind of code:

```
yarp::sig::ImageOf<yarp::sig::PixelRgb> *yarpImg = imgPort.read(true); // imgPort is a BufferedPort<ImageOf<PixelRgb> >
yarp::os::Stamp stamp;
imgPort.getEnvelope(stamp);
if (stamp.isValid())
    yInfo("timestamp is valid");
else
    yWarning("timestamp is NOT valid");
```

When the program receives the image with ```udp``` => ```timestamp is valid```
When the program receives the image with ```udp+recv.bayer``` => ```timestamp is NOT valid```

I tested both with a test_grabber device and with a real iCub camera grabber/dragonfly2 device.

drdanz, 2016-09-22:
Ok, I see the problem now... and I can also confirm that mjpeg is working, while bayer is not.

I tested it also with a callback instead of read, and the issue remains. I also noticed that using `udp+send.bayer` the envelope is correct... that reminds me of #729 and #782.

So I think that the problem here is that somehow **modifying carriers lose the envelope when the carrier is attached at the receiving side**. I'll retitle the issue accordingly

drdanz, 2016-09-26:
After investigating this, I found out that only the bayer and the portmonitor carriers have this problem, because in `yarp::os::ConnectionReader& modifyIncomingData(yarp::os::ConnectionReader& reader)` they return a different reader that does not know how to read the envelope.

I'm preparing a patch to fix this issue, I'm thinking about adding a `ConnectionReader::setParentConnectionReader` method that will save the main connection reader and to use it to read the envelope when needed. Carriers will have to call this method inside modifyIncomingData when they do not return the same ConnectionReader.

drdanz, 2016-09-26:
Fixed in `devel`

commit f7290e5c (2016-09-26): PortMonitor,Bayer: Set parent connection reader in modifyIncomingData

Fixes #909
Fixes #782
Fixes #729
closed 2016-09-26 by drdanz