robotology/yarp #928: branch devel do not compile

davidetome, 2016-10-14:
CMake Error at cmake/YarpFindDependencies.cmake:127 (message):
   Optional package GSL not found.  Please install it or disable the option
   "CREATE_LIB_MATH_USING_GSL" to build yarp.
 Call Stack (most recent call first):
   cmake/YarpFindDependencies.cmake:396 (check_optional_dependency)
   CMakeLists.txt:62 (include)

@drdanz

pattacini, 2016-10-14:
Hi @davidetome 

Do you have `GSL` installed on you system?
It should be in the path pointed by `GSL_DIR` env variable.

pattacini, 2016-10-14:
@davidetome one other possibility might be to purge the cmake cache and configure yarp from scratch.

davidetome, 2016-10-14:
Ciao @pattacini ,
 I did but didn't work. Actually GSL_DIR is not set in .bashrc_iCub; switching to master it compiles.

closed 2016-10-14 by davidetome

traversaro, 2016-10-14:
It is an Ubuntu 16.04 system?

pattacini, 2016-10-14:
Ok you're on linux. Then, it's very likely that GSL is part of `icub-common`, thus you don't need to have GSL_DIR in the `.bashrc_iCub`.

davidetome, 2016-10-14:
@traversaro yes, it is.